pschatzmann/arduino-audio-tools #317: No Sound Stream Analog input to A2DP Bluetooth - example base-adc-a2dp.ino

janczeresnia, 2022-08-28:
I tested the next example. I used the:
 [image] analog microphone module connected to the ADC GPIO34 in the micro-controller ESP32.
I checked if the microphone was working properly on the example of [base-adc-serial.ino](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/base-adc-serial/base-adc-serial.ino) on the graph there was a signal from the microphone centered on 0 in the range of about plus or minus 1100 as I was talking loudly into the microphone.
Then I ran the [base-adc-a2dp.ino](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/base-adc-a2dp/base-adc-a2dp.ino) example and connected to the computer via bluetooth but there was no sound, pulseaudio uv-meter showed silence on the bluetooth microphone.
I connected a bluetooth microphone to my smartphone, but there was also silence.

log in /var/log/messages

```
Aug 28 16:55:10 nuc7 kernel: input: ESP32_A2DP_SRC (AVRCP) as /devices/virtual/input/input25
Aug 28 16:55:10 nuc7 elogind-daemon[7400]: Watching system buttons on /dev/input/event20 (ESP32_A2DP_SRC (AVRCP))
Aug 28 16:55:10 nuc7 pulseaudio[7587]: [pulseaudio] module-loopback.c: Configured latency of 200.00 ms is smaller than minimum latency, using minimum instead
Aug 28 16:55:10 nuc7 pulseaudio[7587]: [pulseaudio] module-loopback.c: Cannot set requested source latency of 66.67 ms, adjusting to 250.00 ms
```

pschatzmann, 2022-08-28:
I am confused: Pulseaudio and A2DP...
Are you really running the sketch on an ESP32 ?

janczeresnia, 2022-08-28:
Yes

My sketch:
```
/*
 * Bluetooth A2DP Analog Microphone
 * https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-adc-a2dp
 * Microphone is connect to GPIO34
 */

#include "AudioTools.h"
#include "BluetoothA2DPSource.h"

AnalogAudioStream adc;
BluetoothA2DPSource a2dp_source;
// The data has a center of around 26427, so we we need to shift it down to bring the center to 0
ConverterScaler<int16_t> scaler(1.0, -26427, 32767);

// callback used by A2DP to provide the sound data
int32_t get_sound_data(Frame* frames, int32_t count) {
  uint8_t *data = (uint8_t*)frames;
  int channels = 2;
  size_t len = count * channels * sizeof(int16_t);
  // the ADC provides data in 16 bits
  size_t result_len = adc.readBytes(data, len);   
  scaler.convert(data, result_len);
  return result_len;
}

void setup(void) {
  Serial.begin(115200);
  while (!Serial && (millis() < 1000)) delay(1000);
  Serial.println(F("\nESP32 Bluetooth A2DP Analog Microphone\n"));
  //AudioLogger::instance().begin(Serial, AudioLogger::Info);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  // start adc input with default configuration
  Serial.println(F("Starting ADC <-- Microphone"));
  adc.begin(adc.defaultConfig(RX_MODE));

  // start the bluetooth
  Serial.println(F("Starting Bluetooth A2DP"));
  a2dp_source.start("BTMic", get_sound_data);  
}

void loop() {
}
```
I pasted the log from a Linux computer to which I connected ESP32 via bluetooth.

pschatzmann, 2022-08-28:
I think there is a bug in the callback. Try it with:
```

 // callback used by A2DP to provide the sound data
int32_t get_sound_data(Frame* frames, int32_t count) {
    uint8_t *data = (uint8_t*)frames;
    int framesize = 4;
   // the ADC provides data in 16 bits
    size_t result_len = adc.readBytes(data, count*framesize);   
    scaler.convert(data, result_len);
    return result_len/framesize;
}
```

janczeresnia, 2022-08-28:
I tried but still silence.

pschatzmann, 2022-08-28:
Try it with a bluetooth speaker to figure out if the issue is on the linux side ?

janczeresnia, 2022-08-28:
On one ESP32 run a bluetooth A2DP speaker and on the other ESP32 run bluetooth A2DP microphone and connect them together?

I have set Core Debug Level in to the Arduino IDE to None.
Will change to Warning.

pschatzmann, 2022-08-28:
Or debug if you want to check that it is still sending data...

janczeresnia, 2022-08-28:
ESP32 A2DP Microphone won't connect to Linux after compiling with Debug.

Serial log:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[     2][D][BluetoothA2DPSource.cpp:111] BluetoothA2DPSource(): ⸮f(⸮A%u⸮BluetoothA2DPSource, 
[     5][D][esp32-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

ESP32 Bluetooth A2DP Analog Microphone

Starting ADC <-- Microphone
Starting Bluetooth A2DP
[    30][D][BluetoothA2DPSource.cpp:153] start(): [BT_API] start, 
[    32][D][BluetoothA2DPSource.cpp:171] start_raw(): [BT_API] start_raw, 
[    38][D][BluetoothA2DPCommon.cpp:149] get_last_connection(): [BT_AV] get_last_connection
[    46][I][BluetoothA2DPCommon.cpp:163] get_last_connection(): [BT_AV] nvs_blob does not exist
[    55][D][BluetoothA2DPCommon.cpp:172] get_last_connection(): [BT_AV] => 00:00:00:00:00:00
[   600][D][BluetoothA2DPSource.cpp:324] bt_app_task_start_up(): [BT_AV] bt_app_task_start_up
[   601][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 0
[   608][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x0
[   617][D][BluetoothA2DPSource.cpp:532] bt_av_hdl_stack_evt(): [BT_AV] bt_av_hdl_stack_evt evt 0
[   627][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[   636][I][BluetoothA2DPSource.cpp:564] bt_av_hdl_stack_evt(): [BT_AV] Starting device discovery...
[   638][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x4, param len 16
[   655][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x4
[   655][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   664][I][BluetoothA2DPSource.cpp:628] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state 1, evt 0x4
[   673][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   682][D][BluetoothA2DPSource.cpp:596] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
[   689][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   708][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   715][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   723][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   734][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   740][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   750][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 1
[   755][I][BluetoothA2DPSource.cpp:474] bt_app_gap_callback(): [BT_AV] Discovery started.
[ 10644][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 13552][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 1
[ 13552][I][BluetoothA2DPSource.cpp:470] bt_app_gap_callback(): [BT_AV] Device discovery failed, continue to discover...
[ 13563][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 1
[ 13569][I][BluetoothA2DPSource.cpp:474] bt_app_gap_callback(): [BT_AV] Discovery started.
```

Linux log:
```
Aug 28 18:14:47 nuc7 bluetoothd[5934]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 0C:B8:15:D8:F0:AA: Host is down (112)
```

janczeresnia, 2022-08-28:
I tried to connect ESP32 Bluetooth A2DP Microphone and ESP32 Bluetooth A2DP Speaker together, but they don't want to connect either.

Microphone a2dp_source Serial log:
```
ESP32 Bluetooth A2DP Analog Microphone

Starting ADC <-- Microphone
[I] AnalogAudioESP32.h : 68 - I2S_MODE_ADC_BUILT_IN
[I] AnalogAudioESP32.h : 196 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 53 - sample_rate: 44100
[I] AudioTypes.h : 54 - channels: 2
[I] AudioTypes.h : 55 - bits_per_sample: 16
[I] AnalogAudioESP32.h : 94 - input pin1: 34
[I] AnalogAudioESP32.h : 96 - input pin2: 1074347410
[I] AnalogAudioESP32.h : 235 - RX_MODE
Starting Bluetooth A2DP
[    73][D][BluetoothA2DPSource.cpp:153] start(): [BT_API] start, 
[    73][D][BluetoothA2DPSource.cpp:171] start_raw(): [BT_API] start_raw, 
[    75][D][BluetoothA2DPCommon.cpp:149] get_last_connection(): [BT_AV] get_last_connection
[    84][D][BluetoothA2DPCommon.cpp:172] get_last_connection(): [BT_AV] => 0c:b8:15:f8:8e:ae
[   607][D][BluetoothA2DPSource.cpp:324] bt_app_task_start_up(): [BT_AV] bt_app_task_start_up
[   607][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 0
[   615][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x0
[   624][D][BluetoothA2DPSource.cpp:532] bt_av_hdl_stack_evt(): [BT_AV] bt_av_hdl_stack_evt evt 0
[   633][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[   642][W][BluetoothA2DPSource.cpp:558] bt_av_hdl_stack_evt(): [BT_AV] Reconnecting to 0c:b8:15:f8:8e:ae
[   644][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x4, param len 16
[   651][W][BluetoothA2DPCommon.cpp:41] connect_to(): [BT_AV] connect_to to 0c:b8:15:f8:8e:ae
[   662][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   670][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[   679][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   696][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   704][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   711][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   720][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   729][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   737][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[ 21114][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 21114][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 21207][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 24952][D][BluetoothA2DPSource.cpp:851] bt_app_rc_ct_cb(): [RCCT] bt_app_rc_ct_cb evt 0
[ 24952][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 12
[ 24959][D][BluetoothA2DPSource.cpp:851] bt_app_rc_ct_cb(): [RCCT] bt_app_rc_ct_cb evt 5
[ 24967][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x5, param len 12
[ 29374][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 13
[ 29375][I][BluetoothA2DPSource.cpp:519] bt_app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT mode:2
[ 31113][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 41113][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 51113][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 61113][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 71113][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 81113][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 81123][E][BluetoothA2DPSource.cpp:282] bt_app_send_msg(): [BT_API] bt_app_send_msg xQueue send failed
[ 91113][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0

```

Speaker a2dp_sink Serial log:
```
ESP32 Bluetooth A2DP Speaker

Starting DAC
[I] AnalogAudioESP32.h : 71 - I2S_MODE_DAC_BUILT_IN
[I] AnalogAudioESP32.h : 196 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 53 - sample_rate: 44100
[I] AudioTypes.h : 54 - channels: 2
[I] AudioTypes.h : 55 - bits_per_sample: 16
[I] AnalogAudioESP32.h : 91 - analog left output pin: 25
[I] AnalogAudioESP32.h : 92 - analog right output pin: 26
[I] AnalogAudioESP32.h : 264 - TX_MODE
[... 30 lines cut ...]
[  1725][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 98632
[  1766][D][BluetoothA2DPSink.cpp:988] app_a2d_callback(): [BT_AV] app_a2d_callback
[  1781][D][BluetoothA2DPSink.cpp:1008] app_a2d_callback(): [BT_AV] app_a2d_callback ESP_A2D_AUDIO_CFG_EVT
[... cut ...]
[  1880][I][BluetoothA2DPSink.cpp:597] av_hdl_a2d_evt(): [BT_AV] A2DP PROF STATE: Init Compl

[  1889][I][BluetoothA2DPSink.cpp:520] app_gap_callback(): [BT_AV] event: 10
```
I don't understand why this is happening. Linux connects to the ESP32 Speaker but ESP32 microphone does not connect to ESP32 Speaker.
That in order for a2dp_source to connect to a2dp_sink, in both cases the first argument of the function must be the same? like:
```
a2dp_source.start("ESP32-BTSpeaker", get_sound_data);
a2dp_sink.start("ESP32-BTSpeaker");
```

pschatzmann, 2022-08-28:
Try to deactivate the autoreconnect before you call start. In your logs I did not see any discovery which would list the found devices...

janczeresnia, 2022-08-28:
After adding to configuration a2dp_source.set_auto_reconnect (true);
ESP32 microphone connected to linux but no sound.

Serial log:
```
ESP32 Bluetooth A2DP Analog Microphone

Starting ADC <-- Microphone
[I] AnalogAudioESP32.h : 68 - I2S_MODE_ADC_BUILT_IN
[I] AnalogAudioESP32.h : 196 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 53 - sample_rate: 44100
[I] AudioTypes.h : 54 - channels: 2
[I] AudioTypes.h : 55 - bits_per_sample: 16
[I] AnalogAudioESP32.h : 94 - input pin1: 34
[I] AnalogAudioESP32.h : 96 - input pin2: 0
[I] AnalogAudioESP32.h : 235 - RX_MODE
Starting Bluetooth A2DP
[    62][D][BluetoothA2DPSource.cpp:153] start(): [BT_API] start, 
[    68][D][BluetoothA2DPSource.cpp:171] start_raw(): [BT_API] start_raw, 
[    74][D][BluetoothA2DPCommon.cpp:149] get_last_connection(): [BT_AV] get_last_connection
[    83][D][BluetoothA2DPCommon.cpp:172] get_last_connection(): [BT_AV] => 0c:b8:15:f8:8e:ae
[   627][D][BluetoothA2DPSource.cpp:324] bt_app_task_start_up(): [BT_AV] bt_app_task_start_up
[   627][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 0
[   635][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x0
[   644][D][BluetoothA2DPSource.cpp:532] bt_av_hdl_stack_evt(): [BT_AV] bt_av_hdl_stack_evt evt 0
[   653][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[   662][W][BluetoothA2DPSource.cpp:558] bt_av_hdl_stack_evt(): [BT_AV] Reconnecting to 0c:b8:15:f8:8e:ae
[   664][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x4, param len 16
[   671][W][BluetoothA2DPCommon.cpp:41] connect_to(): [BT_AV] connect_to to 0c:b8:15:f8:8e:ae
[   682][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   690][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[   699][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   716][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   724][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   731][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   740][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   749][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   757][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[ 21166][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 21264][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x4
[ 21264][I][BluetoothA2DPSource.cpp:628] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state 4, evt 0x4
[ 21264][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 21282][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 24803][D][BluetoothA2DPSource.cpp:851] bt_app_rc_ct_cb(): [RCCT] bt_app_rc_ct_cb evt 0

```
Linux log:
```
Aug 28 19:17:42 nuc7 kernel: input: ESP32_A2DP_SRC (AVRCP) as /devices/virtual/input/input30
Aug 28 19:17:42 nuc7 elogind-daemon[7400]: Watching system buttons on /dev/input/event20 (ESP32_A2DP_SRC (AVRCP))
```

janczeresnia, 2022-08-28:
Afrer change:
```
a2dp_source.set_auto_reconnect(false);
a2dp_sink.set_auto_reconnect(true, false);
```
The ESP32 microphone has connected to the ESP32 speaker but there is no sound.

Microphone Serial log:
```
ESP32 Bluetooth A2DP Analog Microphone

Starting ADC <-- Microphone
[I] AnalogAudioESP32.h : 68 - I2S_MODE_ADC_BUILT_IN
[I] AnalogAudioESP32.h : 196 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 53 - sample_rate: 44100
[I] AudioTypes.h : 54 - channels: 2
[I] AudioTypes.h : 55 - bits_per_sample: 16
[I] AnalogAudioESP32.h : 94 - input pin1: 34
[I] AnalogAudioESP32.h : 96 - input pin2: 1074347410
[I] AnalogAudioESP32.h : 235 - RX_MODE
Starting Bluetooth A2DP
[    73][D][BluetoothA2DPSource.cpp:153] start(): [BT_API] start, 
[    73][D][BluetoothA2DPSource.cpp:171] start_raw(): [BT_API] start_raw, 
[    75][D][BluetoothA2DPCommon.cpp:149] get_last_connection(): [BT_AV] get_last_connection
[    84][D][BluetoothA2DPCommon.cpp:172] get_last_connection(): [BT_AV] => 0c:b8:15:f8:8e:ae
[    91][D][BluetoothA2DPSource.cpp:235] reset_last_connection(): [BT_API] reset_last_connection, 
[   100][D][BluetoothA2DPSource.cpp:237] reset_last_connection(): [BT_API] last connection 0c:b8:15:f8:8e:ae, 
[   110][I][BluetoothA2DPCommon.cpp:64] disconnect(): [BT_AV] disconect a2d: 0c:b8:15:f8:8e:ae
[   118][D][BluetoothA2DPSource.cpp:242] reset_last_connection(): [BT_API] resetting 0c:b8:15:f8:8e:ae, 
[   644][D][BluetoothA2DPSource.cpp:324] bt_app_task_start_up(): [BT_AV] bt_app_task_start_up
[   644][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 0
[   651][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x0
[   661][D][BluetoothA2DPSource.cpp:532] bt_av_hdl_stack_evt(): [BT_AV] bt_av_hdl_stack_evt evt 0
[   670][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[   679][I][BluetoothA2DPSource.cpp:564] bt_av_hdl_stack_evt(): [BT_AV] Starting device discovery...
[   681][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x4, param len 16
[   698][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x4
[   698][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   708][I][BluetoothA2DPSource.cpp:628] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state 1, evt 0x4
[   716][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   725][D][BluetoothA2DPSource.cpp:596] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
[   733][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   751][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   758][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   767][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   775][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
[   783][I][BluetoothA2DPSource.cpp:523] bt_app_gap_callback(): [BT_AV] event: 10
[   793][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 1
[   798][I][BluetoothA2DPSource.cpp:474] bt_app_gap_callback(): [BT_AV] Discovery started.
[  1643][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 0
[  1644][I][BluetoothA2DPSource.cpp:385] filter_inquiry_scan_result(): [BT_AV] Scanned device: 0c:b8:15:f8:8e:ae
[  1651][I][BluetoothA2DPSource.cpp:391] filter_inquiry_scan_result(): [BT_AV] --Class of Device: 0x240414
[  1660][I][BluetoothA2DPSource.cpp:395] filter_inquiry_scan_result(): [BT_AV] --RSSI: -46
[  1668][I][BluetoothA2DPSource.cpp:411] filter_inquiry_scan_result(): [BT_AV] --Compatiblity: Compatible
[  1677][I][BluetoothA2DPSource.cpp:413] filter_inquiry_scan_result(): [BT_AV] --Name: ESP32-BTSpeaker
[  1686][I][BluetoothA2DPSource.cpp:419] filter_inquiry_scan_result(): [BT_AV] --Checking match: ESP32-BTSpeaker
[  1696][I][BluetoothA2DPSource.cpp:428] filter_inquiry_scan_result(): [BT_AV] --Result: Target device found
[  1706][D][BluetoothA2DPCommon.cpp:177] set_last_connection(): [BT_AV] set_last_connection: 0c:b8:15:f8:8e:ae
[  1716][D][BluetoothA2DPCommon.cpp:181] set_last_connection(): [BT_AV] no change!
[  1723][I][BluetoothA2DPSource.cpp:432] filter_inquiry_scan_result(): [BT_AV] Cancel device discovery ...
[  1735][D][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 1
[  1741][I][BluetoothA2DPSource.cpp:465] bt_app_gap_callback(): [BT_AV] Device discovery stopped.
[  1749][I][BluetoothA2DPSource.cpp:466] bt_app_gap_callback(): [BT_AV] a2dp connecting to peer: ESP32-BTSpeaker
[  1759][W][BluetoothA2DPCommon.cpp:41] connect_to(): [BT_AV] connect_to to 0c:b8:15:f8:8e:ae
[  1768][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[  1885][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[  2611][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[  2912][D][BluetoothA2DPSource.cpp:851] bt_app_rc_ct_cb(): [RCCT] bt_app_rc_ct_cb evt 0
[... cut ...]
[ 65693][D][BluetoothA2DPSink.cpp:845] av_hdl_avrc_evt(): [BT_AV] av_hdl_avrc_evt evt 6
[ 65701][I][BluetoothA2DPSink.cpp:896] av_hdl_avrc_evt(): [BT_AV] remote rn_cap: count 0, bitmask 0x0
[ 65710][D][BluetoothA2DPSink.cpp:819] av_new_track(): [BT_AV] av_new_track
[ 68156][D][BluetoothA2DPSink.cpp:1198] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
[ 68156][I][BluetoothA2DPSink.cpp:513] app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT
[ 68161][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 77008
```

janczeresnia, 2022-08-30:
I tested the ESP32 A2DP microphone with JBL Flip bluetooth speaker and it works great.

ESP32 Analog Microphone to A2DP Bluetooth Sketch:
```
/*
 * Bluetooth A2DP Analog Microphone
 * https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-adc-a2dp
 * Microphone is connect to GPIO34
 */

#include "AudioTools.h"
#include "BluetoothA2DPSource.h"

AnalogAudioStream adc;
BluetoothA2DPSource a2dp_source;
// The data has a center of around 26427, so we we need to shift it down to bring the center to 0
ConverterScaler<int16_t> scaler(1.0, -26427, 32767);

// callback used by A2DP to provide the sound data
int32_t get_sound_data(Frame* frames, int32_t count) {
    uint8_t *data = (uint8_t*)frames;
    int framesize = 4;
   // the ADC provides data in 16 bits
    size_t result_len = adc.readBytes(data, count*framesize);   
    scaler.convert(data, result_len);
    return result_len/framesize;
}

void setup(void) {
  Serial.begin(115200);
  while (!Serial && (millis() < 1000)) delay(1000);
  Serial.println(F("\nESP32 Bluetooth A2DP Analog Microphone\n"));
  AudioLogger::instance().begin(Serial, AudioLogger::Info);
  //AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  // start adc input with default configuration
  Serial.println(F("Starting ADC <-- Microphone"));
  adc.begin(adc.defaultConfig(RX_MODE));

  // start the bluetooth
  Serial.println(F("Starting Bluetooth A2DP Sender"));
  // https://github.com/pschatzmann/ESP32-A2DP/wiki/Auto-Reconnect
  a2dp_source.set_auto_reconnect(false);
  a2dp_source.start("JBL Flip", get_sound_data);  
}

void loop() {
}
```

closed 2022-08-30 by janczeresnia