pschatzmann/arduino-audio-tools #320: Opus and OggOpus Codec example - invalid frame_sizes_ms_x2: 10 Error

janczeresnia, 2022-08-30:
Hi I'm testing the codec from your library and I ran into a error with the Opus and OggOpus codec. I changed frame_size_ms_x2 to different values ​​but the error always showed up.

Example : test-codec-opus.ino and test-codec-opusogg.ino

Error:
```
[E] CodecOpus.h : 410 - invalid frame_sizes_ms_x2: 10
```

FugaziScript, 2022-08-30:
I can also confirm, that OPUS decoding is not working. I'm trying to read data from sd card, decode using OpusAudioDecoder(). 
I only get "pops" from i2s dac. Everything seems to be working fine with the mp3 or wav.

This is audio logger info:
[I] AudioCopy.h : 66 - buffer_size=1024
[E] CodecOpus.h : 181 - opus_decode: corrupted stream
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] CodecOpus.h : 181 - opus_decode: corrupted stream
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] CodecOpus.h : 181 - opus_decode: corrupted stream
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] CodecOpus.h : 181 - opus_decode: corrupted stream
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

pschatzmann, 2022-08-30:
Hmm, I don't think that you can use opus w/o a container format.

janczeresnia, 2022-08-30:
A more detailed Debug serial log:
```
ESP32 Opus AudioTools Test

Sample rate 24000 Hz, 2 channels, 16 bit per sample
Setup 1000 Hz sine wave
[I] SoundGenerator.h : 180 - SineWaveGenerator::begin(channels=2, sample_rate=24000, frequency=1000)
[I] SoundGenerator.h : 166 - bool audio_tools::SineWaveGenerator<T>::begin() [with T = short int]
[D] SoundGenerator.h : 35 - bool audio_tools::SoundGenerator<T>::begin() [with T = short int]
[I] SoundGenerator.h : 38 - bits_per_sample: 16
[I] SoundGenerator.h : 39 - channels: 2
[I] SoundGenerator.h : 40 - sample_rate: 24000
[I] SoundGenerator.h : 207 - setFrequency: 1000
[I] SoundGenerator.h : 208 - active: true
Setup DAC
[D] AnalogAudioESP32.h : 169 - defaultConfig
[I] AnalogAudioESP32.h : 71 - I2S_MODE_DAC_BUILT_IN
[I] AnalogAudioESP32.h : 196 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 53 - sample_rate: 24000
[I] AudioTypes.h : 54 - channels: 2
[I] AudioTypes.h : 55 - bits_per_sample: 16
[I] AnalogAudioESP32.h : 91 - analog left output pin: 25
[I] AnalogAudioESP32.h : 92 - analog right output pin: 26
[I] AnalogAudioESP32.h : 264 - TX_MODE
Start decoder
[D] AudioEncoded.h : 304 - void audio_tools::EncodedAudioStream::begin(audio_tools::AudioBaseInfo)
[D] CodecOpus.h : 146 - virtual void audio_tools::OpusAudioDecoder::begin()
Start encoder
[D] AudioEncoded.h : 304 - void audio_tools::EncodedAudioStream::begin(audio_tools::AudioBaseInfo)
[E] CodecOpus.h : 410 - invalid frame_sizes_ms_x2: 10
Opus test started...
[D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
[D] AudioStreams.h : 517 - GeneratedSoundStream::readBytes: 1024
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 2 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 3 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 4 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 5 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 6 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 7 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 8 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 9 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 10 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 11 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 12 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 13 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 14 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 15 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 16 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 17 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 18 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 19 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 20 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[I] AudioCopy.h : 259 - try write - 21 
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[E] AudioCopy.h : 253 - write to target has failed!
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 0 bytes - in 22 hops
[D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
[D] AudioStreams.h : 517 - GeneratedSoundStream::readBytes: 1024

```

pschatzmann, 2022-08-31:
Just to give an update: I corrected the frame_size_ms_x2 problem and the opus test is working again.
If you want to decode ogg files, please use the CodecFLAC.

The opus-ogg is still not working and I am struggling to find the reason: The decoder just never finds any valid pages. Please not that this functionality has never been officially released and is not mentioned in the [documentation](https://github.com/pschatzmann/arduino-audio-tools/wiki/Encoding-and-Decoding-of-Audio).

janczeresnia, 2022-08-31:
I confirm the example works. I tested different bitrate, complexity, application (audio and voip) and sample rate all work. But when changing to mono stream (channels 1) I have an error:
```
Start decoder
[D] AudioEncoded.h : 304 - void audio_tools::EncodedAudioStream::begin(audio_tools::AudioBaseInfo)
[D] CodecOpus.h : 145 - virtual void audio_tools::OpusAudioDecoder::begin()
Start encoder
[D] AudioEncoded.h : 304 - void audio_tools::EncodedAudioStream::begin(audio_tools::AudioBaseInfo)
Opus test started...
[D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
[D] AudioStreams.h : 517 - GeneratedSoundStream::readBytes: 1024
[D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[D] CodecOpus.h : 303 - opus_encode - frame_size: 80

***ERROR*** A stack overflow in task loopTask has been detected.

Backtrace:0x40083721:0x3ffafbb00x40089b71:0x3ffafbd0 0x4008c75d:0x3ffafbf0 0x4008b4a7:0x3ffafc70 0x40089c80:0x3ffafca0 0x40089c30:0xe0038e48  |<-CORRUPTED

```

pschatzmann, 2022-08-31:
Some settings seem to require more stack then we have by default in Arduino. For the time being you just need to increase the available stack size. (e.g. by using a separate task).

I have the opus-ogg decoder now running on the desktop. Unfortunately in Arduino there is not enough heap!

pschatzmann, 2022-08-31:
Oh, I wasn't aware that A2DPStream also has a volume control: So calling out.setVolume(1.0); should give full volume.

pschatzmann, 2022-09-01:
The opus-ogg test is running on Arduino now as well. I am not sure however if it will be useful, because it uses quite some memory and is a kind of slow...

closed 2022-09-01 by pschatzmann