pschatzmann/arduino-audio-tools #338: need to copy audio file from URL . (  mp3 or any other file formate )

aqildad-create, 2022-09-07:
i want to copy file from URL to sd card , ( either MP3, WAVE OR RAW ) .

 i go through your example  ( * @file streams-audiokit-sd-audiokit.ino ) . in which input  source is microphone 

i want to do same , but need to replace MIROPHONE to URL  

i have esp32 dev board , 

What basic approach i should follow .

pschatzmann, 2022-09-07:
I have never tried, but I don't see any reason why this should not work:
- Just use a URLStream as source and a File as destination.
- Open the file first
- I would suggest to call copy.copyAll(); since you want to process the whole file
- Finally close the file.

aqildad-create, 2022-09-07:
i try this code , but it serial out write to target has fail , i am writing url raw to spiffs, Below is the code and serial out result 

/**
 * @file url_raw-I2S_external_dac.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/url_raw-I2S_externel_dac/README.md 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "WiFi.h"
#include "AudioTools.h"
#include <SD.h>
#include <SPI.h>

#include <SPIFFS.h>            // Built-in

const char *file_name = "/rec.raw";

URLStream music;    // Music Stream
//I2SStream i2s;    // I2S as Stream
StreamCopy copier;  //(i2s, music, 1024); // copy music to i2s
File file;          // final output stream

                    // Digital I/O used
//#define SD_CS          5
//#define SPI_MOSI      23
//#define SPI_MISO      19
//#define SPI_SCK       18

void record_start( ){
  Serial.println("Recording...");
  // open the output file
  //file = SD.open(file_name, FILE_WRITE);
  file = SPIFFS.open(file_name, FILE_WRITE);
  copier.begin(file, music);  
   
}
void record_end( ){
   
    Serial.println("Playing...");
    file.close();
     
    
   
}

// Arduino Setup
void setup(void) {
  Serial.begin(115200);

Serial.println("stART");

//    pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);
//    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
//    SPI.setFrequency(1000000);
//    SD.begin(SD_CS);
//    if (!SD.begin(true))  {
//    Serial.println("Error preparing Filing System...");
    
//                          }

  if(!SPIFFS.begin()){
        Serial.println("An Error has occurred while mounting SPIFFS");
        return;
  }
  

  
  
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // connect to WIFI
  WiFi.begin("EVO-Charji-BA88", "JAZZJAZZ");
  while (WiFi.status() != WL_CONNECTED){
    Serial.print(".");
    delay(500); 
  }
 
  // open music stream
  music.begin("https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw");
  record_start();
  
  }

// Arduino loop - repeated processing: copy input stream to output stream
int abb=0;
void loop() {
  abb=abb+1;
  Serial.println(abb);
  record_start();
  if(abb==50)
  {
    record_end();
    while(1);
    }
  
  copier.copy();  
  
  
  
  
  //int len = copier.copy();
  //if (len){
  //    Serial.print(".");
  //} else {
  //    i2s.end();
  //    Serial.println("\nCopy ended");
  //    delay(10000);
 // }
}

Recording...
[I] AudioCopy.h : 74 - buffer_size=1024
[I] AudioCopy.h : 259 - try write - 2 
[I] AudioCopy.h : 259 - try write - 3 
[I] AudioCopy.h : 259 - try write - 4 
[I] AudioCopy.h : 259 - try write - 5 
[I] AudioCopy.h : 259 - try write - 6 
[I] AudioCopy.h : 259 - try write - 7 
[I] AudioCopy.h : 259 - try write - 8 
[I] AudioCopy.h : 259 - try write - 9 
[I] AudioCopy.h : 259 - try write - 10 
[I] AudioCopy.h : 259 - try write - 11 
[I] AudioCopy.h : 259 - try write - 12 
[I] AudioCopy.h : 259 - try write - 13 
[I] AudioCopy.h : 259 - try write - 14 
[I] AudioCopy.h : 259 - try write - 15 
[I] AudioCopy.h : 259 - try write - 16 
[I] AudioCopy.h : 259 - try write - 17 
[I] AudioCopy.h : 259 - try write - 18 
[I] AudioCopy.h : 259 - try write - 19 
[I] AudioCopy.h : 259 - try write - 20 
[I] AudioCopy.h : 259 - try write - 21 
[E] AudioCopy.h : 253 - write to target has failed!
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 0 bytes - in 22 hops

aqildad-create, 2022-09-07:
what i am doing wrong ?

pschatzmann, 2022-09-07:
Just did a test with the following code and it's working perfectly:

```
#include "SD.h"
#include "AudioTools.h"

#define PIN_AUDIO_KIT_SD_CARD_CS 13
#define PIN_AUDIO_KIT_SD_CARD_MISO 2
#define PIN_AUDIO_KIT_SD_CARD_MOSI 15
#define PIN_AUDIO_KIT_SD_CARD_CLK  14

const char *ssid = "SSID";
const char *password = "password";
URLStream url(ssid, password); // Music Stream
StreamCopy copier; //(i2s, music, 1024); // copy music to i2s
File file; // final output stream

// Arduino Setup
void setup(void) {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Info);

    // define custom SPI pins
    SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);

    // intialize SD
    if(!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)){   
        LOGE("SD failed");
        return;
    }

    // open music stream
    url.begin("https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw");

    // open file
    file = SD.open("/audio-8000.raw", FILE_WRITE);
    copier.begin(file, url);
    copier.copyAll();
    file.close();

    file = SD.open("/audio-8000.raw");
    LOGI("file size: %d", file.size());
}

void loop() {
}
```

I am getting
```
13:56:36.636 -> [I] AudioCopy.h : 74 - buffer_size=1024
13:56:36.672 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.672 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.672 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.672 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.706 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.706 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.706 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.706 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.740 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.740 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.740 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.775 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.775 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.775 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.775 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.812 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.812 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:56:36.812 -> [I] AudioCopy.h : 124 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
13:56:36.952 -> [I] test.ino : 38 - file size: 12339
```

aqildad-create, 2022-09-07:
Why the file size is 12339 , and actual size of /audio-8000.raw is much higher . 

i also test it and size i am getting is 4113 , 

why is the file size difference . 

[I] URLStream.h : 248 - WiFiClientSecure
[I] URLStream.h : 248 - WiFiClientSecure
[I] HttpRequest.h : 193 - process connecting to host pschatzmann.github.io port 443
[I] HttpRequest.h : 181 - connected 1 timeout 60000
[I] HttpRequest.h : 204 - Free heap: 121508
[I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /arduino-audio-tools/resources/audio-8000.raw
[I] HttpHeader.h : 254 - HttpHeader::write
[I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 381 - -> GET /arduino-audio-tools/resources/audio-8000.raw HTTP/1.1
[I] HttpHeader.h : 191 -  -> Host: pschatzmann.github.io 
[I] HttpHeader.h : 191 -  -> Connection: close 
[I] HttpHeader.h : 191 -  -> Accept: */* 
[I] HttpHeader.h : 297 -  -> <CR LF> 
[I] HttpRequest.h : 227 - Request written ... waiting for reply
[I] HttpHeader.h : 224 - HttpHeader::read
[W] HttpHeader.h : 231 - Waiting for data...
[I] HttpHeader.h : 161 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 6212544
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Type: application/octet-stream
[I] HttpHeader.h : 161 - HttpHeader::readLine -> permissions-policy: interest-cohort=()
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Last-Modified: Wed, 07 Sep 2022 12:26:27 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 161 - HttpHeader::readLine -> ETag: "63188df3-5ecbc0"
[I] HttpHeader.h : 161 - HttpHeader::readLine -> expires: Wed, 07 Sep 2022 16:32:28 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Cache-Control: max-age=600
[I] HttpHeader.h : 161 - HttpHeader::readLine -> x-proxy-cache: MISS
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-GitHub-Request-Id: 2E18:2F37:B5010:D2F8D:6318C544
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Accept-Ranges: bytes
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Age: 562
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Date: Wed, 07 Sep 2022 16:51:59 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Via: 1.1 varnish
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Served-By: cache-qpg1251-QPG
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache: HIT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache-Hits: 0
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Timer: S1662569519.188851,VS0,VE1
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Vary: Accept-Encoding
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: 7bd33d4555a8c9648ef4e0b5d1b61a47eb2e5b9c
[I] HttpHeader.h : 161 - HttpHeader::readLine -> 
[I] URLStream.h : 93 - size: 6212544
[I] AudioCopy.h : 74 - buffer_size=1024
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] C:\Users\hp\Documents\Arduino\sketch_sep07c\sketch_sep07c.ino : 45 - file size: 4113

pschatzmann, 2022-09-07:
I am not sure: try to copy the data w/o StreamCopy
Then you get a better picture: maybe it is just not coming quick enough. 
The copy stops when available() provides 0..

pschatzmann, 2022-09-07:
I changed the copyAll implementation. The new version is not using the available() method any more:

        /// copies all data - returns true if we copied anything
        bool copyAll(int retryCount=2, int retryWaitMs=2000){

aqildad-create, 2022-09-08:
> I am not sure: try to copy the data w/o StreamCopy Then you get a better picture: maybe it is just not coming quick enough or available() does not provide the correct result. The copy stops when available() provides 0..

how can i copy data w/o streamCopy ? can you provide any related example and reference please . 

Is there any other function by which we can check the file end , and stop only when the file download complete else it retry in copying .

aqildad-create, 2022-09-08:
> I changed the copyAll implementation. The new version is not using the available() method any more:
> 
> ```
>     /// copies all data - returns true if we copied anything
>     bool copyAll(int retryCount=2, int retryWaitMs=2000){
> ```

means first i have to instal new version and then use the copyAll 

Can you please give me any sample example how to use this , i will thank ful .

pschatzmann, 2022-09-08:
No change to the provided example. You can omit all parameters because they are optional.
The retry defines how many times a copy must return 0 until we end the processing.
the wait defains the delay that is added after copy returns 0 - to potentially wait for more data.

aqildad-create, 2022-09-08:
ok i install arduino-audio-tool library again , i will run on the hard and then share the result here later ,

in the mean i want to ask that can i  use MP3 same here in this  url.begin("https://XYZ/ABC.mp3");
and then copy MP3 in SD card ? or their is some other technique use for MP3 downloading/coping ?

pschatzmann, 2022-09-08:
I normally just play it from the URL. There is usually no need to copy...

aqildad-create, 2022-09-08:
> I normally just play it from the URL. There is usually no need to copy...

Yes but many time their is internet and buffering issues , so i am first coping it in SD card and then play from SD card when ever required . 

My question is what technique should i follow to download mp3 from url , is raw data down loading and MP3 downloading will same ? Or some other area need to focus ?

pschatzmann, 2022-09-08:
Just use the copy example from above

aqildad-create, 2022-09-09:
after download latest copy f your library i test again the above code ..
my observation are 
1-file size is still very low as compare to origional size . download file size is 16890 .
2: then i insert the SD card in computer and import the downloaded file in audacity its 2 sec file and plaing no thing on audacity . while the original war file in server is playing and large size . 

now my question have you test the raw downloaded file ?? and have you compare DOWNLODED RAW FILE is same as original one in URL ? 
IF you test this then what i might doing wrong ??

below is my code 
#include "SD.h"
#include "AudioTools.h"
//#include "Audio.h"
#define PIN_AUDIO_KIT_SD_CARD_CS 5
#define PIN_AUDIO_KIT_SD_CARD_MISO 19
#define PIN_AUDIO_KIT_SD_CARD_MOSI 23
#define PIN_AUDIO_KIT_SD_CARD_CLK  18

 #define I2S_DOUT      25
#define  I2S_BCLK      27
#define  I2S_LRC       26
//jAudio audio;

const char *ssid = "EVO-Charji-BA88";
const char *password = "JAZZJAZZ";
URLStream url(ssid, password); // Music Stream
StreamCopy copier; //(i2s, music, 1024); // copy music to i2s
File file; // final output stream

// Arduino Setup
void setup(void) {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Info);

    // define custom SPI pins
    SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);

    // intialize SD
    if(!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)){   
        LOGE("SD failed");
        return;
    }

    // open music stream
    url.begin("https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw");

    // open file
    file = SD.open("/audio-8000.raw", FILE_WRITE);
    copier.begin(file, url);
    copier.copyAll();
    file.close();

    file = SD.open("/audio-8000.raw");
    LOGI("file size: %d", file.size());
     file.close();
    
   // audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
   // audio.setVolume(12); // 0...21
  //  audio.connecttoFS(SD, "audio-8000.raw");

}

void loop() {
//   audio.loop();
}

pschatzmann, 2022-09-09:
Did you try to increase the copyAll() parameter values ? 
Obviously your network behaves differently than mine. What is the minimum retryCount parameter that works reliably for you?

aqildad-create, 2022-09-09:
No , i use it at default like this copier.copyAll(); ... i will try and change it later further do you have any recommended parameter for slow network ? 

and also what is default parameter here copier.copyAll(); i-e in default what are retry counts and what is default delay here ? 

further were can i find documentation of these functions ??

pschatzmann, 2022-09-09:
Just scroll up a bit: I have sent you this information 2 days ago!
In any case I recommend to consult the class documentation...

aqildad-create, 2022-09-10:
were can i find class documentations ?

pschatzmann, 2022-09-10:
Check the Readme and Search for Documentation

aqildad-create, 2022-09-10:
> Just scroll up a bit: I have sent you this information 2 days ago! In any case I recommend to consult the class documentation... ps. I optimized the parameters today a bit, so I suggest that you try with the new values first: copyAll(5,200);

i try different value and also try  copyAll(5,200); and then increase the values upto copyAll(200,400); but none o them give me file size near to original file . 

below uart out , but file size is very low . pls guide

have to computer the original file with downloaded file ? 
[I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 110 - host->pschatzmann.github.io
[I] Url.h : 111 - protocol->https
[I] Url.h : 112 - path->/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 113 - port->443
.
[I] URLStream.h : 248 - WiFiClientSecure
[I] URLStream.h : 248 - WiFiClientSecure
[I] HttpRequest.h : 193 - process connecting to host pschatzmann.github.io port 443
[I] HttpRequest.h : 181 - connected 1 timeout 60000
[I] HttpRequest.h : 204 - Free heap: 169772
[I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /arduino-audio-tools/resources/audio-8000.raw
[I] HttpHeader.h : 254 - HttpHeader::write
[I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 381 - -> GET /arduino-audio-tools/resources/audio-8000.raw HTTP/1.1
[I] HttpHeader.h : 191 -  -> Host: pschatzmann.github.io 
[I] HttpHeader.h : 191 -  -> Connection: close 
[I] HttpHeader.h : 191 -  -> Accept: */* 
[I] HttpHeader.h : 297 -  -> <CR LF> 
[I] HttpRequest.h : 227 - Request written ... waiting for reply
[I] HttpHeader.h : 224 - HttpHeader::read
[W] HttpHeader.h : 231 - Waiting for data...
[I] HttpHeader.h : 161 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 6212544
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Type: application/octet-stream
[I] HttpHeader.h : 161 - HttpHeader::readLine -> permissions-policy: interest-cohort=()
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Last-Modified: Sat, 10 Sep 2022 03:45:37 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 161 - HttpHeader::readLine -> ETag: "631c0861-5ecbc0"
[I] HttpHeader.h : 161 - HttpHeader::readLine -> expires: Sat, 10 Sep 2022 04:14:29 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Cache-Control: max-age=600
[I] HttpHeader.h : 161 - HttpHeader::readLine -> x-proxy-cache: MISS
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-GitHub-Request-Id: 0B08:2F4C:66F14A:8C06FF:631C0CCD
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Accept-Ranges: bytes
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Age: 360
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Date: Sat, 10 Sep 2022 04:32:58 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Via: 1.1 varnish
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Served-By: cache-qpg1282-QPG
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache: HIT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache-Hits: 0
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Timer: S1662784379.651387,VS0,VE12
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Vary: Accept-Encoding
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: f45fde0657f35271d12b1dc73eb5f0e27e37173d
[I] HttpHeader.h : 161 - HttpHeader::readLine -> 
[I] URLStream.h : 93 - size: 6212544
[I] AudioCopy.h : 74 - buffer_size=1024
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 279 -> 279 -> 279 bytes - in 1 hops
[I] C:\Users\hp\Documents\Arduino\sketch_sep08a\sketch_sep08a.ino : 48 - file size: 16384

pschatzmann, 2022-09-10:
Hmm, that's strange: To me it seems that you still use the old version of the library. Can you double check and compare the implementation of copyAll() in Arduino/libraries/AudioTools/AudioCopy.h

aqildad-create, 2022-09-10:
my esp board version is 2.0.4 , and iwll update the lop with time stamp but surely it was not 80sec log lines . and i will add the URL.TIMEOUT TOO .  

This is COPYALL implementation , can you double check is it latest or not ? how to confirm which of your library version is installed , 

        size_t copyAll(int retryCount=2, int retryWaitMs=2000){
            LOGD(LOG_METHOD);
            size_t result = 0;
            int retry = 0;

            if (from==nullptr || to == nullptr) 
                return result;

            // copy while source has data available
            int count=0;
            while (true){
                count = copy();
                result += count;
                if (count==0){
                    // wait for more data
                    retry++;
                    delay(retryWaitMs);
                } else {
                    result = true; // we have some data
                    retry = 0; // after we got new data we restart the counting
                }
                // stop the processing if we passed the retry limit
                if (retry>retryCount){
                    break;
                }
            }
            return result;
        }

pschatzmann, 2022-09-10:
You can use  e.g. 

    const int limit = 500;
    heap_caps_malloc_extmem_enable(limit);

This will allocate all memory > 500 to psram. 
Just make sure that you allocate your objects with new after you executed this call.

pschatzmann, 2022-09-10:
The above should work: It just has a small bug because it does not return the correct total size. The line
   result = true; // we have some data 
should be deleted...

If you want to make sure that this code is used just add a syntax error somewhere. When you compile and you dont get the error the system uses a library from somewhere else...

aqildad-create, 2022-09-10:
I install a fresh copy of your library again and add url.setTimeout(30000); before urlbegin and copyall parameter is  copier.copyAll(200,400);. 

it look good but at the end why write to target fail ???

16:16:59.663 -> [I] AudioCopy.h : 270 - try write - 21 
16:16:59.945 -> [E] AudioCopy.h : 264 - write to target has failed

below is log , log is big so i add few starting logs and few ending .

16:07:29.216 -> [I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
16:07:29.216 -> [I] Url.h : 73 - Url::parse
16:07:29.216 -> [I] Url.h : 109 - url->https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
16:07:29.216 -> [I] Url.h : 110 - host->pschatzmann.github.io
16:07:29.216 -> [I] Url.h : 111 - protocol->https
16:07:29.216 -> [I] Url.h : 112 - path->/arduino-audio-tools/resources/audio-8000.raw
16:07:29.216 -> [I] Url.h : 113 - port->443
16:07:29.358 -> .
16:07:30.334 -> [I] URLStream.h : 248 - WiFiClientSecure
16:07:30.334 -> [I] URLStream.h : 248 - WiFiClientSecure
16:07:30.381 -> [I] HttpRequest.h : 193 - process connecting to host pschatzmann.github.io port 443
16:07:31.872 -> [I] HttpRequest.h : 181 - connected 1 timeout 60000
16:07:31.872 -> [I] HttpRequest.h : 204 - Free heap: 169836
16:07:31.872 -> [I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /arduino-audio-tools/resources/audio-8000.raw
16:07:31.872 -> [I] HttpHeader.h : 254 - HttpHeader::write
16:07:31.872 -> [I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
16:07:31.920 -> [I] HttpHeader.h : 381 - -> GET /arduino-audio-tools/resources/audio-8000.raw HTTP/1.1
16:07:31.920 -> [I] HttpHeader.h : 191 -  -> Host: pschatzmann.github.io 
16:07:31.920 -> [I] HttpHeader.h : 191 -  -> Connection: close 
16:07:31.920 -> [I] HttpHeader.h : 191 -  -> Accept: */* 
16:07:31.920 -> [I] HttpHeader.h : 297 -  -> <CR LF> 
16:07:31.920 -> [I] HttpRequest.h : 227 - Request written ... waiting for reply
16:07:31.920 -> [I] HttpHeader.h : 224 - HttpHeader::read
16:07:31.920 -> [W] HttpHeader.h : 231 - Waiting for data...
16:07:32.430 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> HTTP/1.1 200 OK
16:07:32.430 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Connection: close
16:07:32.430 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 6212544
16:07:32.430 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
16:07:32.478 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Type: application/octet-stream
16:07:32.478 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> permissions-policy: interest-cohort=()
16:07:32.478 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Last-Modified: Sat, 10 Sep 2022 09:30:07 GMT
16:07:32.478 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
16:07:32.478 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> ETag: "631c591f-5ecbc0"
16:07:32.478 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> expires: Sat, 10 Sep 2022 11:15:01 GMT
16:07:32.478 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Cache-Control: max-age=600
16:07:32.526 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> x-proxy-cache: MISS
16:07:32.526 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> X-GitHub-Request-Id: 9694:4A05:852D1B:B1B96B:631C6F5D
16:07:32.526 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Accept-Ranges: bytes
16:07:32.526 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Age: 151
16:07:32.526 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Date: Sat, 10 Sep 2022 11:07:32 GMT
16:07:32.526 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Via: 1.1 varnish
16:07:32.526 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Served-By: cache-qpg1244-QPG
16:07:32.573 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache: HIT
16:07:32.573 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache-Hits: 0
16:07:32.573 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Timer: S1662808053.824890,VS0,VE1
16:07:32.573 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> Vary: Accept-Encoding
16:07:32.573 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: 0978d3aa0fcccef894a831cf668b478f5e11f3f5
16:07:32.573 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> 
16:07:32.573 -> [I] URLStream.h : 93 - size: 6212544
16:07:32.573 -> [I] URLStream.h : 288 - Request written ... waiting for reply
16:07:34.158 -> [I] AudioCopy.h : 74 - buffer_size=1024
16:07:34.158 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
16:07:34.158 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
16:07:34.158 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
16:07:34.158 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
16:07:34.158 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
16:07:34.205 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
16:07:34.205 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
16:07:34.205 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
16:07:34.205 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[... cut ...]
16:16:58.456 -> [I] AudioCopy.h : 270 - try write - 17 
16:16:58.738 -> [I] AudioCopy.h : 270 - try write - 18 
16:16:59.060 -> [I] AudioCopy.h : 270 - try write - 19 
16:16:59.339 -> [I] AudioCopy.h : 270 - try write - 20 
16:16:59.663 -> [I] AudioCopy.h : 270 - try write - 21 
16:16:59.945 -> [E] AudioCopy.h : 264 - write to target has failed!
16:16:59.945 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 4294967295 -> 13376 bytes - in 22 hops
16:18:23.566 -> [I] C:\Users\hp\Documents\Arduino\sketch_sep08a\sketch_sep08a.ino : 49 - file size: 0

pschatzmann, 2022-09-10:
My understanding would be that your SD is full and therefore the writes are failing...

aqildad-create, 2022-09-10:
> My understanding would be that your SD is full and therefore the writes are failing...

its 2 gb SD and it is empty .

pschatzmann, 2022-09-10:
I was running into the same error. - 
Hmm maybe there is a limit on the file size.

aqildad-create, 2022-09-10:
> 

pls do some thing for file size limit ,

in the mean i will download some other small size and then update you .

but i would like to download large size files . pls consider this

pschatzmann, 2022-09-10:
The SD library is not from me.
You can try to use the SDFAT library instead...

aqildad-create, 2022-09-10:
> I was running into the same error. - Hmm maybe there is a limit on the file size. This is definitely the write to the file which is failing.

why write to file WRITE complete file at once , why it not copy small buffers 
? 

> 

yes  pls make changes to write to file with small buffers

pschatzmann, 2022-09-10:
But that's the current logic! You can define the chunk size in StreamCopy: By default it uses 1024 bytes

aqildad-create, 2022-09-10:
> But that's the current logic! You can define the chunk size in StreamCopy: By default it uses 1024 bytes By the way that's what you see in the log.

yes , if you write to file the small chunks i-e 1024 size then may be file write will not give error . . its my assumption because it only give error at the end  , may be you are writing to sd card once complete download it done .

pschatzmann, 2022-09-10:
You were right: the issue was at the end of file. I used a size_t to calculate a difference for the open size which was a bad idea and the URLStream did not return a 0 when the read failed.

The corrections have been committed

aqildad-create, 2022-09-10:
what should i do now ? you mean to say i need to download a latest version of library again and then run the code again .. now it will work properly ?

pschatzmann, 2022-09-10:
yes

aqildad-create, 2022-09-10:
is this possible to add attribute like how much the file downloaded or add downloaded percent etc ?

pschatzmann, 2022-09-10:
I leave this to you as an exercise...

aqildad-create, 2022-09-10:
> yes

i test it .. my observation is some time it work and download complete . 

but many time it give error , my opinion may be the error is due to slow internet and retry over flow , so it stop downloading .

it is very frustrating when downloading stop after 10,15 minutes  and we have no option to start and downloading from zero . 

so is there and way if internet down in the mid we can resume again when internet live again .

pschatzmann, 2022-09-10:
Wait a minute: I see some bugs here that have been corrected and the log lines look different now.
I think you still use the old version!

aqildad-create, 2022-09-10:
> Wait a minute: I see some bugs here that have been corrected and the log lines should look different now. I think you still use the old version!
> 
> Upgrade and test with some small files first!

i test it with one 2,3 minute mp3 file it work , but it stop in the mid with others files , so my observation is it may be due to slow internet , but we have to find a way for slow internet or even if internet down it should resume again later .

pschatzmann, 2022-09-10:
But that's what the parameters are for!
However if the connection was completly lost there is no way to recover!

aqildad-create, 2022-09-10:
> 
ok and why was it retry only 21 time were i set for 200 times , further  can you please verify i have latest library or not ?? because i just install a fresh copy and you were saying it is still old one so i am confuse .

pschatzmann, 2022-09-10:
The retries that you enter are not logged.
I suggest that you physically delete the directory first before reinstalling the library. Or better use git as described in the Readme

aqildad-create, 2022-09-11:
for reinstalling library  i make folder like this , the old one are just for my reference . 

[image]

pschatzmann, 2022-09-11:
I think that might be the source of your problem: Arduino still finds the other ones and you have no control which one it picks!

aqildad-create, 2022-09-11:
> I think that might be the source of your problem: Arduino still finds the other ones and you have no control which one it picks!

Can you please send here the new log , i want to see it ? then I will install the library again  to make the log format same  .

pschatzmann, 2022-09-11:
After the correction:
21:12:59.840 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 4294967295 -> 0 bytes - in 22 hops
will be logged as
21:12:59.840 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 0 -> 0 bytes - in 22 hops

After the correction:
21:12:59.794 -> [I] AudioCopy.h : 271 - try write - 16
should not appear any more but if it does you should get
21:12:59.794 -> [I] AudioCopy.h : 271 - try write - %d (open %ld bytes)

closed 2022-09-14 by pschatzmann

aqildad-create, 2022-09-15:
while stream copy we cant use loop for other task ? how to  free loop while copying 
also 
today while in testing i get this error ? why is it ? 

10:03:43.035 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:03:43.035 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:03:43.035 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:03:43.035 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:03:43.035 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:03:43.454 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:03:43.501 -> [I] AudioCopy.h : 124 - StreamCopy::copy 929 -> 929 -> 929 bytes - in 1 hops
10:03:43.501 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:03:43.501 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:03:43.501 -> [I] AudioCopy.h : 124 - StreamCopy::copy 85 -> 85 -> 85 bytes - in 1 hops
10:03:44.562 -> Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
10:03:44.562 -> 
10:03:44.562 -> Core  1 register dump:
10:03:44.673 -> PC      : 0x400dacdf  PS      : 0x00060930  A0      : 0x800db1b8  A1      : 0x3ffb2510  
10:03:44.673 -> A2      : 0x3fffff9f  A3      : 0x00000200  A4      : 0x00000200  A5      : 0x000c98ed  
10:03:44.673 -> A6      : 0x00000000  A7      : 0x3ffce748  A8      : 0x00003a2f  A9      : 0x40000000  
10:03:44.673 -> A10     : 0x00000061  A11     : 0x3f401eea  A12     : 0x000000c3  A13     : 0x00000010  
10:03:44.673 -> A14     : 0x00000040  A15     : 0x3ffce748  SAR     : 0x0000000a  EXCCAUSE: 0x00000003  
10:03:44.673 -> EXCVADDR: 0x40000000  LBEG    : 0x400845b9  LEND    : 0x400845c1  LCOUNT  : 0x00000027  
10:03:44.673 -> 
10:03:44.673 -> 
10:03:44.673 -> Backtrace:0x400dacdc:0x3ffb25100x400db1b5:0x3ffb2530 0x400db48e:0x3ffb2550 0x400db59d:0x3ffb2590 0x401078d9:0x3ffb25d0 0x40109556:0x3ffb25f0 0x4010b299:0x3ffb2630 0x400e9876:0x3ffb2660 0x4008a309:0x3ffb2680 0x4008911e:0x3ffb26a0 0x40173b1b:0x3ffb26d0 0x40173b7d:0x3ffb2710 0x400dbe2b:0x3ffb2730 0x400dbb93:0x3ffb2750 0x400d4ae8:0x3ffb2770 0x400d4c17:0x3ffb2790 0x400d632f:0x3ffb27d0 0x400df2f2:0x3ffb2820 
10:03:44.673 -> 
10:03:44.673 -> 
10:03:44.673 -> 
10:03:44.673 ->

pschatzmann, 2022-09-15:
You need to analyze the stacktrace to find the source of your problem: see https://github.com/me-no-dev/EspExceptionDecoder. If you you use the copyAll() it will process all data.
If you want to do the copy in the loop which other things work you need to execute individual copy() steps in the loop and add the closing of the file logic to the loop...