pschatzmann/arduino-audio-tools #339: Using UDPStream with AudioEffects

andysheen, 2022-09-07:
Hey,

I've been trying to set up a system that receives UDP audio, runs this through the audio effects and sends this to an I2S output. 

I've got UDP -> I2S working great, and Sine -> AudioEffects -> I2S to work, but having issues implementing AudioEffects to work with `UDPStream` and `GeneratorFromStream`. 

Here's my code:

```
include "AudioConfigLocal.h"
#include "AudioTools.h"
#include "AudioLibs/Communication.h"

//AUDIO
uint16_t sample_rate = 44100;
uint8_t channels = 1;  // The stream will have 2 channels

//ROUTING
I2SStream out;
UDPStream udp;
AudioEffects<GeneratorFromStream<int16_t>> effects(udp);
GeneratedSoundStream<int16_t> effectOut(effects);
StreamCopy copier(out, effectOut);

//WIFI
const int udpPort = 7000;
const char* ssid = "SSID";
const char* password = "PASS";

void setup() {
  Serial.begin(115200);
  // AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // connect to WIFI
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(WiFi.localIP());
  esp_wifi_set_ps(WIFI_PS_NONE);
  WiFi.setSleep(false);
  udp.begin(udpPort);

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  config.pin_data = 23;
  config.sample_rate = sample_rate;
  config.pin_data = 22;
  config.pin_bck = 23;
  config.pin_ws = 21;
  config.channels = channels;
  config.bits_per_sample = 16;
  config.buffer_size = 512;
  config.buffer_count = 6;
  out.begin(config);

  Serial.println("started...");
}

void loop() {
  copier.copy();
}
```

Any pointers would be appreciated.

Thanks!

pschatzmann, 2022-09-08:
If I understand you right, you want to replace the I2SStream output with UDPStream: Just delete the I2SStream out;
and the compiler will tell you where you messed up...

andysheen, 2022-09-08:
I'm trying to manipulate the UDPStream input with the AudioEffects and then Output this to I2SStream so:

UDPStream > AudioEffects > I2SStream

I've had each section working respectively, but having issues with UDPStream > AudioEffects. The program uploads and I'm not getting any compiler errors, i'm just not getting any sound output. I think it's getting stuck somewhere as i'm not getting any print out from `AudioLogger::instance().begin(Serial, AudioLogger::Info);` in loop.

Am I using UDPStream incorrectly for inputting into AudioEffects?

pschatzmann, 2022-09-08:
Select the log level Debug for better visibility of what is called.
Not sure what the issue is: I relied on the read functionality of WiFiUDP.

I suggest that you try to add a new method in UDPStream which has a log statement and calls the WiFiUDP::readBytes 
  size_t readBytes(uint8_t *buffer, size_t size) override {

This way you should be better see if the reading is triggered...

andysheen, 2022-09-08:
Adding readbytes to udpstream seemed to fix it! 

Thanks for the help

closed 2022-09-09 by pschatzmann