pschatzmann/arduino-audio-tools #340: PLAY RAW audio from SD card

aqildad-create, 2022-09-08:
i did not find example related to playing raw audio from SD . 

what example or technique should i need to follow in order to achieve this .

pschatzmann, 2022-09-08:
That's not different like any other Audio Source. Just use the file as copy source!

aqildad-create, 2022-09-08:
means some thing like will work for raw audio playing ? is their any better approach  ? 

{   
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(12); // 0...21
    audio.connecttoFS(SD, "audio-8000.raw");
}

void loop() {
   audio.loop();
}

pschatzmann, 2022-09-08:
I have no clue what your audio is. It is definitely not from me!

aqildad-create, 2022-09-08:
ok  @file streams-sd_mp3-i2s.ino i just need to change file name in order to play audio file right ? 

audioFile = SD.open("/Music/Conquistadores.mp3");  to  audioFile = SD.open("/audio-8000.raw"); 

what to do with codec 
  // begin copy
  copier.begin(decoder, audioFile);

no need to have AudioCodecs if play raw data ?

pschatzmann, 2022-09-08:
Codecs translate encoded files to raw (=PCM) format: so there is no need ...

mageeagle, 2022-09-11:
I tried recording some audio files by writing raw files onto an SD card based on the example here:
https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-sd-audiokit/streams-audiokit-sd-audiokit.ino

which I removed all the audiokit stuff away and record through internal ADC, and playing it back through an external DAC using
`
file = SD.open("record.raw");
copier.begin(i2s, file);
`
I opened the raw file on audacity on my computer and it plays fine.
While using copier produces a bunch of these errors occur:

E (327888) I2S: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1214 (i2s_write):tx NULL
[E] I2SESP32.h : 237 - static size_t audio_tools::I2SBase::writeExpandChannel(i2s_port_t, int, const void*, size_t)

Wonder if there are leads?

pschatzmann, 2022-09-11:
Did I get you right that you record via the AnalogAudioStream and play back via an I2SStream ?
If so I suggest that you try to use port_no 1 for the I2SStream...

mageeagle, 2022-09-11:
Yes, I was using the AnalogAudioStream and I wasn't aware that it would clash on the I2SStream. Changing the port number solves the problem, thanks.

closed 2022-09-12 by pschatzmann