pschatzmann/arduino-audio-tools #346: AnalogRead not working after AnalogAudioStream begins

mageeagle, 2022-09-11:
I am trying to get values from a mic while recording with it.
I start the audio stream in setup, while putting analogRead in loop
when the audio stream begins, analogRead will not work and cause the loop to stop
Wonder if there is a workaround on this? Such as if it's possible to get the value of the audio stream like analogRead?

pschatzmann, 2022-09-11:
You would copy the data from the AnalogAudioStream to a MultiOutput. 
https://github.com/pschatzmann/arduino-audio-tools/wiki/Splitting-and-Merging-Audio

mageeagle, 2022-09-11:
I would like to just read the audio data from the stream, since I'm working with a distance sensor which has delays that does not work when the copier is running with .copy(),  but I'm not sure what (buffer and size) would fit into the function?

pschatzmann, 2022-09-11:
https://github.com/pschatzmann/arduino-audio-tools/wiki/Introduction
If you use bits_per_sample=16 you can just cast the buffer to int16_t* to read the values. If you want to process a single sample it would be readBytes(buffer,2);

mageeagle, 2022-09-11:
I'll look into the readBytes

But on the topic, for instance, I wired the mic output to another pin for analogRead, but simply calling analogRead stops the code from running, even if I add an nth-time thing in the loop:

int incre = 0;
void loop() {
  copier.copy();
  incre++;
  if (incre > 100) {
     Serial.println(analogRead(PIN_NUMBER));
     incre = 0;
  }
}

I think digitalWrite also don't work in this case as well: for the distance sensors, I thought the delay here is actually slower than the sample rate
  // Clear the trigPin by setting it LOW:
  digitalWrite(trigPin, LOW);
  delayMicroseconds(5);
  // Trigger the sensor by setting the trigPin high for 10 microseconds:
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  // Read the echoPin, pulseIn() returns the duration (length of the pulse) in microseconds:
  duration = pulseIn(echoPin, HIGH);
  // Calculate the distance:
  distance = duration * 0.343 / 2;

But the distance sensor is a bit off topic, but just to show some stuff doesn't work while the copier is doing its thing

Sorry I'm kind of a beginner on this! Thanks for your patience.

pschatzmann, 2022-09-11:
An analogRead() should be fast and should not disturb.
The print however is very bad!

mageeagle, 2022-09-11:
I made a simple mistake by calling analogRead with a small delay in loop, hence it's blowing up
I didn't know the loop() is running at such a high speed, it was fixed by pushing the increment to a much higher number!

But indeed, when a pin is in use by AnalogAudioStream, the pin will not be available for analogRead and the whole code will not run for some reason.

mageeagle, 2022-09-12:
Another observation is, after AnalogAudioStream started, if there are analogReads afterwards, no matter which pin it is from, it would stop the analog stream... copying from the stream results in a blank sound file...

pschatzmann, 2022-09-12:
https://github.com/espressif/arduino-esp32/issues/4782

mageeagle, 2022-09-12:
Thanks, I didn't notice that my ESP32 library is very outdated. I don't have the means to test the code again, but I assume that it should be fine once I update my library, I'll reopen this issue if needed.

closed 2022-09-12 by mageeagle