pschatzmann/arduino-audio-tools #354: Audio control on audiokit only functioning for speaker outs not earphone out

iannecj, 2022-09-16:
Mr S!

Love your work!  Thank you!

One question, I am wondering if I am  using something incorrectly or there is functionality not yet completed.

I am using aithinker audiokit board.

All the example sketches work for me in audiotools and in audiokit github repositories.

I am noticing a volume control issue with the earphone output.  I have 2 speaker configurations, one using a small set of speakers plugged directly into on board speaker jacks and the other I am running into a set of amplified speakers from earphone output jack .

In the example, streams-audiokit-sd-audiokit

when I connect speakers to onboard ports, the line kit.setVolume(0.9); controls volume from mute to fairly high volume by 0 to 1 setting in argument.

However, when I connect amplified speakers to the earphone out 3.5 mm jack, this setting doesn't seem to control the earphone out volume and it remains very low.

I would like to be able to get volume control on 3.5mm earphone out jack in a similar manner to the control I have of the onboard speaker jacks.

Do you have any suggestions?

Thanks,

Chris

pschatzmann, 2022-09-16:
I never use the earphones: The simple solution would be to use the [VolumeControl which is available in the Framework](https://github.com/pschatzmann/arduino-audio-tools/wiki/Volume-Control). This will impact both the speakers and the jack.

If you look for a challenging solution you might dig into the [ES8388 Datasheet](https://datasheet.lcsc.com/lcsc/1912111437_Everest-semi-Everest-Semiconductor-ES8388_C365736.pdf)...

iannecj, 2022-09-16:
Thank you Mr. S!  Since I am a software novice, looking at the volumestream
example you linked I am struggling to apply it to the two example code
below one of which which is a modified version  of one of your other
examples.

Could you help?  My issue is would it apply to "kit" and where in the flow.

Thank you very much!!!

-C

/**
 * @file player-sd-audiokit.ino
 * @brief see
https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/player-sdfat-audiokit/README.md
 * Make sure that the pins are set to off, on, on, off, off
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "AudioLibs/AudioSourceSDFAT.h" // or AudioSourceIdxSDFAT.h
#include "AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/";
const char* ext="mp3";
SdSpiConfig sdcfg(PIN_AUDIO_KIT_SD_CARD_CS, DEDICATED_SPI, SD_SCK_MHZ(10) ,
&AUDIOKIT_SD_SPI);
AudioSourceSDFAT source(startFilePath, ext, sdcfg);
AudioKitStream kit;
MP3DecoderHelix decoder;  // or change to MP3DecoderMAD
AudioPlayer player(source, kit, decoder);

void next(bool, int, void*) {
   player.next();
}

void previous(bool, int, void*) {
   player.previous();
}
void printMetaData(MetaDataType type, const char* str, int len){
  Serial.print("==> ");
  Serial.print(toStr(type));
  Serial.print(": ");
  Serial.println(str);
}
void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
    cfg.channels = 1;
  cfg.sample_rate = 24000;

  kit.begin(cfg);

 // setup additional buttons
  kit.addAction(PIN_KEY4, next);
  kit.addAction(PIN_KEY3, previous);
kit.setVolume(90);

  // setup player
  player.setVolume(0.9);
 source.setFileFilter("*iron*");
  player.setMetadataCallback(printMetaData);
  player.begin();

  // select file with setPath() or setIndex()
  //player.setPath("/ZZ Top/Unknown Album/Lowrider.mp3");
  //player.setIndex(1); // 2nd file

}

void loop() {
  player.copy();
  kit.processActions();
}

OR

/**
 * @file player-sd-audiokit.ino
 * @brief see
https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/player-sdfat-audiokit/README.md
 * Make sure that the pins are set to off, on, on, off, off
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "AudioLibs/AudioSourceSDFAT.h" // or AudioSourceIdxSDFAT.h
#include "AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/";
const char* ext="mp3";
SdSpiConfig sdcfg(PIN_AUDIO_KIT_SD_CARD_CS, DEDICATED_SPI, SD_SCK_MHZ(10) ,
&AUDIOKIT_SD_SPI);
AudioSourceSDFAT source(startFilePath, ext, sdcfg);
AudioKitStream kit;
MP3DecoderHelix decoder;  // or change to MP3DecoderMAD
AudioPlayer player(source, kit, decoder);

void next(bool, int, void*) {
   player.next();
}

void previous(bool, int, void*) {
   player.previous();
}

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  kit.begin(cfg);

 // setup additional buttons
  kit.addAction(PIN_KEY4, next);
  kit.addAction(PIN_KEY3, previous);

kit.setVolume(0.7);
  // setup player
  player.setVolume(0.7);
  player.begin();

  // select file with setPath() or setIndex()
  //player.setPath("/ZZ Top/Unknown Album/Lowrider.mp3");
  //player.setIndex(1); // 2nd file

}

void loop() {
  player.copy();
  kit.processActions();
}

OR

iannecj, 2022-09-16:
> Oh, there is not need if you use the player!
> It has the volume control already included: Just call setVolume() on the
> player!

As you can see in my example I try both kit and player volume control.  The
adjust speaker outputs but not earphone out volume.

pschatzmann, 2022-09-16:
I guarantee that if you set   player.setVolume(0.0); you won't have any audio on the earphones!

iannecj, 2022-09-16:
Hi Mr. S, 

Thanks again for staying with me!

I tested more and in fact **as you said** player volume control is modulating volume on earphone out.
I still have a problem though.  

The reason I didn't realize this is I had (2) 3W speakers plugged direct into board and at the same time had earphone out plugged into a 2x20 lepy amplifier (LP-2020A) into (2) 50W speakers with volume on knob on amplifier turned fairly high.

When player volume is adjusted to .9 the 3W speakers when up and I expected the 2x20W amplifer to be blaring but wasnt.  In fact the 3W speakers dwarfed the sound from the 2x20W amplifier so I couldn't tell that in reality the big speakers and amp are moduling volume its just so low it could not be heard.  

If I disconnect the 3W speakers and leave only the earphone out to the amp with amp volume at full....I can hear an output on 20W amp at full knob turn but its light/.  Much less volume than the 3W even though Amp is at full.

Seems like while earphone out is modulated its volume overall is very very low.  So low the big amp cant get the signal louder than what the 3W speakers can get to with on board DACs

  I am not sure how to set registers is ES8388 SoC but I think I see some registers I might want to set.

Earphone out not really usable at these levels.

Knowing player is modulating volume but a very low volume do you have any other ideas?

Thanks again!

pschatzmann, 2022-09-16:
Did you try to set both the player and kit to max volume (1.0)  if you get an acceptable volume out of your amplifier ?

I tend to think that this is rather an issue of your amplifier. I can double check this tomorrow with my amplifier but I am pretty sure that there will not be any problem if I use Aux In. You can double check this on your side with some earphones which I am certain will be very loud.

Unfortunately the registers will not help, because you can only attenuate!

iannecj, 2022-09-16:
I just plugged in passive earphones...so faint you cant hardly hear.  While if I plug in 3W speakers very loud.

earphone jack is very low best I can tell, even with power 2x20W amp out of test.

iannecj, 2022-09-17:
> Strange - Did you try the AI_THINKER_ES8388_VOLUME_HACK in the
> AudioKitSettings.h
> ?

Yes, it worked!!
Thank you very much!

closed 2022-09-17 by pschatzmann