pschatzmann/arduino-audio-tools #358: Afrer update git to current version. Player has stopped compiling.

janczeresnia, 2022-09-17:
Hi I just updated your library to today's version and my MP3 player stopped compiling. Shows the error:
```
ESP32-MP3-Radio-Player2-I2S-SSD1306:39:13: error: cannot declare variable 'player' to be of abstract type 'audio_tools::AudioPlayer'
 AudioPlayer player(source, i2s, decoder);
             ^~~~~~
In file included from /root/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:31,
                 from /root/Arduino/szkice/ESP32/ESP32-MP3-Radio-Player2-I2S-SSD1306/ESP32-MP3-Radio-Player2-I2S-SSD1306.ino:9:
/root/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioPlayer.h:32:11: note:   because the following virtual functions are pure within 'audio_tools::AudioPlayer':
     class AudioPlayer : public AudioBaseInfoDependent {
           ^~~~~~~~~~~
In file included from /root/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimerDef.h:2,
                 from /root/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimerESP32.h:4,
                 from /root/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimer.h:3,
                 from /root/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:11,
                 from /root/Arduino/szkice/ESP32/ESP32-MP3-Radio-Player2-I2S-SSD1306/ESP32-MP3-Radio-Player2-I2S-SSD1306.ino:9:
/root/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioTypes.h:66:29: note: 	'virtual audio_tools::AudioBaseInfo audio_tools::AudioBaseInfoDependent::audioInfo()'
       virtual AudioBaseInfo audioInfo() = 0;
                             ^~~~~~~~~
exit status 1
cannot declare variable 'player' to be of abstract type 'audio_tools::AudioPlayer'
```
Sketch:
```
/**
 * ESP32 MP3 internet Radio Player2 with metadata Icecast
 * Output on external DAC MAX98357 GPIO 13,14,15
 * Button GPIO 36
 * OLED Display SSD1306 I2C
 * https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-url_icy-i2s
 */

#include "AudioTools.h"         // https://github.com/pschatzmann/arduino-audio-tools
#include "AudioCodecs/CodecMP3Helix.h"
#include "Adafruit_SSD1306.h"   // https://github.com/adafruit/Adafruit_SSD1306

const char *urls[] = {
  "http://stream4.nadaje.com:12854/test",             // Radio AK Jutrzenka
  "http://fsspx.pl:8000/radio.mp3",                   // Fsspx Radio
  "https://ice1.somafm.com/cliqhop-128-mp3",          // Soma
  "http://icecast.westerfunk.net:8004/radiofunk.mp3", // radioFunk
  "http://uk5.internet-radio.com:8185/stream",        // Polymer Radio
  "https://stream.laut.fm/elektronisch_querbeat",     // Elektronisch Querbeat
  "http://s1.a1radio.nl:8054/stream"                  // The SID Station - Commodore 64 SID music 24/7
};

// I2S DAC MAX98357
#define DIN_PIN 13    // DATA | SD_OUT
#define BCLK_PIN 14   // BCK  | SCK
#define LRCLK_PIN 15  // WS   | FS
// Swithes
#define PTT_PIN 36
// SSD1306
#define SSD1306_NO_SPLASH       // Disable Adafruit splash screen
#define SCREEN_WIDTH 128        // OLED display width, in pixels
#define SCREEN_HEIGHT 64        // OLED display height, in pixels
#define SCREEN_ADDRESS 0x3C

ICYStream url("ssid","password");
AudioSourceURL source(url, urls, "audio/mp3");
I2SStream i2s;
MP3DecoderHelix decoder;
AudioPlayer player(source, i2s, decoder);
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, false);
Debouncer ButtonDebouncer(2000);

float vol = 0.6;
String msg[2];

void clearDisplay(void) {
  display.setCursor(0,0); // Start at top-left corner
  display.clearDisplay(); // Clear the display buffer
}

// function for meta data display on oled
void displayMetadata(void) {
  clearDisplay();
  display.println(msg[0]);
  display.display();
  display.println(msg[1]);
  display.display();
}

void metadataUnknown(void) {
  msg[0] = String("Name: Unknown");
  msg[1] = String("Title: Unknown");
}

// callback for meta data display
void printMetaData(MetaDataType type, const char* str, int len) {
  String t = toStr(type);
  String s = str;
  Serial.print("==> ");
  Serial.print(t);
  Serial.print(": ");
  Serial.println(s);
  if (t == String("Name")) {
    msg[0] = String("Name: " + s);
  }
  if (t == String("Title")) {
    msg[1] = String("Title: " + s);
  }
  displayMetadata();
}

void updateSwitch() {
  // Moves to the next url when we touch the pin
  if (digitalRead(PTT_PIN) == LOW) {
    if (ButtonDebouncer.debounce()) {
      Serial.println(F("Moving to next url"));
      metadataUnknown();
      player.next();
    }
  }
}

void setup() {
  String s;
  Serial.begin(115200);
  while (!Serial && (millis() < 1000)) delay(1000);
  Serial.println(F("\nESP32 MP3 Radio Player2 I2S SSD1306\n"));
  //AudioLogger::instance().begin(Serial, AudioLogger::Info);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial, AudioLogger::Debug); 

  // SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
  if(!display.begin(SSD1306_SWITCHCAPVCC,SCREEN_ADDRESS)) {
    while(true) {
      Serial.println(F("SSD1306 allocation failed"));
      delay(1000);
    }
  }
  // setup SSD1306 display
  clearDisplay();
  display.setTextSize(1);      // Normal 1:1 pixel scale
  display.setTextColor(SSD1306_WHITE); // Draw white text
  display.cp437(true);         // Use full 256 char 'Code Page 437' font
  // setup switch
  s = F("Setup Controllers");
  Serial.println(s);
  display.println(s);
  display.display();
  pinMode(PTT_PIN, INPUT_PULLUP);
  // setup output digital i2s dac
  s = F("Starting I2S DAC");
  Serial.println(s);
  display.println(s);
  display.display();
  auto config = i2s.defaultConfig(TX_MODE);
  config.pin_bck = BCLK_PIN;  //  BCLK
  config.pin_ws = LRCLK_PIN;  //  LRCLK
  config.pin_data = DIN_PIN;  //  DIN
  config.is_master = true;
  config.port_no = 0;
  i2s.begin(config);
  // setup player
  s = F("Starting Player");
  Serial.println(s);
  display.println(s);
  display.display();
  player.setMetadataCallback(printMetaData);
  player.begin();
  player.setVolume(vol);
}

void loop() {
  updateSwitch();
  if(player.isActive()) {
    player.copy();
  } else {
    vTaskDelay(100);
  }
}
```

pschatzmann, 2022-09-17:
Thanks for letting me know. I have just added the missing method...

janczeresnia, 2022-09-17:
Everything is already going on fine, thank you.

Regards.

closed 2022-09-17 by janczeresnia

pschatzmann, 2022-09-18:
I just finished the compile of all examples: I can confirm that, this was the only thing that I missed!