pschatzmann/arduino-audio-tools #368: Enhancement: Make ESP32 Audio Kit Louder

Yadefaulty, 2022-09-20:
So i been searching through github and I found this: 

[https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/issues/26](url)

here's the code that makes the audiokit louder:
`esp_err_t es8388_set_voice_volume(int volume) 
{ 
    esp_err_t res = ESP_OK; 
    if (volume < 0) volume = 0; else if (volume > 100) volume = 100; 
    volume /= 3; 
    res = es_write_reg(ES8388_ADDR, ES8388_DACCONTROL4, 0); // LDACVOL  0..-96db  in 0.5steps (0=loud, 192=silent)
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL5, 0); // RDACVOL 0..-96db  in 0.5steps (0=loud, 192=silent)
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL24, volume);  // LOUT1 volume 0..33 dB
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL25, volume);  // ROUT1 volume 0..33 dB
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL26, volume);  // LOUT2 volume 0..33 dB
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL27, volume);  // ROUT2 volume 0..33 dB
    return res; 
}`

pschatzmann, 2022-09-20:
Isn't this the same like the currently implemented
#define AI_THINKER_ES8388_VOLUME_HACK 1

Yadefaulty, 2022-09-20:
> 

yea basically but makes it a little louder

pschatzmann, 2022-09-20:
I am not sure if I want to act to correct a bug on some boards that most peaple don't have and that I can't reproduce.
Feel free to submit a PR

Yadefaulty, 2022-09-20:
ok got u

closed 2022-09-20 by Yadefaulty