pschatzmann/arduino-audio-tools #372: Not working with Espressif 32 library version 5.1.1

CliveDent, 2022-09-22:
Hello.

Firstly, thank you very much for the hard work on this library - I am very impressed.

I have found a strange issue which basically broke everything. In the end, I have found there are two things that are not compatible:

AudioCopy.h
`// If we try to write to a server we might not have any output destination yet`
` int to_write = to->availableForWrite();`
` if (to_write<=0){`
`     delay(500);`
`     return 0;`
` }`

and

Print.h
`// add availableForWrite to make compatible with Arduino Print.h`
`// default to zero, meaning "a single write may block"`
`// should be overriden by subclasses with buffering`
`virtual int availableForWrite() { return 0; }`

I don't know what version of Espressif 32 library should I use so I used the latest version (5.1.1). Older versions gave me compile error saying that `availableForWrite()` function is not defined. The 5.1.1 version seams to make all streams unavailable for writing if 0 is not accepted. Even the example for writing generated sound into serial didn't work.

When I changed the condition to  `if (to_write<0)` everything started working but I'm not sure if this is the correct way.

pschatzmann, 2022-09-22:
Did you upgrade to the latest version. I think this problem has already been corrected...

CliveDent, 2022-09-22:
Just upgraded the arduino-audio-tools library and it works. Thank you!

closed 2022-09-22 by CliveDent