pschatzmann/arduino-audio-tools #377: [ESP32] Streaming from server to client doesn't work because of missing availableForWrite

vlastahajek, 2022-09-26:
Thanks for the wonderful library!

I've started two weeks ago with this library on ESP32 with streaming demos using INMP441 mic.
`streams-i2s-serial` with changes for using INMP441 didn't work at the beginning. Problem was that CvsStream missed override for `int availableForWrite()` used in used in the copier and the inherited function from `Print` returns 0. 
When I was about to create PR, I noticed you fixed that in the commit [8e6a8914](https://github.com/vlastahajek/arduino-audio-tools/commit/8e6a8914cda704bbd292cd08032e5824f2805814#diff-c1a75caeb040f9f4b34b238ea707d7e7c095ed25bf1e747829a02e5f06fd58f2). 

Great work with often updates! 

However, the `streams-i2s-webserver_wav` demo doesn't work on ESP32 for a similar reason: WiFiClient doesn't override `availableForWrite`.

pschatzmann, 2022-09-26:
Thanks for letting me know: I could reproduce the issue and have committed a correction...

vlastahajek, 2022-09-26:
@pschatzmann, Thanks for the quick fix. Awesome work!
I can confirm it works without a workaround.

vlastahajek, 2022-09-26:
Fixed by [d2cfd6](https://github.com/pschatzmann/arduino-audio-tools/commit/d2cfd6fbbca4dce3b88779332d9c28846983a27a)

closed 2022-09-26 by vlastahajek