pschatzmann/arduino-audio-tools #379: Speed up reading speed

luke2023, 2022-09-27:
I made a test before, and I found out that the mp3 speed becomes slower when the cpu is 80mhz, is there any other way to speed up the system?Thanks

https://youtu.be/fl0i-7D50IU

pschatzmann, 2022-09-27:
I didn't even know that you can go that low! I am always using my ESP32 at max speed.
You could use wav files instead of mp3 or do the docoding with a separate module: https://github.com/pschatzmann/arduino-vs1053

luke2023, 2022-09-27:
> I didn't even know that you can go that low! I am always using my ESP32 at max speed.
> 
> You could use wav files instead of mp3 or do the MP3 docoding with a separate [module](https://github.com/pschatzmann/arduino-vs1053)

Thank you,what example do you suggest if I want to transform a mp3 to wav and place it in my sd card?

pschatzmann, 2022-09-27:
Just read the Wiki about Codecs...

luke2023, 2022-09-27:
@pschatzmann I found AudioSourceSDFAT ,wavEncoder , MP3DecoderHelix and audioWriter,but I don't find any functions in audioWriter to transform data,could you please give me some hints?Thank you for your help!

pschatzmann, 2022-09-27:
You would do this on the PC e.g. with Audacity

luke2023, 2022-09-27:
@pschatzmann I want to let my students save mp3 file in sd because this is more common but,anyway, Thank you for helping!
Lin Che

luke2023, 2022-09-29:
@pschatzmann hello, I think I finally found the code for decode and encoding,:
https://github.com/pschatzmann/arduino-audio-tools/wiki/Encoding-and-Decoding-of-Audio#encoding

I'll modify the in put to sd card and try the codes !I'll probably make a video for that.

Thank you!