pschatzmann/arduino-audio-tools #383: Timeout when reading WAV from SD card

Benforce, 2022-09-30:
Hi, using your player.  Appreciate your work! Reading mp3  from SD and playing to BT speaker works. To save processing power I switched to WAV files. I added some Wav files on the SD card and using WAVdecoder. Somehow it doesn't work. I get timeouts on reading the WAV from SD card. Tried different WAV files. 

12:42:57.551 -> [W] AudioA2DP.h : 307 - ==> state: Connecting
12:42:57.900 -> [W] AudioA2DP.h : 307 - ==> state: Connected
12:43:00.380 -> [W] AudioPlayer.h : 313 - -> timeout - moving to next stream
12:43:01.996 -> [W] AudioPlayer.h : 313 - -> timeout - moving to next stream
12:43:03.641 -> [W] AudioPlayer.h : 313 - -> timeout - moving to next stream
12:43:05.261 -> [W] AudioPlayer.h : 313 - -> timeout - moving to next stream
12:43:06.906 -> [W] AudioPlayer.h : 313 - -> timeout - moving to next stream

Any ideas?  Thanks!

pschatzmann, 2022-09-30:
Did you try with the latest version from github?
Looks like a problem to me that has already been solved...

Benforce, 2022-09-30:
Yes, latest versions.

pschatzmann, 2022-09-30:
Hmm, I would expect to see these messages when the files can not be read..

Benforce, 2022-09-30:
A bit more readable with loglevel to info:

load:0x40080400,len:5856
entry 0x400806a8
[I] AudioA2DP.h : 135 - Connecting to EE4922
[I] AudioA2DP.h : 142 - Starting a2dp_source...
[W] AudioA2DP.h : 307 - ==> state: Connecting
[W] AudioA2DP.h : 307 - ==> state: Connected
[I] AudioA2DP.h : 152 - a2dp_source is connected...
[I] AudioPlayer.h : 276 - setVolume(0.700000)
[I] AudioStreams.h : 1103 - setVolume: 0.700000
[I] AudioStreams.h : 1103 - setVolume: 0.700000
[I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSourceSDFAT.h : 104 - selectStream: 0
[I] AudioSourceSDFAT.h : 121 - -> selectStream: /V.wav
[I] AudioCopy.h : 66 - buffer_size=1024
[I] CodecWAV.h : 138 - WAVHeader::write: 1024 -> 44 -> 44
[I] CodecWAV.h : 49 - WAVHeader::begin: 44
[I] CodecWAV.h : 218 - WAVHeader sound_pos: 0
[I] CodecWAV.h : 219 - WAVHeader channels: 2 
[I] CodecWAV.h : 220 - WAVHeader bits_per_sample: 16
[I] CodecWAV.h : 221 - WAVHeader sample_rate: 44100 
[I] CodecWAV.h : 222 - WAVHeader format: 1
[I] CodecWAV.h : 323 - WAV sample_rate: 44100
[I] CodecWAV.h : 324 - WAV data_length: 0
[I] CodecWAV.h : 325 - WAV is_streamed: 0
[I] CodecWAV.h : 326 - WAV is_valid: true
[I] CodecWAV.h : 344 - isValid: true
[I] AudioPlayer.h : 190 - sample_rate: 44100
[I] AudioPlayer.h : 191 - bits_per_sample: 16
[I] AudioPlayer.h : 192 - channels: 2
[I] AudioStreams.h : 1103 - setVolume: 0.700000
[I] AudioStreams.h : 1103 - setVolume: 0.700000
[I] AudioTypes.h : 53 - sample_rate: 44100
[I] AudioTypes.h : 54 - channels: 2
[I] AudioTypes.h : 55 - bits_per_sample: 16
[I] CodecWAV.h : 347 - WAVDecoder writing first sound data
[I] AudioCopy.h : 295 - try write - 2 (open 0 bytes) 
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 2 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[... 22 lines cut ...]
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 135 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
[W] AudioPlayer.h : 313 - -> timeout - moving to next stream
[I] AudioSourceSDFAT.h : 99 - nextStream: 1
[I] AudioSourceSDFAT.h : 104 - selectStream: 1
[I] AudioSourceSDFAT.h : 121 - -> selectStream: /1.wav
[I] AudioEncoded.h : 319 - void audio_tools::EncodedAudioStream::end()
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 66 - buffer_size=1024
[W] AudioPlayer.h : 313 - -> timeout - moving to next stream
[I] AudioSourceSDFAT.h : 99 - nextStream: 1
[I] AudioSourceSDFAT.h : 104 - selectStream: 2
[I] AudioSourceSDFAT.h : 121 - -> selectStream: /hundred.wav
[I] AudioEncoded.h : 319 - void audio_tools::EncodedAudioStream::end()
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 66 - buffer_size=1024
[W] AudioPlayer.h : 313 - -> timeout - moving to next stream
[I] AudioSourceSDFAT.h : 99 - nextStream: 1
[I] AudioSourceSDFAT.h : 104 - selectStream: 3
[E] AudioSourceSDFAT.h : 112 - Filename is null
[I] AudioEncoded.h : 319 - void audio_tools::EncodedAudioStream::end()
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()

Benforce, 2022-09-30:
From this it's still not clear to me what causes the timeout. Apparently not the SD read speed so increasing the SPi frequency will not help with this problem.

pschatzmann, 2022-09-30:
So, the first file is working ? It just gets into trouble at the second file...
What happens if you send it to another output stream (e.g. I2SStream) instead ?

Benforce, 2022-09-30:
No, I don't hear any music when trying to play wav's. When debuglevel is set to 'warning' I only see the timeout messages:

12:42:57.551 -> [W] AudioA2DP.h : 307 - ==> state: Connecting
12:42:57.900 -> [W] AudioA2DP.h : 307 - ==> state: Connected
12:43:00.380 -> [W] AudioPlayer.h : 313 - -> timeout - moving to next stream
12:43:01.996 -> [W] AudioPlayer.h : 313 - -> timeout - moving to next stream
12:43:03.641 -> [W] AudioPlayer.h : 313 - -> timeout - moving to next stream

only when debug level set lower I can see that apparently something is done with streamcopy otherwise I would not have known. But I do not hear anything. I wonder if the whole first wav is streamcopied. I will see what happens if the first WAV file is longer, if the whole streamcopy process takes longer.

Hmm, I2Sstream. Not familiar with it with your player. So you mean I stream to an i2s output and play through a i2s decoder like max98357? Will look into how this works and let you know.

luke2023, 2022-10-01:
@Benforce hi, did you make sure you can play the file from computer ?

pschatzmann, 2022-10-27:
issue can not be reproduced...

closed 2022-10-27 by pschatzmann