pschatzmann/arduino-audio-tools #384: Example sketch streams-url_aac-audiokit.ino not working

timtamneil, 2022-10-01:
I am using ESP32 Audio Kit v2.2 2957 board purchased from Aliexpress, and cannot get the sketch streams-url_aac-audiokit to work.
This board has the AC101 codec, and I have changed the board to type 6 in the file AudioKitSettings.h

The only change to the sketch is for the ssid and password to suit my wifi network.
I compile the sketch with ESP32 Dev Module and with all other settings set to their defaults.
There are three warnings:
D:\software\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h:129:34: warning: narrowing conversion of 'AudioKitConfig::sampleRate()' from 'uint32_t {aka unsigned int}' to 'int' inside { } [-Wnarrowing]
         .sample_rate = sampleRate(),
                                  ^
C:\Users\neil\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:\Users\neil\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:\Users\neil\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;

When the sketch runs, the serial monitor shows:
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[W] AudioKit.h : 51 - sd_active = true
[W] AudioKit.h : 602 - Mode Button ignored because of conflict: 13 
[W] AudioKit.h : 610 - Headphone detection ignored because of conflict: 5 
.....
[W] HttpHeader.h : 231 - Waiting for data...
[W] HttpRequest.h : 153 - no CONTENT_LENGTH found in reply

Changing the URL to http://peacefulpiano.stream.publicradio.org/peacefulpiano.aac causes the ESP32 to continually reboot.

As a comparison, the sketch streams-url_mp3-audiokit compiles with only one warning:
D:\software\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h:129:34: warning: narrowing conversion of 'AudioKitConfig::sampleRate()' from 'uint32_t {aka unsigned int}' to 'int' inside { } [-Wnarrowing]
         .sample_rate = sampleRate(),
With the only changes being the ssid and password, the sketch runs correctly with the default URL.

I would like to make a simple internet radio using the ESP32 Audio Kit, which can play both mp3 and AAC streams.
Any suggestion as to why the streams-url_aac-audiokit.ino example sketch does not work would be appreciated

pschatzmann, 2023-01-22:
I guess you forgot to update the ESP32 in the Board Manager. The current version is 2.0.6

pschatzmann, 2023-02-13:
Just retested with the indicate URL and I could not reproduce your issue.

closed 2023-02-13 by pschatzmann