pschatzmann/arduino-audio-tools #385: Noice canceling

podaen, 2022-10-01:
I have this crazy idea  to cancel the noice that is recorded through a mic. Invert the sinewave and send it to the spreaker. I am seriously interested to try this. If it works we have a win, if it doesn't at leasted we tried.

The dogs of the neighbors are barking me out of bed when somebody is passing by. I m convinced that most of the sound comes beneath the gap of the door. And luckely that's where my spreakers are. But first I will try it with a radio and put some relaxing music on. 🙂

Do you think it this could work?

pschatzmann, 2022-10-01:
I expect that whatever you try to do, It will  not be fast enough: you will have latency when reading the signal and add more latency when you try to output your negative signal: so instead of cancelling you might increase the noise.

I think a better approach might be that, as soon as you notice a noise level above a threshold, you start your favourite music or sound that is louder then the annoying dog!

podaen, 2022-10-01:
That's an idea. I will give it a try. Better than a frustated dog that want to take a walk. Thanks for the advice.

luke2023, 2022-10-01:
@podaen great idea,I suggest we can also cancel the noise after recording

podaen, 2022-10-01:
I like some inspiration. If we could make are home a happier place to live in, we would sigh for that. Not!? Never the less it's not an easy thing to do. I think we need some equipment for that and counter the latency of the mic that is verry minimized. Until than we enjoying the sound of music. 😆