pschatzmann/arduino-audio-tools #410: Question on streams-audiokit-tf

iannecj, 2022-10-15:
Again thanks for your excellent work!

Question:  I don't see serial monitor response to yes/no in example.

I expect I am not operating this correctly.  This should use on board mic correct?

My audiokit version is selected as 5 in setting header file and this works for other examples so I am guessing that is not it.

If I re-configure audiologger to info I see:
`
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[I] AudioTypes.h : 53 - sample_rate: 16000
[I] AudioTypes.h : 54 - channels: 1
[I] AudioTypes.h : 55 - bits_per_sample: 16
[I] I2SConfig.h : 95 - rx/tx mode: RX
[I] I2SConfig.h : 96 - port_no: 0
[I] I2SConfig.h : 97 - is_master: Master
[I] I2SConfig.h : 98 - sample rate: 16000
[I] I2SConfig.h : 99 - bits per sample: 16
[I] I2SConfig.h : 100 - number of channels: 1
[I] I2SConfig.h : 101 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 103 - auto_clear:1
[I] I2SConfig.h : 110 - buffer_count:16
[I] I2SConfig.h : 111 - buffer_size:512
[I] I2SConfig.h : 115 - pin_bck: 14
[I] I2SConfig.h : 116 - pin_ws: 15
[I] I2SConfig.h : 117 - pin_data: -1515870811
[W] AudioKit.h : 51 - sd_active = true
[I] AudioKit.h : 199 - Channels 1->2
[I] AudioKit.h : 590 - void audio_tools::AudioKitStream::setupActions()
[W] AudioKit.h : 602 - Mode Button ignored because of conflict: 13 
[I] AudioActions.h : 42 - ActionLogic::add pin: 39 / logic: 2
[I] AudioActions.h : 55 - pin 39 -> INPUT
[I] AudioKit.h : 340 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 52 - pin 18 -> INPUT_PULLUP
[I] AudioKit.h : 340 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 52 - pin 5 -> INPUT_PULLUP
[I] TfLiteAudioStream.h : 961 - virtual bool audio_tools::TfLiteAudioStream::setupInterpreter()
[I] TfLiteAudioStream.h : 836 - AllocateTensors
[I] TfLiteAudioStream.h : 844 - Get Input
[I] TfLiteAudioStream.h : 856 - Get Buffer
[I] TfLiteAudioStream.h : 870 - done
[I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] TfLiteAudioStream.h : 567 - ->slices: 2

`

pschatzmann, 2022-10-25:
I could not reproduce your issue. So I decided to merge the latest changes to https://github.com/pschatzmann/tflite-micro-arduino-examples and retry again: For me it is still working....

iannecj, 2022-10-26:
Thanks for looking into this!

Still looks the same with fresh libraries installs and using default example.  Maybe I am not using the example correctly?  What should I see in serial monitor output below?
details:
Assuming some inadvertent change somewhere caused a problem I delete libraries for 
arduino-audiokit
arduino-audio-tools
arduino-tflite 
tflite-micro-arduino-examples

then reinstalled fresh

I did a quick test with a previously working example related to SD card and It still works.

I then retried example streams-audiokit-tf

I see the following in serial monitor with the default example (i.e. not changing Autologger settings to Info):

Message (Ctrl + Enter to send message to 'ESP32 Dev Module' on 'COM4')
New Line
115200 baud
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[W] AudioKit.h : 53 - sd_active = true
[W] AudioKit.h : 604 - Mode Button ignored because of conflict: 13

pschatzmann, 2022-10-26:
Strange, when I start the sketch I see the evaluation results right away.

If you have installed arduino-tflite, that's maybe the disturbing piece. I only have
- pschatzmann/arduino-audiokit
- pschatzmann/arduino-audio-tools
- pschatzmann/tflite-micro-arduino-examples

iannecj, 2022-10-26:
Ok thanks for this tip.  I think we  are onto something different between our systems here.  That's what I am thinking to look for anyway since we have the same hardware, and same arduino source code.

I started by deleted other tflite related libraries as you suggested.

I recompiled and now paid attention to complier window in Arduino IDE.  Again it seems to compile fine and will upload.
However, I see this message in compiler output:
`
Library tflite-micro-arduino-examples-main has been declared precompiled:
Precompiled library in "D:\Documents\Arduino\libraries\tflite-micro-arduino-examples-main\src\esp32" not found
Sketch uses 888797 bytes (67%) of program storage space. Maximum is 1310720 bytes.
Global variables use 26772 bytes (8%) of dynamic memory, leaving 300908 bytes for local variables. Maximum is 327680 bytes.
`

I then go and turn on verbose output in arduinp IDE preferences and see many things that may be related. I am deleting some lines that dont seem related to get size down.

**I wonder if I can force it to compile fresh one time?**

```
Using board 'esp32' from platform in folder: C:\Users\Chris Iannello\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5
Using core 'esp32' from platform in folder: C:\Users\Chris Iannello\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5
cmd /c if exist "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2022926-14580-l12zc.gcyswr\\streams-audiokit-tf\\partitions.csv" COPY /y "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2022926-14580-l12zc.gcyswr\\streams-audiokit-tf\\partitions.csv" "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\partitions.csv"
cmd /c if not exist "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\partitions.csv" if exist "C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5\\variants\\esp32\\partitions.csv" COPY "C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5\\variants\\esp32\\partitions.csv" "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\partitions.csv"
cmd /c if not exist "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\partitions.csv" COPY "C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5\\tools\\partitions\\default.csv" "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\partitions.csv"
cmd /c IF EXIST "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2022926-14580-l12zc.gcyswr\\streams-audiokit-tf\\bootloader.bin" ( COPY /y "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2022926-14580-l12zc.gcyswr\\streams-audiokit-tf\\bootloader.bin" "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\streams-audiokit-tf.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5\\variants\\esp32\\bootloader.bin" ( COPY "C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5\\variants\\esp32\\bootloader.bin" "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\streams-audiokit-tf.ino.bootloader.bin" ) ELSE ( "C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.2.1/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\streams-audiokit-tf.ino.bootloader.bin" "C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5\\tools\\sdk\\esp32\\bin\\bootloader_qio_80m.elf" ) )
esptool.py v4.2.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
cmd /c if exist "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2022926-14580-l12zc.gcyswr\\streams-audiokit-tf\\build_opt.h" COPY /y "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2022926-14580-l12zc.gcyswr\\streams-audiokit-tf\\build_opt.h" "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\build_opt.h"
cmd /c if not exist "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\build_opt.h" type nul > "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2\\build_opt.h"
Detecting libraries used...
"C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.2\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/freertos/include" "-IC:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/includ [... cut ...]
Using library Wire at version 2.0.0 in folder: C:\Users\Chris Iannello\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\Wire 
"C:\\Users\\Chris Iannello\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-size" -A "C:\\Users\\Chris Iannello\\AppData\\Local\\Temp\\arduino-sketch-EDD121CFCA5FBF60847BEE9067E6C0B2/streams-audiokit-tf.ino.elf"
Sketch uses 888797 bytes (67%) of program storage space. Maximum is 1310720 bytes.
Global variables use 26772 bytes (8%) of dynamic memory, leaving 300908 bytes for local variables. Maximum is 327680 bytes.
```

pschatzmann, 2022-10-27:
I don't see any issue here: most messages are coming because of precompiled=full in library.properties
You can increase the program storage by changing the partition scheme...

iannecj, 2022-10-27:
So I could force it to recompile everything by changing that property?

Do you note how it says a precomposed file not found at top of my comment
in first code block.  You think that note is ok too? Not an issue?

pschatzmann, 2022-10-27:
Why would you want to recompile? This is automatically done if necessary...
For precompiled=full see the documentation:
https://arduino.github.io/arduino-cli/0.28/library-specification/

iannecj, 2022-10-27:
Only because with the same sketch and I believe hardware you have this one (of those from audio tools I have tried which is many) example does not work for me but does you.

When I saw the message below I thought this could be reason  So you get the same message?

Library tflite-micro-arduino-examples-main has been declared precompiled: Precompiled library in "D:\Documents\Arduino\libraries\tflite-micro-arduino-examples-main\src\esp32" not found"

pschatzmann, 2022-10-27:
yes

iannecj, 2022-10-27:
Ok thanks for trying to get me running here on audiokit hardware with Arduino audio tools library.   Well I am out of ideas.    I will move over to another tensor flow for esp32 project and see if I am able to get voice recognition going.

pschatzmann, 2022-11-19:
I am closing this issue, since I can't reproduce it...

closed 2022-11-19 by pschatzmann