pschatzmann/arduino-audio-tools #413: Playback streaming while saving mp3 file and DMA usage for streamcopy

0x0fe, 2022-10-16:
I did not found if there is an API to stream playback of an mp3 file while saving it to SD card, that is, we provide a standard file url, start the download and whenever enough data is buffered it starts the playback at the same time.
Also, i wonder if there is any implementation of copier.copy with DMA or if it has to be called in an endless loop no matter what, besides, i wonder if copier.copy can live in a separate thread.
Thanks.

pschatzmann, 2022-10-16:
Sure, you can execute the copy in your own task.
You can use a MultiOutput class to send the output to multiple destinations...

⁣Get TypeApp for Android ​

0x0fe, 2022-10-16:
okay, could you clairfy the usage of #define USE_URLSTREAM_TASK?
when i enable it, i get stuck when the stream should start

[image]

I am not sure if copier.copy(); shoudl still be used when enabled, but earlier i tried to enable USE_URLSTREAM_TASK and comment copier.copy(); but it did not work either.

Also, when the stream is started (USE_URLSTREAM_TASK disabled) i notice there is numerous drops in the audio whenever streamcopy doesnt get 1024 bytes, so i wonder how to mitigate this, i tried to increase #define URL_STREAM_BUFFER_COUNT 10 to 20, but it did not change anything.
My test just uses 
`url.begin("http://stream.srg-ssr.ch/m/rsj/mp3_128", "audio/mp3");`

[image]

pschatzmann, 2022-10-17:
the api is still the same! It just adds another queue to the processing. 

⁣Get TypeApp for Android ​

***@***.***>

closed 2022-10-27 by pschatzmann