pschatzmann/arduino-audio-tools #417: player-sd-audiokit no data found in mp3 file

DharmikEC, 2022-10-20:
Thank you for such great library. 

I am using ESP32 based audiokit v2.2 and trying to get data from SD card. I am using example player-sd-audiokit.

Problem is that it detects the mp3 file in the SD card but on the next line it says no data found and I dont hear voice too. 

 `17:41:13.255 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
17:41:13.255 -> configsip: 0, SPIWP:0xee
17:41:13.255 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:41:13.255 -> mode:DIO, clock div:1
17:41:13.255 -> load:0x3fff0030,len:1344
17:41:13.255 -> load:0x40078000,len:13864
17:41:13.255 -> load:0x40080400,len:3608
17:41:13.255 -> entry 0x400805f0
17:41:13.473 -> [I] AudioTypes.h : 53 - sample_rate: 44100
17:41:13.473 -> [I] AudioTypes.h : 54 - channels: 2
17:41:13.473 -> [I] AudioTypes.h : 55 - bits_per_sample: 16
17:41:13.473 -> [I] I2SConfig.h : 98 - rx/tx mode: TX_MODE
17:41:13.473 -> [I] I2SConfig.h : 99 - port_no: 0
17:41:13.473 -> [I] I2SConfig.h : 100 - is_master: Master
17:41:13.505 -> [I] I2SConfig.h : 101 - sample rate: 44100
17:41:13.505 -> [I] I2SConfig.h : 102 - bits per sample: 16
17:41:13.505 -> [I] I2SConfig.h : 103 - number of channels: 2
17:41:13.505 -> [I] I2SConfig.h : 104 - i2s_format: I2S_STD_FORMAT
17:41:13.505 -> [I] I2SConfig.h : 106 - auto_clear:0
17:41:13.505 -> [I] I2SConfig.h : 113 - buffer_count:10
17:41:13.505 -> [I] I2SConfig.h : 114 - buffer_size:512
17:41:13.505 -> [I] I2SConfig.h : 118 - pin_bck: 14
17:41:13.537 -> [I] I2SConfig.h : 119 - pin_ws: 15
17:41:13.537 -> [I] I2SConfig.h : 120 - pin_data: -1515870811
17:41:13.537 -> [W] AudioKit.h : 53 - sd_active = true
17:41:13.537 -> [I] AudioKit.h : 201 - Channels 2->2
17:41:13.537 -> [I] AudioKit.h : 592 - void audio_tools::AudioKitStream::setupActions()
17:41:13.537 -> [I] AudioKit.h : 342 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
17:41:13.569 -> [I] AudioActions.h : 42 - ActionLogic::add pin: 13 / logic: 0
17:41:13.569 -> [I] AudioActions.h : 52 - pin 13 -> INPUT_PULLUP
17:41:13.569 -> [I] AudioActions.h : 42 - ActionLogic::add pin: 39 / logic: 2
17:41:13.569 -> [I] AudioActions.h : 55 - pin 39 -> INPUT
17:41:13.569 -> [W] AudioKit.h : 622 - Volume Buttons ignored because of conflict: 18 
17:41:13.601 -> [I] AudioPlayer.h : 276 - setVolume(0.700000)
17:41:13.601 -> [I] AudioStreams.h : 1103 - setVolume: 0.700000
17:41:13.601 -> [I] AudioStreams.h : 1103 - setVolume: 0.700000
17:41:13.641 -> [I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
17:41:13.641 -> [I] AudioSourceSD.h : 64 - selectStream: 0
17:41:13.641 -> [E] SDDirect.h : 69 - Open failed: System Volume Information
17:41:13.641 -> [I] AudioSourceSD.h : 68 - Using file example_mp3.mp3
17:41:13.641 -> [W] AudioPlayer.h : 141 - -> begin: no data found
`

It can be seen it has found example_mp3.mp3 file means no issue in SD card interface I think. Also the file I am using, runs on PC and I have tried separate esp32 with external card reader and same file is working. Yet I have tried different file and same output.

Any help is appreciated. Thank you.

pschatzmann, 2022-10-20:
Did you consider the [AudioKit Wiki](https://github.com/pschatzmann/arduino-audiokit/wiki) ?

DharmikEC, 2022-10-20:
I tried defining SPI pins without initiating AudioKitStream config if that is what you meant but no luck.

```
// setup output
  // auto cfg = kit.defaultConfig(TX_MODE);
  // // sd_active is setting up SPI with the right SD pins by calling 
  SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
  // cfg.sd_active = true;
  // kit.begin(cfg);
```
In this case output looks like this
```
20:41:05.179 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
20:41:05.179 -> configsip: 0, SPIWP:0xee
20:41:05.179 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:41:05.225 -> mode:DIO, clock div:1
20:41:05.225 -> load:0x3fff0030,len:1344
20:41:05.225 -> load:0x40078000,len:13864
20:41:05.225 -> load:0x40080400,len:3608
20:41:05.225 -> entry 0x400805f0
20:41:05.440 -> [I] AudioPlayer.h : 276 - setVolume(0.700000)
20:41:05.440 -> [I] AudioStreams.h : 1103 - setVolume: 0.700000
20:41:05.440 -> [I] AudioStreams.h : 1103 - setVolume: 0.700000
20:41:05.440 -> [I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
20:41:05.487 -> [I] AudioSourceSD.h : 64 - selectStream: 0
20:41:05.487 -> [E] SDDirect.h : 69 - Open failed: System Volume Information
20:41:05.487 -> [I] AudioSourceSD.h : 68 - Using file example_mp3.mp3
20:41:05.487 -> [W] AudioPlayer.h : 141 - -> begin: no data found
```
DIP switches are 0,1,1,0,0 too. 
I tried different cards too. Currently I am using microSDHC. I have tried microSDXC too but that fails in the beginning.
SD card was formatted with FAT32. 
Is there any other intermediate way to debug or some other test code I can try ? I was never able to make SD card work. 
In audiokit-sd-audiokit example also, code runs smoothly but I cant see file created in SD card. Separate example of microphone and speaker is working fine. So the issue is only with SD card. 

Thank you very much for help.

pschatzmann, 2022-10-20:
I suggest to use the standard SD formatter as explained in the link. 

For me SD, SDMMC and SDFAT are working and it does not depend on the AudioKit version since all use the same wiring.
Don't forget to restart the kit after changing the dip settings.