pschatzmann/arduino-audio-tools #420: player-sd-audiokit error: 'class fs::File' has no member named 'open'

Theagainmen, 2022-10-20:
Hi,

## Setup and error output
Using an ESP32 Lyrat board from Espressif (audiokit). I've tried and used the `streams-audiokit-webserver_wav` succesfully with my setup already. The `player-sd-audiokit` however spits out the following error:

```
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h: In instantiation of 'FileT audio_tools::SDDirect<SDT, FileT>::open(const char*) [with SDT = fs::SDFS; FileT = fs::File]':
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:65:24:   required from 'void audio_tools::SDDirect<SDT, FileT>::listDir(const char*) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:42:14:   required from 'const char* audio_tools::SDDirect<SDT, FileT>::operator[](int) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/AudioSourceSD.h:66:26:   required from here
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:230:11: error: 'class fs::File' has no member named 'open'
       if (!result.open(name)){
           ^
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h: In instantiation of 'bool audio_tools::SDDirect<SDT, FileT>::isDirectory(FileT) [with SDT = fs::SDFS; FileT = fs::File]':
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:71:23:   required from 'void audio_tools::SDDirect<SDT, FileT>::listDir(const char*) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:42:14:   required from 'const char* audio_tools::SDDirect<SDT, FileT>::operator[](int) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/AudioSourceSD.h:66:26:   required from here
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:127:14: error: 'class fs::File' has no member named 'isDir'
       result = f.isDir();
              ^
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h: In instantiation of 'void audio_tools::SDDirect<SDT, FileT>::rewind(FileT) [with SDT = fs::SDFS; FileT = fs::File]':
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:82:15:   required from 'void audio_tools::SDDirect<SDT, FileT>::listDir(const char*) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:42:14:   required from 'const char* audio_tools::SDDirect<SDT, FileT>::operator[](int) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/AudioSourceSD.h:66:26:   required from here
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:118:7: error: 'class fs::File' has no member named 'rewind'
       f.rewind();
       ^
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h: In instantiation of 'FileT audio_tools::SDDirect<SDT, FileT>::openNext(FileT&) [with SDT = fs::SDFS; FileT = fs::File]':
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:85:28:   required from 'void audio_tools::SDDirect<SDT, FileT>::listDir(const char*) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:42:14:   required from 'const char* audio_tools::SDDirect<SDT, FileT>::operator[](int) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/AudioSourceSD.h:66:26:   required from here
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:139:11: error: 'class fs::File' has no member named 'openNext'
       if (!result.openNext(&dir, O_READ)){
           ^
.pio\libdeps\esp32dev\audiokit\src\audio_board\lyrat_v4_2\board.c:32:20: warning: 'TAG' defined but not used [-Wunused-variable]
 static const char *TAG = "AUDIO_BOARD";
                    ^
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h: In instantiation of 'const char* audio_tools::SDDirect<SDT, FileT>::fileName(FileT&) [with SDT = fs::SDFS; FileT = fs::File]':
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:90:28:   required from 'void audio_tools::SDDirect<SDT, FileT>::listDir(const char*) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:42:14:   required from 'const char* audio_tools::SDDirect<SDT, FileT>::operator[](int) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/AudioSourceSD.h:66:26:   required from here
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:183:8: error: 'class fs::File' has no member named 'getName'
        file.getName(name,MAX_FILE_LEN);        
        ^
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h: In instantiation of 'bool audio_tools::SDDirect<SDT, FileT>::isHidden(FileT) [with SDT = fs::SDFS; FileT = fs::File]':
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:172:33:   required from 'bool audio_tools::SDDirect<SDT, FileT>::isValidAudioFile(FileT&) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:95:31:   required from 'void audio_tools::SDDirect<SDT, FileT>::listDir(const char*) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:42:14:   required from 'const char* audio_tools::SDDirect<SDT, FileT>::operator[](int) [with SDT = fs::SDFS; FileT = fs::File]'
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/AudioSourceSD.h:66:26:   required from here
.pio\libdeps\esp32dev\audio-tools\src/AudioLibs/SDDirect.h:219:25: error: 'class fs::File' has no member named 'isHidden'
       return f.isHidden();
                         ^
.pio\libdeps\esp32dev\audiokit\src\audio_board\lyrat_v4_2\board_pins_config.c:36:20: warning: 'TAG' defined but not used [-Wunused-variable]
 static const char *TAG = "LYRAT_V4_2";
                    ^
*** [.pio\build\esp32dev\src\player-sd-audiokit.cpp.o] Error 1
```

## Platformio.ini file and libraries
My platformio.ini file is setup according to the wiki:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

board_build.f_cpu = 240000000L      ; 240MHz
upload_speed = 921600           
monitor_speed = 115200

; From the Wiki https://github.com/pschatzmann/arduino-audio-tools/wiki/Working-with-PlatformIO 
lib_ldf_mode = deep+
monitor_filters = esp32_exception_decoder
build_flags =  -DCORE_DEBUG_LEVEL=5 -DAUDIOKIT_BOARD=1 -DUSE_SDFAT -DUSE_HELIX

lib_deps = 
    https://github.com/pschatzmann/arduino-audio-tools
    https://github.com/pschatzmann/arduino-audiokit
    https://github.com/pschatzmann/arduino-libhelix
    https://github.com/greiman/SdFat
```

I've looked through the [audokit wiki](https://github.com/pschatzmann/arduino-audiokit/wiki) but couldn't find a solution there either. Am I missing a library or something?

Thanks for any help!

pschatzmann, 2022-10-20:
I suggest that you consult the [Wiki](https://github.com/pschatzmann/arduino-audio-tools/wiki/The-Audio-Player-Class).
You cant mix the different libraries and if you include the SDFAT library you need to use the corresponding implementation!

Theagainmen, 2022-10-20:
Thanks for the quick reply, I copied the lib_deps from another example I think, it all works fine when I use the other example called `player-sdfat-audiokit` after setting some DIP switches :)!

closed 2022-10-20 by Theagainmen