pschatzmann/arduino-audio-tools #423: M5Stack Core2

NDC-SC, 2022-10-23:
I'd like to use the M5Stack Core2 module rather than the combination of ESP32 Dev Board and SD card read/write module to transmit sound from MP3 files held on an SD card to a Bluetooth speaker with the player-sdfat-a2dp sketch.
The SPI pins of the ESP32 Dev Board are the same as the M5Stack Core2 (CS: 5, SCK: 18, MOSI: 23), except for MOSI which is GPIO 19 on ESP32 Dev Board and GPIO 38 for M5Stack Core2.
In the AudioConfig.h file, several parameters can be changed (such as PIN_CS) but I can't see how to update the MOSI pin.
Any help would be appreciated.

NDC-SC, 2022-10-23:
Typo in comment. The MISO pin is different GPIO 19 on ESP32 Dev Board and GPIO 38 for M5Stack Core2.
It's MISO that requires changing.

NDC-SC, 2022-10-23:
One possible option is to change the pins_arduino.h file, but I would prefer to make the change in the sketch,

pschatzmann, 2022-10-24:
This has nothing to do with my library and this is a basic ESP32 functionality. 
You can define the SPI pins in the begin call:   void begin(int8_t sck=-1, int8_t miso=-1, int8_t mosi=-1, int8_t ss=-1);

I suggest that you have a look at the .h file provided by the ESP32: https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h

closed 2022-10-24 by NDC-SC

NDC-SC, 2023-05-06:
In the examples> audio-tools> player-sdfat-a2dp sketch
replacement of the line: AudioSourceSDFAT source(startFilePath, ext),
with the lines
SdSpiConfig sdcfg(SD_CS, DEDICATED_SPI, SD_SCK_MHZ(10))
AudioSourceSDFAT source(startFilePath, ext, sdcfg)
resolves the error message of 
[W] AudioA2DP.h : 319 - ==> state: Connecting
[W] AudioA2DP.h : 319 - ==> state: Connected
[E] AudioSourceSDFAT.h : 89 - sd.begin failed
[E] AudioSourceSDFAT.h : 114 - Filename is null
[W] AudioPlayer.h : 161 - -> begin: no data found

The M5Stack Core2 SPI pins are defined as
int SD_CS = 4, SD_SCK = 18, SD_MISO = 38, SD_MOSI = 23; 
and in void setup, SPI and SD pins are initialised with
SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS)
SD.begin(SD_CS)