pschatzmann/arduino-audio-tools #442: Use INMP441 with AudioKit

iannecj, 2022-11-05:
Goal is use INMP441 with Audiokit to make use of its SD, integrated buttons, and line OUT.  In addition to learning how to use AudioTools frame work better I need to locate the microphone away from the board.

1. Ran audio tools example streams-i2s-a2dp.ino on normal esp32 to test out my INMP441  - **Success**
2. Changed default pins in Audioconfig.h to pins accessible on pin header of Audiokit and switched to streams-i2s-a2dp.ino and moved over to audio kit board - **Success**
3. Now the example that is most like my application is audiokit-sd-audiokit but I thought I would start with audiokit-ram-audiokit before I tried to get SD working.  
4. I run the default example fine using onboard analog mics.  **My problem starts when I try to use kit for I2S output and INMP mic for I2S input**
5. The code I am trying to integrate is below.  I am showing with example that compiles and runs fine but with same original functionality.
6. You can see I have moved parts from the working streams-i2s-a2dp.ino over but they are **_mostly_** commented out.
7. If I merely uncomment the I2SStream configuration/begin lines of code the base project functionality with analog mic still works as it should but volume is drastically effected and the debug I2S error to register receive mode object as well as alternating audio kit volume change messages in the serial monitor.

```
    auto config_mic = i2sStream.defaultConfig(RX_MODE);  // //Chris add  
    i2sStream.begin(config_mic);
```

I then see the following in debug:

```
[I] I2SConfig.h : 120 - pin_data: 5  
E (208) I2S: register I2S object to platform failed  
[E] I2SESP32.h : 168 - begin - i2s_driver_install  
[I] AudioActions.h : 42 - ActionLogic::add pin: 36 / logic: 0
[I] AudioActions.h : 52 - pin 36 -> INPUT_PULLUP
Press Key 1 to record
[I] AudioKit.h : 392 - static void audio_tools::AudioKitStream::actionVolumeDown(bool, int, void*)
[I] AudioKit.h : 374 - incrementVolume: -2 -> 98
Playing...
[I] AudioCopy.h : 74 - buffer_size=1024
[I] AudioKit.h : 383 - static void audio_tools::AudioKitStream::actionVolumeUp(bool, int, void*)
[I] AudioKit.h : 374 - incrementVolume: 2 -> 100
[I] AudioKit.h : 392 - static void audio_tools::AudioKitStream::actionVolumeDown(bool, int, void*)
[I] AudioKit.h : 374 - incrementVolume: -2 -> 98
```

9. If I go for broke and make what seems like logical changes taking kit to TX Mode  and using INMP as RX Mode, copying the converter filler function output to recording instead of kit etc. **you can see the commented version below of what I would add once I figure out how to register an I2S object separate from those on kit**.

```cpp
/**
 * @file streams-audiokit-sd-audiokit.ino
 * @author Phil Schatzmann
 * @brief We record the input from the microphone to SPI RAM and constantly repeat to play the file
 * The input is triggered by pressing key 1. Recording stops when key 1 is released!
 * @version 0.1
 * @date 2022-09-01
 * 
 * @copyright Copyright (c) 2022
 * 
 */
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "AudioLibs/MemoryManager.h"

uint16_t sample_rate = 16000;
uint8_t channels = 1;  // The stream will have 2 channels 
AudioKitStream kit;
MemoryManager memory(500); // Activate SPI RAM for objects > 500 bytes
DynamicMemoryStream recording(true); // Audio stored on heap 
StreamCopy copier; // copies data

I2SStream i2sStream;                             // Chris Adds for I2S Mic
ConverterFillLeftAndRight<int16_t> filler(Auto);  // fill both channels Chris Adds for I2S Mic

void record_start(bool pinStatus, int pin, void* ref){
  Serial.println("Recording...");
  recording.begin();
  copier.begin(recording, kit);
  //copier.begin(recording, filler);   
}

void record_end(bool pinStatus, int pin, void* ref){
  Serial.println("Playing...");

  // Remove popping noise, from button: we delete 6 segments at the beginning and end 
  // and on the resulting audio we slowly raise the volume on the first segment
  // end decrease it on the last segment
  recording.postProcessSmoothTransition<int16_t>(channels, 0.01, 6);
  copier.begin(kit, recording);  // start playback
}

void setup(){
  Serial.begin(115200);
  while(!Serial); // wait for serial to be ready
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  

  // setup input and output
  auto cfg = kit.defaultConfig(RXTX_MODE);
 //auto cfg = kit.defaultConfig(TX_MODE); //Chris add
  cfg.sd_active = true;
  cfg.sample_rate = sample_rate;
  cfg.channels = channels;
  cfg.input_device = AUDIO_HAL_ADC_INPUT_LINE2;
  kit.begin(cfg);
  kit.setVolume(1.0);

 auto config_mic = i2sStream.defaultConfig(RX_MODE);  // //Chris add
  //config_mic.allow_boost = true;  // uses linear control allowing values > 1.0. 
 // config_mic.sample_rate = sample_rate;
 // config_mic.channels = channels;
  i2sStream.begin(config_mic);

  // record when key 1 is pressed
  kit.audioActions().add(PIN_KEY1, record_start, record_end);
  Serial.println("Press Key 1 to record");

}

void loop(){

  // record or play recording
  copier.copy(); 
  //copier.copy(filler); //Chris Adds for I2S Mic 
 
  // Process keys
  kit.processActions();

}

```

**_So how can I modify this code above to work?_**

pschatzmann, 2022-11-05:
I am surprised that you have INMP441 working. Did you need to change the hardware somehow ?
If you use a I2SStream in parallel with the AudioKitStream you need to assign them to port 1 since 0 is used by the audiokit.

iannecj, 2022-11-05:
Thanks for the lightning fast response especially since these are all novice questions.  Let me answer your question immediately since this might help others.

There was no hardware change though I had my doubts it would work.  I chose pins 23, 18, and 5 because the Audio kit uses these for the last 3 buttons.  My supposition which has since seemed to work is that the resistive pull up is there is weak enough that the resistor to header pin can still allow the INMP441 to communicate.  I know its using digital mic as code I use does not configure kit or analog mics and if I disconnect INMP441 power I cant record audio as you'd expect.

[image]

[image]

Project defaults from example sketch readme file with my revisions in RED overlaid

[image]

[image]

And the following code which is from base audiotools library slightly modified works.  **_Of course it doesn't try to use AudiokitStream in parallel._**

Also, it still has pins redefined in AudioConfig.h to use the Audiokit Pin Header pins 23, 18, and 5 as shown below because when I use the pin config inside I2Sconfig **doesn't seem to work**.  You can see my alternate try commented out just below config block in code snipped below.  What is here and is  uncommented works but as you can see is still changing defaults.  Why would that be?  

```cpp
#define PWM_FREQENCY 30000
#define PIN_PWM_START 12
//#define PIN_I2S_BCK 14
#define PIN_I2S_BCK 23
//#define PIN_I2S_WS 15
#define PIN_I2S_WS 18
//#define PIN_I2S_DATA_IN 32
#define PIN_I2S_DATA_IN 5
#define PIN_I2S_DATA_OUT 22
#define I2S_USE_APLL false  
```

```cpp
/**
 * @file streams-i2s-a2dp.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-i2s-a2dp/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioLibs/AudioA2DP.h"

I2SStream i2sStream;                             // Access I2S as stream
A2DPStream a2dpStream = A2DPStream::instance();  // access A2DP as stream
VolumeStream volume(a2dpStream);
StreamCopy copier(volume, i2sStream);             // copy i2sStream to a2dpStream
ConverterFillLeftAndRight<int16_t> filler(Auto);  // fill both channels

// Arduino Setup
void setup(void) {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // set intial volume
  volume.setVolume(1.0);

  // start bluetooth
  Serial.println("starting A2DP...");
  auto cfgA2DP = a2dpStream.defaultConfig(TX_MODE);
  cfgA2DP.name = "BT-WUZHI";
  a2dpStream.begin(cfgA2DP);
 
  // start i2s input with default configuration
  Serial.println("starting I2S...");
  a2dpStream.setNotifyAudioChange(i2sStream);  // i2s is using the info from a2dp
  
  
  auto config_mic = i2sStream.defaultConfig(RX_MODE);
[... 114 lines cut ...]
  auto config_mic = i2sStream.defaultConfig(RX_MODE);
  // config.allow_boost = true;  // uses linear control allowing values > 1.0. 
   config_mic.pin_bck = 23;
  config_mic.pin_ws = 18;
  config_mic.pin_data_rx = 5;
  i2sStream.begin(config_mic);

   auto config = volume.defaultConfig();
  config.allow_boost = true;  // uses linear control allowing values > 1.0. 
  volume.begin(config);
  volume.setVolume(5); // Too big values lead to clipping!

   
}

// Arduino loop - copy data
void loop() {
   copier.copy(filler);
  //copier.copy();
}
```

```

```

pschatzmann, 2022-11-05:
Of cause you can use I2SStream and AudioKitStream in parallel. Just apply my suggested solution above!

iannecj, 2022-11-05:
In the code above I wasn't using the audiokit example I am working on as that issue post was mainly answering your question and showing that I was able to use audiokit hardware with INMP441 mic.  That code does not need port definition since only one i2s port right? That example based on your streams-i2s-a2dp

In the code I try to adjust for my stated goal above (streams-audiokit-ram-audiokit.ino) which I now show below  I did add port when you suggested and did correct _rx pin definition as instructed.  Apologies if _rx pin definition should have been clear to me from your suggestion immediately.  You also provided a link to class docs and there was an option therefor _rx.  I thought in your example since you did not use my pin numbers it was just notional and not specific input and you wanted me to review docs. 

Nevertheless, see the code below which I believe addresses all your inputs thus far.  I need to go though documentation link again and again since as I see it suggests since inmp441 outputs 32 bit I likely need converter.  I am attempting to add suggested converter with some **SUCCESS**.  **However I get much lower volume but I hear it noise free from both channels.**

I think its related to some incompatibility between audiokit and I2S stream that I have not corrected yet. The reason I think that is  I still see continuous audiokit info messages on changing volume that only appear when I have defined I2S stream even if I use default example in every other way and before any buttons are pushed.  That makes me wonder, no physiucal button push but I wonder of the way I wire I2S mic into that header is it.   Were those IO pins used for some volume control and they are now somehow dual purposed?

 **If so any idea about how to deconflict these?**  Whenever these are present volume seems to be reduced dramatically also.

```
[I] AudioTypes.h : 54 - sample_rate: 16000
[I] AudioTypes.h : 55 - channels: 1
[I] AudioTypes.h : 56 - bits_per_sample: 16
[I] I2SConfig.h : 98 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 99 - port_no: 0
[I] I2SConfig.h : 100 - is_master: Master
[I] I2SConfig.h : 101 - sample rate: 16000
[I] I2SConfig.h : 102 - bits per sample: 16
[I] I2SConfig.h : 103 - number of channels: 1
[I] I2SConfig.h : 104 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 106 - auto_clear:0
[I] I2SConfig.h : 113 - buffer_count:10
[I] I2SConfig.h : 114 - buffer_size:512
[I] I2SConfig.h : 118 - pin_bck: 14
[I] I2SConfig.h : 119 - pin_ws: 15
[I] I2SConfig.h : 120 - pin_data: -1515870811
[W] AudioKit.h : 53 - sd_active = true
[I] AudioKit.h : 201 - Channels 1->2
[I] AudioKit.h : 592 - void audio_tools::AudioKitStream::setupActions()
[W] AudioKit.h : 604 - Mode Button ignored because of conflict: 13 
[I] AudioActions.h : 42 - ActionLogic::add pin: 39 / logic: 2
[I] AudioActions.h : 55 - pin 39 -> INPUT
[I] AudioKit.h : 342 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 52 - pin 18 -> INPUT_PULLUP
[I] AudioKit.h : 342 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 52 - pin 5 -> INPUT_PULLUP
[W] AudioKit.h : 308 - Volume is > 1.0: 5.000000
[I] I2SConfig.h : 98 - rx/tx mode: RX_MODE
[I] I2SConfig.h : 99 - port_no: 1
[I] I2SConfig.h : 100 - is_master: Master
[I] I2SConfig.h : 101 - sample rate: 16000
[I] I2SConfig.h : 102 - bits per sample: 16
[I] I2SConfig.h : 103 - number of channels: 1
[I] I2SConfig.h : 104 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 106 - auto_clear:0
[I] I2SConfig.h : 113 - buffer_count:10
[I] I2SConfig.h : 114 - buffer_size:512
[I] I2SConfig.h : 118 - pin_bck: 23
[I] I2SConfig.h : 119 - pin_ws: 18
[I] I2SConfig.h : 120 - pin_data: 5
[I] AudioActions.h : 42 - ActionLogic::add pin: 36 / logic: 0
[I] AudioActions.h : 52 - pin 36 -> INPUT_PULLUP

```
Note the messages below continue regardless of whether the recording button has been pushed after a reboot.  They persist even if I comment out all changes to example except the I2SStream configuration with begin.
```
[I] AudioKit.h : 383 - static void audio_tools::AudioKitStream::actionVolumeUp(bool, int, void*)
[I] AudioKit.h : 374 - incrementVolume: 2 -> 84
[I] AudioKit.h : 392 - static void audio_tools::AudioKitStream::actionVolumeDown(bool, int, void*)
[I] AudioKit.h : 374 - incrementVolume: -2 -> 82
[I] AudioKit.h : 383 - static void audio_tools::AudioKitStream::actionVolumeUp(bool, int, void*)
[I] AudioKit.h : 374 - incrementVolume: 2 -> 84
[I] AudioKit.h : 392 - static void audio_tools::AudioKitStream::actionVolumeDown(bool, int, void*)
[I] AudioKit.h : 374 - incrementVolume: -2 -> 82
[I] AudioKit.h : 383 - static void audio_tools::AudioKitStream::actionVolumeUp(bool, int, void*)
[I] AudioKit.h : 374 - incrementVolume: 2 -> 84
Press Key 1 to record
[... 83 lines cut ...]
  
config_mic.pin_bck = 23;
config_mic.pin_ws = 18;
config_mic.pin_data = 5;
config_mic.port_no = 1;
i2sStream.begin(config_mic);

  // record when key 1 is pressed
  kit.audioActions().add(PIN_KEY1, record_start, record_end);
  Serial.println("Press Key 1 to record");

}

void loop(){

  // record or play recording
  copier.copy(); 
  //copier.copy(filler); //Chris Adds for I2S Mic 
 
  // Process keys
  kit.processActions();

}

```

pschatzmann, 2022-11-05:
You might need to deactivate the actions that act on pins which have a conflict with your definitions! 
or comment out kit.processActions(); and implement your own logic

iannecj, 2022-11-06:
Yes thank you very much for helping this novice.
The final code with volume control added.

```cpp
/**
 * @file streams-audiokit-sd-audiokit.ino
 * @author Phil Schatzmann
 * @brief We record the input from the microphone to SPI RAM and constantly repeat to play the file
 * The input is triggered by pressing key 1. Recording stops when key 1 is released!
 * @version 0.1
 * @date 2022-09-01
 * 
 * @copyright Copyright (c) 2022
 * 
 */
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "AudioLibs/MemoryManager.h"

uint16_t sample_rate = 24000;
uint8_t channels = 1;  // The stream will have 2 channels 
AudioKitStream kit;
MemoryManager memory(500); // Activate SPI RAM for objects > 500 bytes
DynamicMemoryStream recording(true); // Audio stored on heap 
I2SStream i2sStream; // Chris Adds for I2S Mic
VolumeStream volume(kit); // Chris adds for volume control
StreamCopy copier; 
ConverterFillLeftAndRight<int16_t> filler(Auto);  // fill both channels Chris Adds for I2S Mic

void record_start(bool pinStatus, int pin, void* ref){
  Serial.println("Recording...");
  recording.begin();
  copier.begin(recording, i2sStream);  //Chris adds to chnage record source
}

void record_end(bool pinStatus, int pin, void* ref){
  Serial.println("Playing...");
  copier.begin(volume, recording);  // start playback
}

void setup(){
  Serial.begin(115200);
  while(!Serial); // wait for serial to be ready
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  

  // setup output
 auto cfg = kit.defaultConfig(TX_MODE); //Chris add
  cfg.sd_active = true;
  cfg.sample_rate = sample_rate;
  cfg.bits_per_sample = 16;
  cfg.channels = channels;
  cfg.port_no = 0;
  kit.begin(cfg);
  kit.setVolume(1.0);
  
auto config_mic = i2sStream.defaultConfig(RX_MODE);  // Chris add config for INMP441
config_mic.sample_rate = sample_rate;
config_mic.channels = channels;
config_mic.bits_per_sample = 16;
config_mic.pin_bck = 23;
config_mic.pin_ws = 18;
config_mic.pin_data = 5;
config_mic.port_no = 1;
i2sStream.begin(config_mic);

auto configvol = volume.defaultConfig();
configvol.allow_boost = true;  // uses linear control allowing values > 1.0. 
configvol.bits_per_sample = 16;
volume.begin(configvol);
volume.setVolume(7.0); // Too big values lead to clipping!

  // record when key 1 is pressed
  kit.audioActions().add(PIN_KEY1, record_start, record_end);
  kit.audioActions().setEnabled(23, false);
  kit.audioActions().setEnabled(18, false);
  kit.audioActions().setEnabled(5, false);
  Serial.println("Press Key 1 to record");

}

void loop(){

// record or play recording
  //copier.copy(); 
  copier.copy(filler); //Chris Adds for I2S Mic 
// Process keys
  kit.processActions();

}
```

closed 2022-11-06 by iannecj