pschatzmann/arduino-audio-tools #447: How to mix multiple MP3 streams

321dev, 2022-11-09:
Suppose i have this code fragment taken from one of the samples:

```
URLStream url("SSID","password");
I2SStream i2s; // final output of decoded stream
VolumeStream volume(i2s);
EncodedAudioStream dec(&volume,new MP3DecoderHelix()); // Decoding stream
StreamCopy copier(dec, url); // copy url to decoder
```
Now i want multiple EncodedAudioStream instances, and they all should output to the mixer, and then the mixer should output to the VolumeStream. Unfortunately i couldn't find an example that does this.

The MP3Decoder writes its output directly into the VolumeStream. But i can't pass an InputMixer instance as output stream.
Instead i need something to pass to InputMixer's add() method.
So, i guess i need some kind of "adapter" class that i can pass to the EncodedAudioStream constructor as well as to InputMixer::add() ?

How can i do it?

321dev, 2022-11-09:
Thanks for the quick response. Could you recommend another encoding that could be decoded fast enough. We're trying to mix four audio streams.

pschatzmann, 2022-11-09:
The best is just to unencoded PCM data. 
The next best is a WAV file (with PCM data)

321dev, 2022-11-09:
How about ADPCM?

pschatzmann, 2022-11-09:
Since I don't have a CODEC for this, I don't know

321dev, 2022-11-09:
Ok. thanks for your quick responses.

closed 2022-11-09 by 321dev