pschatzmann/arduino-audio-tools #452: error: 'audio_tools:: .... ' declared as an 'inline' variable

avidadearthur, 2022-11-16:
Hello, while trying to run the `streams-analog-serial.ino` example the compilation failed due to multiple  inline variable declaration errors. How can I resolve this issue? 

Please let me know if I need to provide more details.
```
(...)
In file included from c:\Users\arthu\OneDrive\Documentos\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/AudioCodecs.h:3:0,
                 from c:\Users\arthu\OneDrive\Documentos\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:29,
                 from C:\Users\arthu\AppData\Local\Temp\.arduinoIDE-unsaved20221016-25256-m24fdy.4uams\streams-adc-serial\streams-adc-serial.ino:10:
c:\Users\arthu\OneDrive\Documentos\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/CodecWAV.h:34:24: error: 'audio_tools::wav_mime' declared as an 'inline' variable
 INLINE_VAR const char* wav_mime = "audio/wav";
                        ^

exit status 1

Compilation error: exit status 1
```

pschatzmann, 2022-11-16:
- Use the latest Arduino ESP32 version and make sure that the compiler warnings are set to default (in the preferences)
- Or change the the USE_INLINE_VARS to 0 in the AudioConfig.h

avidadearthur, 2022-11-16:
Thanks for the quick reply.

> * Use the latest Arduino ESP32 version and make sure that the compiler warnings are set to default (in the preferences)

The compiler warnings were already set to default.
> * Or change the the USE_INLINE_VARS to 0 in the AudioConfig.h

How can I change the file if the directory is set to read-only mode?

pschatzmann, 2022-11-16:
The latest ESP32 version 2.0.5. You can check this in the Board Manager.
I have never seen this to be read only. Anyhow, you should have the right to change this...

pschatzmann, 2022-11-19:
In the latest version, I changed the default setting of USE_INLINE_VARS to 0

closed 2022-11-19 by pschatzmann