pschatzmann/arduino-audio-tools #455: Compiling error

Dzsonzii, 2022-11-18:
Hey guys,
I want to test the example on my ESP32-A1S audio kit (AI Thinker one), and I get these errors when compiling. Any solution to that?
`In file included from D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioI2S/I2SStream.h:9,
                 from D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioTools.h:18,
                 from D:\Arduino\libraries\arduino-audio-tools-0.9.2\examples\examples-audiokit\basic-audiokit-a2dp\basic-audiokit-a2dp.ino:9:
D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioI2S/I2SESP32.h: In member function 'bool audio_tools::I2SBase::begin(audio_tools::I2SConfig, int, int)':
D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioI2S/I2SESP32.h:146:28: warning: narrowing conversion of '(eps32_i2s_sample_rate_type)cfg.audio_tools::I2SConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate' from 'eps32_i2s_sample_rate_type' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
             .sample_rate = (eps32_i2s_sample_rate_type)cfg.sample_rate,
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioI2S/I2SESP32.h:181:9: error: 'i2s_pin_config_t' has no non-static data member named 'mck_io_num'
         };
         ^
In file included from D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioAnalog/AnalogAudio.h:3,
                 from D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioTools.h:20,
                 from D:\Arduino\libraries\arduino-audio-tools-0.9.2\examples\examples-audiokit\basic-audiokit-a2dp\basic-audiokit-a2dp.ino:9:
D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioAnalog/AnalogAudioESP32.h: In member function 'bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)':
D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioAnalog/AnalogAudioESP32.h:205:39: warning: narrowing conversion of '(eps32_i2s_sample_rate_type)cfg.audio_tools::AnalogConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate' from 'eps32_i2s_sample_rate_type' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
             .sample_rate            = (eps32_i2s_sample_rate_type)cfg.sample_rate,
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioLibs/AudioKit.h:3,
                 from D:\Arduino\libraries\arduino-audio-tools-0.9.2\examples\examples-audiokit\basic-audiokit-a2dp\basic-audiokit-a2dp.ino:10:
D:\Arduino\libraries\arduino-audiokit-0.6.0\src/AudioKitHAL.h: In member function 'i2s_config_t AudioKitConfig::i2sConfig()':
D:\Arduino\libraries\arduino-audiokit-0.6.0\src/AudioKitHAL.h:124:34: warning: narrowing conversion of '((AudioKitConfig*)this)->AudioKitConfig::sampleRate()' from 'uint32_t' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
         .sample_rate = sampleRate(),
                        ~~~~~~~~~~^~
D:\Arduino\libraries\arduino-audiokit-0.6.0\src/AudioKitHAL.h: In member function 'i2s_pin_config_t AudioKitConfig::i2sPins()':
D:\Arduino\libraries\arduino-audiokit-0.6.0\src/AudioKitHAL.h:142:13: error: 'struct i2s_pin_config_t' has no member named 'mck_io_num'; did you mean 'bck_io_num'?
      result.mck_io_num = I2S_PIN_NO_CHANGE; //mclk_gpio;
             ^~~~~~~~~~
             bck_io_num
In file included from D:\Arduino\libraries\arduino-audio-tools-0.9.2\examples\examples-audiokit\basic-audiokit-a2dp\basic-audiokit-a2dp.ino:10:
D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioLibs/AudioKit.h: In member function 'virtual void audio_tools::AudioKitStream::setAudioInfo(audio_tools::AudioBaseInfo)':
D:\Arduino\libraries\arduino-audio-tools-0.9.2\src/AudioLibs/AudioKit.h:267:11: error: 'class AudioKit' has no member named 'setSampleRate'
       kit.setSampleRate(cfg.toSampleRate());
           ^~~~~~~~~~~~~
`

pschatzmann, 2022-11-18:
How can I reproduce this issue ?

Dzsonzii, 2022-11-18:
I just downloaded the library with all the others you mentioned in the readme (ESP32-A2DP, arduino-audoikot), connected the board to my computer, and tried to upload it. This exact error is from the base-audiokit-a2dp example.

pschatzmann, 2022-11-18:
How did you download and what versions ? The latest version of the AudioKit class definitly has the setSampleRate() method available

pschatzmann, 2022-11-18:
Ok, to prevent that others run into the same issue, I just created a new release...

Dzsonzii, 2022-11-18:
I installed from the code button, now I get these
```
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimerDef.h:2,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimerESP32.h:4,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimer.h:3,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:11,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:9:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioTypes.h:13:38: warning: inline variables are only available with -std=c++17 or -std=gnu++17
 INLINE_VAR const char* RxTxModeNames[]={"UNDEFINED_MODE","TX_MODE","RX_MODE","RXTX_MODE" };
                                      ^
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioTypes.h:97:24: warning: inline variables are only available with -std=c++17 or -std=gnu++17
 INLINE_VAR const char* mime_pcm = "audio/pcm";
                        ^~~~~~~~
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimer.h:3,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:11,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:9:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimerESP32.h:41:15: warning: inline variables are only available with -std=c++17 or -std=gnu++17
 } INLINE_VAR *simpleUserCallback = nullptr;
               ^~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimerESP32.h:93:15: warning: inline variables are only available with -std=c++17 or -std=gnu++17
 } INLINE_VAR *timerCallbackArray = nullptr;
               ^~~~~~~~~~~~~~~~~~
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SStream.h:8,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:18,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:9:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SConfig.h:24:36: warning: inline variables are only available with -std=c++17 or -std=gnu++17
 INLINE_VAR const char* i2s_formats[] = {"I2S_STD_FORMAT","I2S_LSB_FORMAT","I2S_MSB_FORMAT","I2S_PHILIPS_FORMAT","I2S_RIGHT_JUSTIFIED_FORMAT","I2S_LEFT_JUSTIFIED_FORMAT","I2S_PCM_LONG","I2S_PCM_SHORT"};
                                    ^
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SStream.h:9,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:18,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:9:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h: In member function 'bool audio_tools::I2SBase::begin(audio_tools::I2SConfig, int, int)':
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h:146:28: warning: narrowing conversion of '(eps32_i2s_sample_rate_type)cfg.audio_tools::I2SConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate' from 'eps32_i2s_sample_rate_type' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
             .sample_rate = (eps32_i2s_sample_rate_type)cfg.sample_rate,
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h:181:9: error: 'i2s_pin_config_t' has no non-static data member named 'mck_io_num'
         };
         ^
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioPWM/PWMAudioESP32.h:4,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioPWM/AudioPWM.h:4,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:19,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:9:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioPWM/PWMAudioBase.h: At global scope:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioPWM/PWMAudioBase.h:93:14: warning: inline variables are only available with -std=c++17 or -std=gnu++17
 } INLINE_VAR default_config;
              ^~~~~~~~~~~~~~
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudio.h:3,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:20,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:9:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h: In member function 'bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)':
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h:205:39: warning: narrowing conversion of '(eps32_i2s_sample_rate_type)cfg.audio_tools::AnalogConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate' from 'eps32_i2s_sample_rate_type' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
             .sample_rate            = (eps32_i2s_sample_rate_type)cfg.sample_rate,
[... cut ...]
In file included from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:10:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioKit.h: In member function 'virtual void audio_tools::AudioKitStream::setAudioInfo(audio_tools::AudioBaseInfo)':
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioKit.h:274:11: error: 'class AudioKit' has no member named 'setSampleRate'
       kit.setSampleRate(cfg.toSampleRate());
```

pschatzmann, 2022-11-18:
You still have the old version of the AudioKit library...

Dzsonzii, 2022-11-18:
I deleted, downloaded, and installed both audiokit, and audio-tools, I still get these.
```
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SStream.h:9,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:18,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:9:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h: In member function 'bool audio_tools::I2SBase::begin(audio_tools::I2SConfig, int, int)':
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h:146:28: warning: narrowing conversion of '(eps32_i2s_sample_rate_type)cfg.audio_tools::I2SConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate' from 'eps32_i2s_sample_rate_type' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
             .sample_rate = (eps32_i2s_sample_rate_type)cfg.sample_rate,
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h:181:9: error: 'i2s_pin_config_t' has no non-static data member named 'mck_io_num'
         };
         ^
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudio.h:3,
                 from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:20,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:9:
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h: In member function 'bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)':
D:\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h:205:39: warning: narrowing conversion of '(eps32_i2s_sample_rate_type)cfg.audio_tools::AnalogConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate' from 'eps32_i2s_sample_rate_type' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
             .sample_rate            = (eps32_i2s_sample_rate_type)cfg.sample_rate,
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from D:\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioKit.h:4,
                 from D:\Arduino\libraries\arduino-audio-tools-main\examples\examples-audiokit\basic-a2dp-audiokit\basic-a2dp-audiokit.ino:10:
D:\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h: In member function 'i2s_config_t AudioKitConfig::i2sConfig()':
D:\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h:129:34: warning: narrowing conversion of '((AudioKitConfig*)this)->AudioKitConfig::sampleRate()' from 'uint32_t' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
         .sample_rate = sampleRate(),
                        ~~~~~~~~~~^~
D:\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h: In member function 'i2s_pin_config_t AudioKitConfig::i2sPins()':
D:\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h:149:14: error: 'struct i2s_pin_config_t' has no member named 'mck_io_num'; did you mean 'bck_io_num'?
       result.mck_io_num = (gpio_num_t)mclk_gpio; //mclk_gpio;
              ^~~~~~~~~~
              bck_io_num

```

pschatzmann, 2022-11-18:
I think you must be using a very old ESP32 Arduino version. If you check in the Board Manager: the latest version is 2.0.5!

Dzsonzii, 2022-11-18:
> It's getting better, but I think you must be using a very old ESP32 Arduino version. If you check in the Board Manager: the latest version is 2.0.5!
> 
> I am supporting 1.0.6 and the latest 2 releases (2.0.4 and 2.0.5)...

jup, that was the final issue. Thanks for everything!

closed 2022-11-18 by Dzsonzii

pschatzmann, 2022-11-18:
Just out of curiosity: what release did cause this issue?