pschatzmann/arduino-audio-tools #472: Unable to compile using latest version

andrewwakeling, 2022-11-24:
Appears to be some issue with `BaseBuffer`.

```
In file included from .pio/libdeps/esp32dev/audio-tools/src/AudioHttp/AudioHttp.h:3,
                 from .pio/libdeps/esp32dev/audio-tools/src/AudioTools.h:30,
                 from .pio/libdeps/esp32dev/audio-tools/src/AudioLibs/AudioKit.h:3,
                 from src/main.cpp:3:
.pio/libdeps/esp32dev/audio-tools/src/AudioHttp/URLStreamESP32.h:115:97: error: use of deleted function 'audio_tools::NBuffer<unsigned char>::NBuffer(const audio_tools::NBuffer<unsigned char>&)'
         NBuffer<uint8_t> buffers = NBuffer<uint8_t>(DEFAULT_BUFFER_SIZE, URL_STREAM_BUFFER_COUNT);       
                                                                                                 ^        
In file included from .pio/libdeps/esp32dev/audio-tools/src/AudioTools.h:13,
                 from .pio/libdeps/esp32dev/audio-tools/src/AudioLibs/AudioKit.h:3,
                 from src/main.cpp:3:
.pio/libdeps/esp32dev/audio-tools/src/AudioTools/Buffers.h:314:7: note: 'audio_tools::NBuffer<unsigned char>::NBuffer(const audio_tools::NBuffer<unsigned char>&)' is implicitly deleted because the default definition would be ill-formed:
 class NBuffer : public BaseBuffer<T> {
       ^~~~~~~
Compiling .pio\build\esp32dev\lib81b\libopus\opus-1.3.1\silk\decode_frame.c.o
.pio/libdeps/esp32dev/audio-tools/src/AudioTools/Buffers.h:314:7: error: use of deleted function 'audio_tools::BaseBuffer<T>::BaseBuffer(const audio_tools::BaseBuffer<T>&) [with T = unsigned char]'
.pio/libdeps/esp32dev/audio-tools/src/AudioTools/Buffers.h:26:3: note: declared here
   BaseBuffer(BaseBuffer const&) = delete;
```

andrewwakeling, 2022-11-24:
I'm building in PlatformIO.

I can reproduce it with just this code:

```
#include <Arduino.h>
#include "AudioLibs/AudioKit.h"

void setup() {

}

void loop() {

}
```

I'm using the latest version of `arduino-audiokit` and `arduino-audio-tools`.

andrewwakeling, 2022-11-24:
It looks as though some functions were explicitly disabled/deleted via this commit:
https://github.com/pschatzmann/arduino-audio-tools/commit/7b22542f9cbed476a283011d9c56a54618432078

pschatzmann, 2022-11-24:
I don't think that you can include AudioLibs/AudioKit.h w/o including #include "AudioTools.h" first.
```
#include <Arduino.h>
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

void setup() {

}

void loop() {

}

Library Manager: Installing git+https://github.com/pschatzmann/arduino-audio-tools
git version 2.38.1
Cloning into '/Users/pschatzmann/.platformio/.cache/tmp/pkg-installing-j1qv7ats'...
remote: Enumerating objects: 2874, done.
remote: Counting objects: 100% (2874/2874), done.
remote: Compressing objects: 100% (2045/2045), done.
remote: Total 2874 (delta 1554), reused 1584 (delta 759), pack-reused 0
Receiving objects: 100% (2874/2874), 57.07 MiB | 1.84 MiB/s, done.
Resolving deltas: 100% (1554/1554), done.
Updating files: 100% (2562/2562), done.
Library Manager: audio-tools@0.9.3+sha.b5cc8de has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing git+https://github.com/pschatzmann/arduino-libhelix
git version 2.38.1
Cloning into '/Users/pschatzmann/.platformio/.cache/tmp/pkg-installing-iggnpi86'...
remote: Enumerating objects: 212, done.
remote: Counting objects: 100% (212/212), done.
remote: Compressing objects: 100% (161/161), done.
remote: Total 212 (delta 76), reused 141 (delta 50), pack-reused 0
Receiving objects: 100% (212/212), 2.24 MiB | 780.00 KiB/s, done.
Resolving deltas: 100% (76/76), done.
Library Manager: libhelix@0.1.0+sha.ee1023c has been installed!
Library Manager: Installing git+https://github.com/pschatzmann/arduino-audiokit
git version 2.38.1
Cloning into '/Users/pschatzmann/.platformio/.cache/tmp/pkg-installing-77ie6hy3'...
remote: Enumerating objects: 314, done.
remote: Counting objects: 100% (314/314), done.
remote: Compressing objects: 100% (232/232), done.
remote: Total 314 (delta 159), reused 143 (delta 78), pack-reused 0
Receiving objects: 100% (314/314), 380.56 KiB | 932.00 KiB/s, done.
Resolving deltas: 100% (159/159), done.
Library Manager: audiokit@0.6.1+sha.1f733c6 has been installed!
Library Manager: Installing git+https://github.com/greiman/SdFat
git version 2.38.1
Cloning into '/Users/pschatzmann/.platformio/.cache/tmp/pkg-installing-zn_p7_it'...
remote: Enumerating objects: 304, done.
remote: Counting objects: 100% (304/304), done.
remote: Compressing objects: 100% (218/218), done.
remote: Total 304 (delta 66), reused 162 (delta 30), pack-reused 0
Receiving objects: 100% (304/304), 2.79 MiB | 1.39 MiB/s, done.
Resolving deltas: 100% (66/66), done.
Library Manager: SdFat@2.2.0+sha.628effa has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.2.0+sha.03254d4) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 51 compatible libraries
Scanning dependencies...
Dependency Graph
[... 31 lines cut ...]
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_driver/es7210/es7210.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_driver/es7243/es7243.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_driver/es8311/es8311.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_driver/es8374/es8374.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_driver/es8388/es8388.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_driver/tas5805m/tas5805m.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_hal/audio_hal.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_hal/audio_mem.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_hal/audio_mutex.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_hal/i2c_bus.c.o
Compiling .pio/build/esp32dev/lib9d5/audiokit/audio_hal/i2c_wire.cpp.o
Archiving .pio/build/esp32dev/lib9d5/libaudiokit.a
Indexing .pio/build/esp32dev/lib9d5/libaudiokit.a
Linking .pio/build/esp32dev/firmware.elf
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.6% (used 25056 bytes from 327680 bytes)
Flash: [====      ]  42.6% (used 557973 bytes from 1310720 bytes)
Building .pio/build/esp32dev/firmware.bin
esptool.py v4.2.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
```

andrewwakeling, 2022-11-24:
I'm still getting the same errors after manually deleting `build` and `libdeps` and compiling your example code. (See output below).

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 2.0.0
|-- audiokit @ 0.6.1+sha.1f733c6
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- audio-tools @ 0.9.3+sha.b5cc8de
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- audiokit @ 0.6.1+sha.1f733c6
|   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Building .pio\build\esp32dev\bootloader.bin
Generating partitions .pio\build\esp32dev\partitions.bin
esptool.py v4.2.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32dev\libd16\Wire\Wire.cpp.o
Compiling .pio\build\esp32dev\libdfb\SPI\SPI.cpp.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\ai_thinker_ac101\board.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\ai_thinker_ac101\board_pins_config.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\ai_thinker_es8388_2957\board.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\ai_thinker_es8388_2957\board_pins_config.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\ai_thinker_es8388_3478\board.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\ai_thinker_es8388_3478\board_pins_config.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\esp32_s2_kaluga_1_v1_2\board.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\esp32_s2_kaluga_1_v1_2\board_pins_config.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\generic_es8388\board.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\generic_es8388\board_pins_config.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\lyrat_mini_v1_1\board.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\lyrat_mini_v1_1\board_pins_config.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\lyrat_v4_2\board.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\lyrat_v4_2\board_pins_config.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\lyrat_v4_3\board.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_board\lyrat_v4_3\board_pins_config.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_driver\ac101\ac101.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_driver\es7148\es7148.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_driver\es7210\es7210.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_driver\es7243\es7243.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_driver\es8311\es8311.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_driver\es8374\es8374.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_driver\es8388\es8388.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_driver\tas5805m\tas5805m.c.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_hal\audio_gpio.cpp.o
Compiling .pio\build\esp32dev\libe46\audiokit\audio_hal\audio_hal.c.o
In file included from .pio/libdeps/esp32dev/audio-tools/src/AudioTools.h:10,
                 from src/main.cpp:2:
.pio/libdeps/esp32dev/audio-tools/src/AudioConfig.h:195: warning: "URL_STREAM_PRIORITY" redefined
 #define URL_STREAM_PRIORITY 2

[... 42 lines cut ...]
                 from src/main.cpp:2:
.pio/libdeps/esp32dev/audio-tools/src/AudioTools/Buffers.h:314:7: note: 'audio_tools::NBuffer<unsigned char>::NBuffer(const audio_tools::NBuffer<unsigned char>&)' is implicitly deleted because the default definition would be ill-formed:
 class NBuffer : public BaseBuffer<T> {
       ^~~~~~~
.pio/libdeps/esp32dev/audio-tools/src/AudioTools/Buffers.h:314:7: error: use of deleted function 'audio_tools::BaseBuffer<T>::BaseBuffer(const audio_tools::BaseBuffer<T>&) [with T = unsigned char]'
.pio/libdeps/esp32dev/audio-tools/src/AudioTools/Buffers.h:26:3: note: declared here
   BaseBuffer(BaseBuffer const&) = delete;
   ^~~~~~~~~~
Compiling .pio\build\esp32dev\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\USB.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\USBCDC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\USBMSC.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
```

andrewwakeling, 2022-11-24:
Hold up. I think I've realized that I have `AudioConfigLocal.h` which could be creating problems.

pschatzmann, 2022-11-24:
It seems that you are using the USE_URLSTREAM_TASK option, which is usually not recommended.
I just committed a correction...

andrewwakeling, 2022-11-24:
So, the source of this problem was `#define USE_URLSTREAM_TASK`.

andrewwakeling, 2022-11-24:
> It seems that you were using the USE_URLSTREAM_TASK option, which is usually not recommended. I just committed a correction...

Thank you! I can disable it for now.

pschatzmann, 2022-11-24:
yes, but with the latest correction this will work as well

pschatzmann, 2022-11-24:
by the way: AudioConfigLocal.h is a concept which should be relevant for Arduino only. In PlatformIO it is much more flexible to define this as build_flags in platform.ini

andrewwakeling, 2022-11-24:
> by the way: AudioConfigLocal.h is a concept which should be relevant for Arduino only. In PlatformIO it is much more flexible to define this as build_flags in platform.ini

Thank you! I am using build_flags for `AUDIOKIT_BOARD` and `AI_THINKER_ES8388_VOLUME_HACK` but forgotten that I'd had these before migrating from Arduino IDE.

closed 2022-11-24 by andrewwakeling