pschatzmann/arduino-audio-tools #482: Can't compile player-sd-audiokit and any other audiokit examples

frgithub35, 2022-11-29:
Hi, 
I'd like to compile examples for AI AudioKit, but on my case, none of this example can be compiled.
I've checked for prerequisites librairies but there's still compilation errors that are beginning with : 

_In file included from F:\ArduinoIDE_1.8.57.0_x86\portable\sketchbook\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimerDef.h:2:0,
                 from F:\ArduinoIDE_1.8.57.0_x86\portable\sketchbook\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimerESP32.h:4,
                 from F:\ArduinoIDE_1.8.57.0_x86\portable\sketchbook\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimer.h:3,
                 from F:\ArduinoIDE_1.8.57.0_x86\portable\sketchbook\libraries\arduino-audio-tools-main\src/AudioTools.h:11,
                 from F:\ESP32-player-sd-audiokit\ESP32-player-sd-audiokit.ino:10:
F:\ArduinoIDE_1.8.57.0_x86\portable\sketchbook\libraries\arduino-audio-tools-main\src/AudioTools/AudioTypes.h:13:38: error: 'audio_tools::RxTxModeNames' declared as an 'inline' variable
 INLINE_VAR const char* RxTxModeNames[]={"UNDEFINED_MODE","TX_MODE","RX_MODE","RXTX_MODE" };
                                      ^
F:\ArduinoIDE_1.8.57.0_x86\portable\sketchbook\libraries\arduino-audio-tools-main\src/AudioTools/AudioTypes.h:96:24: error: 'audio_tools::mime_pcm' declared as an 'inline' variable
 INLINE_VAR const char* mime_pcm = "audio/pcm";
                        ^_

Is there compilation options for this card, on anything else that I missed
Thanks for your help

pschatzmann, 2022-11-29:
How can I reproduce your issue ? I tried this example in Arduino and did not get any errors...
If you double check in the AudioConfig.h the INLINE_VAR is defined at the bottom of the code.

frgithub35, 2022-11-30:
Hi,
I've made a complete reinstall of Arduino IDE on a new PC and it's OK now.
Maybe it doesn't work because the first is on portable Arduino with a lot of librairies...

I think that this issue can be close.
Thanks a lot for your help.

closed 2022-11-30 by frgithub35