pschatzmann/arduino-audio-tools #487: Audio Player Looping Single File

pauldupuyjr, 2022-12-02:
Using the AudioPlayer, I want to seamlessly loop playing a single file. I've tried using `player.setAutoNext(false) ` in setup and versions of the following in the main loop:
```
    player.copy();
    if (player.isActive() == false) {
        player.setAudioSource(source);
        player.begin(0, false);
    }
```
However, player doesn't seem to honor the autonext setting. It plays the file once, then attempts to move to the nextStream:

```
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 2048 bytes - in 1 hops
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 2048 bytes - in 1 hops
[I] AudioCopy.h : 134 - StreamCopy::copy 290 -> 290 -> 580 bytes - in 1 hops
[W] AudioPlayer.h : 316 - -> timeout - moving to next stream
[I] AudioSourceLittleFS.h : 40 - nextStream: 1
[I] AudioSourceLittleFS.h : 45 - selectStream: 1
[I] AudioEncoded.h : 324 - void audio_tools::EncodedAudioStream::end()
[I] MetaDataID3.h : 550 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 544 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSourceLittleFS.h : 45 - selectStream: 0
[W] AudioPlayer.h : 144 - -> begin: no data found
[I] MetaDataID3.h : 544 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSourceLittleFS.h : 45 - selectStream: 0
[W] AudioPlayer.h : 144 - -> begin: no data found
```
How can I play a single file in a seamless loop?

pschatzmann, 2022-12-02:
I don't see this as standard feature. However you should be able to easily implement this yourself by creating a subclass your AudioSource and overwriting the next() method. If next returns the same file it will be repeated, this way you can also implement your specific logic which decides which files should be repeated and how often...

pauldupuyjr, 2022-12-02:
Looping audio is has a long history ( [Wikipedia: Loop (music)](https://en.wikipedia.org/wiki/Loop_(music))) and I'd wager others might find it useful. 

Regardless, thank you very much for the workaround Phil. Your library is a gift of marvelous software engineering and I'm very appreciative of your rapid response in addressing issues.

pauldupuyjr, 2022-12-03:
Trying to `seek(0)` on the File* causes a crash but doing it with the file directly in a modified AudioSource class works.
```
  virtual Stream *nextStream(int offset = 1) override {
    file.seek(0);
    return file ? &file : nullptr;
  }
```

However, there's some noise between loops when resetting in this way -- a buzzy click. Is there some way to have a seamless loop without extraneous noise or delay?

pschatzmann, 2022-12-03:
I suggest that you have a look at your audio file that creates the click. Audio files start usually with a low amplitue around 0. If they end with a very high or very low value and you move back to the beginning you can hear the difference as click.

Therefore audio is usually fading out at the end: In Audacity e.g. you can apply a Fade out or Studio Fade Out.
If the delay between the loop is too long, you can adjust it by calling setTimeoutAutoNext(milliscends) on the AudioSource.

pauldupuyjr, 2022-12-03:
Reducing timeout_auto_next_value from the default 500 did the trick. Thanks Phil!

closed 2022-12-05 by pauldupuyjr

timmcarson, 2023-04-08:
@pauldupuyjr 

Would you mind sharing your code?  I would like to implement looping in my project.