pschatzmann/arduino-audio-tools #491: File name sd card

gnpaone, 2022-12-04:
I am actually trying to display the file name of mp3 in sd card on the oled but I couldn't find any function for getting file name, please help

pschatzmann, 2022-12-04:
Just look the documentation of the Audio Source!

closed 2022-12-04 by pschatzmann

gnpaone, 2022-12-04:
🥵 tried various ways but couldn't make it work, can you please help like for this example: https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/basic-a2dp-audiokit and this example: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/player-sdfat-audiokit/player-sdfat-audiokit.ino how you would implement I can manage I think

pschatzmann, 2022-12-04:
You should not try, but read the [Class Documentation](https://pschatzmann.github.io/arduino-audio-tools/html/namespaceaudio__tools.html) !

gnpaone, 2022-12-04:
> You should not try, but look up the relevant function from the [Class Documentation](https://pschatzmann.github.io/arduino-audio-tools/html/namespaceaudio__tools.html) ! There you can find how to get the file name...

Yaah, there's some under sddirect and sdindex but nothing under audiosourcesdfat nor a2dpsink 🤧😵‍💫, please help

pschatzmann, 2022-12-04:
You must be blind!
const char *toStr  provides the actual file name

gnpaone, 2022-12-06:
Umm sorry 😳
Can you please tell for a2dpsink too? 😳😳😳
Thanks by the way, toStr() works for sdfat