pschatzmann/arduino-audio-tools #497: Compilation error when using Esp32-s3

Thecactusman0, 2022-12-06:
When trying to compile basic-a2dp-i2s for the Esp32-s3 dev board I get the following errors, however if I instead select Esp32-wrover from the boards manager without changing anything it compiles without issue. Is this an issue specific to the board or is there an issue on my end?

`C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp: In member function 'virtual void BluetoothA2DPSink::init_i2s()':
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:210:31: error: 'I2S_MODE_DAC_BUILT_IN' was not declared in this scope
         if (i2s_config.mode & I2S_MODE_DAC_BUILT_IN) {
                               ^~~~~~~~~~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:210:31: note: suggested alternative: 'I2S_MODE_MASTER'
         if (i2s_config.mode & I2S_MODE_DAC_BUILT_IN) {
                               ^~~~~~~~~~~~~~~~~~~~~
                               I2S_MODE_MASTER
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:214:34: error: 'I2S_DAC_CHANNEL_BOTH_EN' was not declared in this scope
             if (i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN)!= ESP_OK){
                                  ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:214:34: note: suggested alternative: 'I2S_CHANNEL_STEREO'
             if (i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN)!= ESP_OK){
                                  ^~~~~~~~~~~~~~~~~~~~~~~
                                  I2S_CHANNEL_STEREO
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:214:17: error: 'i2s_set_dac_mode' was not declared in this scope
             if (i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN)!= ESP_OK){
                 ^~~~~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:214:17: note: suggested alternative: 'i2s_set_clk'
             if (i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN)!= ESP_OK){
                 ^~~~~~~~~~~~~~~~
                 i2s_set_clk
In file included from C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/esp_rom/include/esp32s3/rom/ets_sys.h:19,
                 from C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:114,
                 from C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPCommon.h:25,
                 from C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.h:17,
                 from C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:16:
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp: In member function 'virtual esp_err_t BluetoothA2DPSink::i2s_mclk_pin_select(uint8_t)':
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:233:53: error: 'FUNC_GPIO0_CLK_OUT1' was not declared in this scope
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
                                                     ^~~~~~~~~~~~~~~~~~~
C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:136:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:233:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
             ^~~~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:233:53: note: suggested alternative: 'FUNC_GPIO20_CLK_OUT1'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
                                                     ^~~~~~~~~~~~~~~~~~~
C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:136:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:233:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
             ^~~~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:237:53: error: 'FUNC_U0TXD_CLK_OUT3' was not declared in this scope
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
                                                     ^~~~~~~~~~~~~~~~~~~
C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:136:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:237:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
             ^~~~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:237:53: note: suggested alternative: 'FUNC_U0TXD_CLK_OUT1'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
                                                     ^~~~~~~~~~~~~~~~~~~
C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:136:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:237:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
             ^~~~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp: In member function 'size_t BluetoothA2DPSink::i2s_write_data(const uint8_t*, size_t)':
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:1227:33: error: 'I2S_MODE_DAC_BUILT_IN' was not declared in this scope
     if (this->i2s_config.mode & I2S_MODE_DAC_BUILT_IN) {
                                 ^~~~~~~~~~~~~~~~~~~~~
C:\Users\nic\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:1227:33: note: suggested alternative: 'I2S_MODE_MASTER'
     if (this->i2s_config.mode & I2S_MODE_DAC_BUILT_IN) {
                                 ^~~~~~~~~~~~~~~~~~~~~
                                 I2S_MODE_MASTER
`

pschatzmann, 2022-12-06:
Just read the documentation:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Support-for-esp32-c3%3F

closed 2022-12-06 by pschatzmann

Thecactusman0, 2022-12-06:
My apologies I was looking for references to the S3 I should have been more thorough

pschatzmann, 2022-12-06:
Unfortunately I did not find a reliable way to identify the processor and issue an error on compile time...