pschatzmann/arduino-audio-tools #510: Mp3 stream to I2S Stutter, pauses, noises etc.

alexliit, 2022-12-09:
Hello!

I have played a bit with the provided example (streams-url_mp3-i2s.ino) but the audio output suffer from stutters, noises, pauses etc.. randomly: it may play perfectly for 1 or 2 minutes, then started to make some noise or stops for a second...restart to play ok for 30" then noise ...etc. 
No warning seen on serial output 
`AudioLogger::instance().begin(Serial, AudioLogger::Warning);  `
changed also buffer size without any effect on the issue
`URLStream url("SIK","WirelessZone10",4048);`
also changed the source stream
`url.begin("http://direct.fipradio.fr/live/fip-midfi.mp3","audio/mp3");`
Also changed some different power sources (USB powerbank, PC USB..).

Any hint?

May be it is some chip issues (ESP32 or I2S MAX98357A)?

Below the full code.

Thanks!
------
```
`/**
 * @file streams-url_mp3-i2s.ino
 * @author Phil Schatzmann
 * @brief decode MP3 stream from url and output it on I2S
 * @version 0.1
 * @date 2021-96-25
 * 
 * @copyright Copyright (c) 2021
 */

// install https://github.com/pschatzmann/arduino-libhelix.git

#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

URLStream url("SIK","WirelessZone10",4048);
I2SStream i2s; // final output of decoded stream
EncodedAudioStream dec(&i2s, new MP3DecoderHelix()); // Decoding stream
StreamCopy copier(dec, url); // copy url to decoder

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);  

  // setup i2s
  auto config = i2s.defaultConfig(TX_MODE);
  // you could define e.g your pins and change other settings
  config.pin_ws=13;
  config.pin_bck=12;
  config.pin_data=15;

  LogarithmicVolumeControl lvc(0.5);

  //config.mode = I2S_STD_FORMAT;
  i2s.begin(config);

  // setup I2S based on sampling rate provided by decoder
  dec.setNotifyAudioChange(i2s);
  dec.begin();

// mp3 radio
  url.begin("http://stream.srg-ssr.ch/m/rsj/mp3_128","audio/mp3");
  //url.begin("http://direct.fipradio.fr/live/fip-midfi.mp3","audio/mp3");
  

}

void loop(){
  copier.copy();
}`

pschatzmann, 2022-12-09:
I tried your french URL http://direct.fipradio.fr/live/fip-midfi.mp3" and for me it is working perfectly:
- To test if your output is working properly use[streams-generator-i2s](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-generator-i2s)
- Make sure that you have a proper wifi/network connection you can send the output to a [MeasuringStream](https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_measuring_stream.html) e.g. ```MeasuringStream out(50, &Serial);```

I am getting the following values over my phone network
```
18:59:02.541 -> ==> Bytes per second: 75000
18:59:03.260 -> ==> Bytes per second: 322000
18:59:04.013 -> ==> Bytes per second: 323000
18:59:04.725 -> ==> Bytes per second: 322000
18:59:05.467 -> ==> Bytes per second: 322000
18:59:06.189 -> ==> Bytes per second: 323000
18:59:06.908 -> ==> Bytes per second: 322000
18:59:07.656 -> ==> Bytes per second: 321000
18:59:08.370 -> ==> Bytes per second: 322000
18:59:09.088 -> ==> Bytes per second: 323000
18:59:09.835 -> ==> Bytes per second: 322000
18:59:10.543 -> ==> Bytes per second: 322000
18:59:11.586 -> ==> Bytes per second: 228000
18:59:12.906 -> ==> Bytes per second: 176000
18:59:14.052 -> ==> Bytes per second: 208000
18:59:15.270 -> ==> Bytes per second: 191000
18:59:16.491 -> ==> Bytes per second: 191000
18:59:17.810 -> ==> Bytes per second: 177000
18:59:19.067 -> ==> Bytes per second: 191000
18:59:20.192 -> ==> Bytes per second: 208000
18:59:21.397 -> ==> Bytes per second: 191000
18:59:22.523 -> ==> Bytes per second: 208000
18:59:23.841 -> ==> Bytes per second: 177000
18:59:25.201 -> ==> Bytes per second: 176000
18:59:26.308 -> ==> Bytes per second: 208000
18:59:27.544 -> ==> Bytes per second: 191000
18:59:28.767 -> ==> Bytes per second: 191000
18:59:29.979 -> ==> Bytes per second: 191000
18:59:31.203 -> ==> Bytes per second: 191000
18:59:32.459 -> ==> Bytes per second: 191000
18:59:33.689 -> ==> Bytes per second: 191000
18:59:34.976 -> ==> Bytes per second: 182000
18:59:36.190 -> ==> Bytes per second: 191000
18:59:37.403 -> ==> Bytes per second: 191000
18:59:38.634 -> ==> Bytes per second: 191000
18:59:39.752 -> ==> Bytes per second: 208000
18:59:41.207 -> ==> Bytes per second: 164000
18:59:42.321 -> ==> Bytes per second: 208000
18:59:43.446 -> ==> Bytes per second: 208000
18:59:44.761 -> ==> Bytes per second: 176000
18:59:45.985 -> ==> Bytes per second: 191000
18:59:47.111 -> ==> Bytes per second: 208000
18:59:48.338 -> ==> Bytes per second: 191000
18:59:49.598 -> ==> Bytes per second: 191000
18:59:50.799 -> ==> Bytes per second: 191000
18:59:51.943 -> ==> Bytes per second: 208000
18:59:53.246 -> ==> Bytes per second: 176000
18:59:54.487 -> ==> Bytes per second: 191000
18:59:55.708 -> ==> Bytes per second: 191000

```

alexliit, 2022-12-14:
Hello

Thanks for the hints! 

After few tweaks I managed to fix a lot the issue.
Basically seems that the pins used to control the I2S amplifiers were not good  for a TTGO Display EP32; finally this configuration worked very well
```
  //config.pin_ws=13;
  config.pin_bck=17;
  //config.pin_data=15;

```
on the same skech posted above.

But nevertheless some very few glitches still came out. This can be WIFI issues, since it happend when the traffic rised on the home net. 

There is some remediation to this, for instance buffer increase?

Thanks

pschatzmann, 2022-12-15:
I never had any issues with the URLStream class. You could try to use URLStreamBuffered which has it's own task management and buffering...

closed 2023-01-09 by pschatzmann