pschatzmann/arduino-audio-tools #513: Compile fails when audio toolsis included

vvstubbs, 2022-12-10:
In file included from /home/vstubbs/Development/Arduino_Dev/arduino-sketchbook/libraries/arduino-audio-tools-main/src/AudioTools.h:55,
                 from /home/vstubbs/Development/Arduino_Dev/arduino-sketchbook/ESP32-Wrover-Arduino-Nano-project_phase-2b/ESP32-Wrover-Arduino-Nano-project_phase-2b.ino:45:
/home/vstubbs/Development/Arduino_Dev/arduino-sketchbook/libraries/arduino-audio-tools-main/src/AudioTools/SynchronizedBuffers.h:8:12: fatal error: FreeRTOS.h: No such file or directory
 #  include <FreeRTOS.h>
            ^~~~~~~~~~~~
compilation terminated.
Not sure where I'm supposed to find FreeRTOS.h

pschatzmann, 2022-12-10:
What Board did you select ? For ESP32 FreeRTOS is part of Arduino...

pschatzmann, 2022-12-10:
I could not reproduce your issue in Arduino, but in PlatformIO: I just committed some corrections...

vvstubbs, 2022-12-11:
Board: ESP32 WROVER KIT (all version) on /dev/ttyUSB1

/*
ESP32-Wrover-Arduino-Nano-project_phase-1a: This project explores the limits of the
FREENOVE ESP32-WROVER-DEV module and the BM004 compass module. This code is based on the
BM004_Arduino_compass_tilt_compensated:  This program reads the magnetometer and
accelermoter registers from ST Micro's LSM303DLHC.  The register values are used to generate 
a tilt compensated heading value.  NOTE:  placing the compass near metallic objects can impact
readings. And additional code provided by the Freenove_Ultimate_Starter_Kit_for_ESP32-master
libraries and code snippets as in the 'C' folder included. Also employing a SSD1306 OLED 
display module from DIYmail, using the Adafruit_GFX and Adafruit_SSD1306 libraries.

Schematics associated with the BM004 moudle may be used for hardware wiring information.
see www.solutions-cubed.com for additional information.

OLEDMeter was written to utilize any 128x64 display. I have only seen marginal attempts to
animate meters and I hope this one will set a standard. Please feel free to modify and share
this code for any 128x64 LCD or OLED. OLEDMeter sketch was written for use with I2C SH1106.
This code must be modified to work with other display devices.
  
Working portion of code was taken from Adafruit Example Sound Level Sketch for the
Adafruit Microphone Amplifier
https://learn.adafruit.com/adafruit-microphone-amplifier-breakout/measuring-sound-levels
  
Remaining code was written by Greg Stievenart with no claim to or any images or information 
provided in this code. Freely published May 26, 2016.
  
Software to convert background mask to 128x64 at: http://www.ablab.in/image2glcd-software/
  
IMPORTANT: Sound source must be grounded to the Arduino or other MCU's to work. Usually the
base sleeve contact on TRS or TRRS connector is the ground.

Additional code snippets and libraries used are documented with in the source code below.
*/

#include <Math.h>
#include <WiFi.h>
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>                               // For SSD1306 128x32 and 128x64 0.91 and 0.96 OLED displays
#include <Adafruit_SH110X.h>                                // For SH1106 128x64 1.3 OLED displays
#include <Cardinal.h>                                       // Cardinal library (https://github.com/DaAwesomeP/arduino-cardinal)

#include <Arduino.h>
#include <BluetoothA2DPSource.h>
#include <AudioTools.h>                                 // This causes the error

#define SCREEN_WIDTH 128                                    // OLED display width, in pixels
#define SCREEN_HEIGHT 64                                    // OLED display height, in pixels

// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
#define OLED_RESET   4                                      // Reset pin # (or -1 if sharing Arduino reset pin)
#define OLED1_RESET -1

/* Uncomment the initialize the I2C address , uncomment only one, If you get a totally blank screen try the other*/
#define i2c_Address 0x3D                                    //initialize with the I2C addr 0x3C for 1.3 OLED
//#define i2c_Address 0x3d                                  //initialize with the I2C addr 0x3D Typically Adafruit OLED's
#define SCREEN_ADDRESS 0x3C                                 //< See datasheet for Address; 0x3D for 128x64, 0x3C for 128x32

Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
//Adafruit_SH1106G oled = Adafruit_SH1106G(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED1_RESET);
Adafruit_SH1106G oled(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED1_RESET);

float Heading;
float Pitch;
float Roll;

float Accx;
float Accy;
[... 588 lines cut ...]
  delay(1000);

}

void Scanner(){
  display.setCursor(0, 20);
  byte count = 0;
  Wire.begin();
  for (byte i = 8; i < 120; i++)
  {
    Wire.beginTransmission (i);                             // Begin I2C transmission Address (i)
    if (Wire.endTransmission () == 0)                       // Receive 0 = success (ACK response) 
    {
      display.print ("I2C: ");
      display.print (i, DEC);
      display.print (" (0x");
      display.print (i, HEX);                               // PCF8574 7 bit address
      display.println (")");
      count++;
    }
  }
  //display.print ("Found ");      
  //display.print (count, DEC);                               // numbers of devices
  //display.println (" device(s).");
}

vvstubbs, 2022-12-11:
Well what ever commits you did it has indeed found the FreeRTOS. and is in the midst of compiling. It's a long compile....

vvstubbs, 2022-12-11:
/home/vstubbs/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-g++ -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.1-1-gb8050b365e\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -I/home/vstubbs/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/config -I/home/vstubbs/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/newlib/platform_include -I/home/vstubbs/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/freertos/include -I/home/vstubbs/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/freertos/include/esp_additions/freertos -I/home/vstubbs/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/freertos/port/xtensa/include -I/home/vstubbs/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/freertos/include/esp_additions ....

As you can see it has found it and is still compiling

closed 2022-12-11 by vvstubbs

pschatzmann, 2022-12-11:
> I/home/vstubbs/.arduino15/packages/esp32/hardware/esp32/2.0.3

Is there any reason why you still use this old, obsolete release. 

Go to the Board Manager and type ESP32: The actual release is 2.0.5