pschatzmann/arduino-audio-tools #517: Compile Error NodeMCU-32 / Arduino 1.8.19 - pitch shifting test sketch

kabub17, 2022-12-11:
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioPrint.h:6:0,
                 from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioFilter/Equilizer.h:3,
                 from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:58,
                 from C:\Users\43677\Documents\Arduino\sketch_dec11a\sketch_dec11a.ino:1:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h:74:15: error: 'virtual int audio_tools::AudioStream::availableForWrite()' marked 'override', but does not override
   virtual int availableForWrite() override { return DEFAULT_BUFFER_SIZE; }
               ^
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h: In member function 'virtual int audio_tools::AudioStreamWrapper::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h:147:54: error: 'class Stream' has no member named 'availableForWrite'
   virtual int availableForWrite() { return p_stream->availableForWrite(); }
                                                      ^
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h: In member function 'virtual int audio_tools::ConverterStream<T, ConverterT>::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h:910:60: error: 'class Stream' has no member named 'availableForWrite'
         virtual int availableForWrite() { return p_stream->availableForWrite(); }
                                                            ^
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h: In member function 'virtual int audio_tools::VolumeStream::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h:1064:47: error: 'class Print' has no member named 'availableForWrite'
             return p_out==nullptr? 0 : p_out->availableForWrite();
                                               ^
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h: In member function 'virtual int audio_tools::MeasuringStream::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h:1266:23: error: 'class Print' has no member named 'availableForWrite'
       return p_print->availableForWrite();
                       ^
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h: In member function 'virtual int audio_tools::FilteredStream<T, TF>::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h:1454:28: error: 'class Stream' has no member named 'availableForWrite'
           return p_stream->availableForWrite();
                            ^
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioFilter/Equilizer.h:3:0,
                 from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:58,
                 from C:\Users\43677\Documents\Arduino\sketch_dec11a\sketch_dec11a.ino:1:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioPrint.h: At global scope:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioPrint.h:32:21: error: 'virtual int audio_tools::AudioPrint::availableForWrite()' marked 'override', but does not override
         virtual int availableForWrite() override {
                     ^
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:58:0,
                 from C:\Users\43677\Documents\Arduino\sketch_dec11a\sketch_dec11a.ino:1:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioFilter/Equilizer.h: In member function 'virtual int audio_tools::Equilizer3Bands::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioFilter/Equilizer.h:109:29: error: 'class Print' has no member named 'availableForWrite'
             return p_print->availableForWrite();
                             ^
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:65:0,
                 from C:\Users\43677\Documents\Arduino\sketch_dec11a\sketch_dec11a.ino:1:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h: In member function 'virtual int audio_tools::ChannelFormatConverterStreamT<T>::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:71:43: error: 'class Print' has no member named 'availableForWrite'
           return 1.0f / factor * p_print->availableForWrite();
                                           ^
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:65:0,
                 from C:\Users\43677\Documents\Arduino\sketch_dec11a\sketch_dec11a.ino:1:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h: In member function 'virtual int audio_tools::NumberFormatConverterStreamT<TFrom, TTo>::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:329:27: error: 'class Print' has no member named 'availableForWrite'
           return p_print->availableForWrite();
                           ^
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h: In member function 'virtual int audio_tools::NumberFormatConverterStream::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:457:29: error: 'class Print' has no member named 'availableForWrite'
             return p_print->availableForWrite();
                             ^
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:67:0,
                 from C:\Users\43677\Documents\Arduino\sketch_dec11a\sketch_dec11a.ino:1:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/Resample.h: In member function 'int audio_tools::Resample<T>::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/Resample.h:76:58: error: 'class Print' has no member named 'availableForWrite'
         int availableForWrite() override { return p_out->availableForWrite(); }
                                                          ^
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:68:0,
                 from C:\Users\43677\Documents\Arduino\sketch_dec11a\sketch_dec11a.ino:1:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h: In member function 'size_t audio_tools::StreamCopyT<T>::copy()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:95:32: error: 'class Print' has no member named 'availableForWrite'
             int to_write = to->availableForWrite();
                                ^
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:69:0,
                 from C:\Users\43677\Documents\Arduino\sketch_dec11a\sketch_dec11a.ino:1:
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/AudioEncoded.h: In member function 'virtual int audio_tools::EncodedAudioPrint::availableForWrite()':
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/AudioEncoded.h:344:54: error: 'class Print' has no member named 'availableForWrite'
   int availableForWrite() override { return ptr_out->availableForWrite(); }
                                                      ^
In file included from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioHttp/AudioHttp.h:8:0,
                 from C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:75,
[... cut ...]
C:\Users\43677\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioHttp/URLStream.h:242:35: error: 'class WiFiClientSecure' has no member named 'setInsecure'
                     clientSecure->setInsecure();
                                   ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\43677\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board NodeMCU-32S.

pschatzmann, 2022-12-11:
I am not able to reproduce your issues. Go to the Board Manager and check that you use an actual version of the ESP32 implementation. The current version is 2.0.5!

kabub17, 2022-12-12:
THX for advice. Had two parallel installations of Arduino IDE with different versions. Made a clean installation with 2.0.3 – now it works.
Thank You
Wolfgang

closed 2022-12-12 by pschatzmann