pschatzmann/arduino-audio-tools #520: Error compiling example stream pwm on ESP32

AlbertoManske, 2022-12-12:
Hi, I'm having problem trying to compile the example to esp32, I'm using the latest arduino IDE and the latest version of esp32.
Erro msg:

C:\Users\alber\AppData\Local\Temp\arduino-sketch-CAA39A6BAE645F5021AC374169B62D89\sketch\streams-generator-pwm.ino.cpp.o: in function `audio_tools::defaultPWMAudioOutputCallback()':
c:\Users\alber\OneDrive\Documentos\Arduino\libraries\arduino-audio-tools-main\src/AudioPWM/PWMAudioESP32.h:152:(.iram1.40[_ZN11audio_tools29defaultPWMAudioOutputCallbackEv]+0x3): dangerous relocation: l32r: literal placed after use: .literal._ZN11audio_tools29defaultPWMAudioOutputCallbackEv
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

pschatzmann, 2022-12-12:
Thanks for letting me know: I am currently refactoring the classes.
I have committed the correction...

AlbertoManske, 2022-12-12:
Thanks friend, now working perfectly

pschatzmann, 2022-12-12:
I forgot to mention that you can also use AnalogAudioStream or I2SStream with signal_type PDM to get some analog output

AlbertoManske, 2022-12-12:
Yes, but in my case the pwm signal is needed, I am using it to generate a square wave modulated by the 60hz sine wave, to be able to use it in a voltage inverter

pschatzmann, 2022-12-12:
Oh, I see - so it has nothing to do with Audio then....

AlbertoManske, 2022-12-12:
Not this time, but I already used your lib together with the ADAU1701 processor, to be able to get the bluetooth i2s signal, I managed to make it work perfectly, but I had a certain amount of work to generate the M_clk through esp32 to synchronize the DSP