pschatzmann/arduino-audio-tools #521: URLStream does not name a type

VictorHAS, 2022-12-12:
Hi, I'm trying to run example "streams-url-file" I alread imported lib as zip on arduino ide, but its returning this:

`C:\Users\vherm\Documents\Arduino\streams-url-file\streams-url-file.ino:12:1: error: 'URLStream' does not name a type; did you mean 'Stream'?
 URLStream url(ethClient); // Music Stream
 ^~~~~~~~~
 Stream
C:\Users\vherm\Documents\Arduino\streams-url-file\streams-url-file.ino: In function 'void setup()':
C:\Users\vherm\Documents\Arduino\streams-url-file\streams-url-file.ino:23:122: error: no matching function for call to 'SPIClass::begin(int, int, int, int)'
     SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
                                                                                                                          ^
In file included from C:\Users\vherm\Documents\Arduino\streams-url-file\streams-url-file.ino:4:0:
C:\Users\vherm\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI\src/SPI.h:159:15: note: candidate: static void SPIClass::begin()
   static void begin();
               ^~~~~
C:\Users\vherm\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI\src/SPI.h:159:15: note:   candidate expects 0 arguments, 4 provided
C:\Users\vherm\Documents\Arduino\streams-url-file\streams-url-file.ino:32:5: error: 'url' was not declared in this scope
     url.begin("https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw");
     ^~~

exit status 1

Compilation error: 'URLStream' does not name a type; did you mean 'Stream'?`

I'm also trying to use Arduino Ethernet shield and URLStream ethernet client, It's possible?

pschatzmann, 2022-12-12:
I never tried, but in theory it might work. Here are the steps that you need to do
- in the AudioConfig.h add then entry #define USE_URL_ARDUINO best in the AVR section to activate the functionality.
- To the URLStream you need to pass the EthernetClient that you need to setup yourself in the sketch

VictorHAS, 2022-12-12:
USE_URL_ARDUINO added but now, new error: `In file included from c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools.h:61:0,
                 from C:\Users\vherm\Documents\Arduino\streams-url-file\streams-url-file.ino:2:
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:28:17: error: 'PWMConfig' does not name a type
         virtual PWMConfig defaultConfig() {
                 ^~~~~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:32:9: error: 'PWMConfig' does not name a type
         PWMConfig config() {
         ^~~~~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:53:59: error: 'PWMCallbackType' has not been declared
         bool begin(uint16_t sampleRate, uint8_t channels, PWMCallbackType cb) {
                                                           ^~~~~~~~~~~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:63:20: error: 'PWMConfig' has not been declared
         bool begin(PWMConfig config ) {
                    ^~~~~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:97:9: error: 'PWMConfig' does not name a type
         PWMConfig audio_config;
         ^~~~~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:98:9: error: 'PWMDriver' does not name a type
         PWMDriver driver; // platform specific driver
         ^~~~~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h: In destructor 'virtual audio_tools::PWMAudioStream::~PWMAudioStream()':
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:23:17: error: 'driver' was not declared in this scope
             if (driver.isTimerStarted()){
                 ^~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:23:17: note: suggested alternative: 'write'
             if (driver.isTimerStarted()){
                 ^~~~~~
                 write
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h: In member function 'virtual void audio_tools::PWMAudioStream::setAudioInfo(audio_tools::AudioBaseInfo)':
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:39:13: error: 'PWMConfig' was not declared in this scope
             PWMConfig cfg = audio_config;
             ^~~~~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:48:26: error: no matching function for call to 'audio_tools::PWMAudioStream::begin(audio_tools::AudioBaseInfo&)'
                 begin(cfg);
                          ^
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:53:14: note: candidate: bool audio_tools::PWMAudioStream::begin(uint16_t, uint8_t, int)
         bool begin(uint16_t sampleRate, uint8_t channels, PWMCallbackType cb) {
              ^~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:53:14: note:   candidate expects 3 arguments, 1 provided
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:63:14: note: candidate: bool audio_tools::PWMAudioStream::begin(int)
         bool begin(PWMConfig config ) {
              ^~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:63:14: note:   no known conversion for argument 1 from 'audio_tools::AudioBaseInfo' to 'int'
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:69:14: note: candidate: bool audio_tools::PWMAudioStream::begin()
         bool begin()  {
              ^~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:69:14: note:   candidate expects 0 arguments, 1 provided
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h: In member function 'bool audio_tools::PWMAudioStream::begin(uint16_t, uint8_t, int)':
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:55:13: error: 'audio_config' was not declared in this scope
             audio_config.channels = channels;
             ^~~~~~~~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:55:13: note: suggested alternative: 'audioInfo'
             audio_config.channels = channels;
             ^~~~~~~~~~~~
             audioInfo
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:57:13: error: 'driver' was not declared in this scope
             driver.setUserCallback(cb);
             ^~~~~~
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:57:13: note: suggested alternative: 'write'
             driver.setUserCallback(cb);
             ^~~~~~
             write
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h: In member function 'bool audio_tools::PWMAudioStream::begin(int)':
c:\Users\vherm\Documents\Arduino\libraries\arduino-audio-tools\src/AudioPWM/AudioPWM.h:65:19: error: 'class audio_tools::PWMAudioStream' has no member named 'audio_config'; did you mean 'audioInfo'?
             this->audio_config = config;
                   ^~~~~~~~~~~~
                   audioInfo
[... cut ...]
    // open music stream
    url.begin("https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw");

    // copy file
    file = SD.open("/audio-8000.raw", FILE_WRITE);
    file.seek(0); // overwirte from beginning
    copier.begin(file, url);
    copier.copyAll();
    file.close();

    file = SD.open("/audio-8000.raw");
    LOGI("file size: %d", file.size());
}

void loop() {
}
`
Arduino shield already have sd card slot, do I need to use SPI.begin with params? becausa, was throwing some errors too

pschatzmann, 2022-12-12:
What processor did you select ?

VictorHAS, 2022-12-12:
I tried with Arduino Uno and Arduino Ethernet

pschatzmann, 2022-12-12:
I committed some corrections: the compile errors are gone but I still have bad news: the sketch is just too big for the Uno!

VictorHAS, 2022-12-12:
Ouch, I'm just trying to get audio file from S3 and save on SD card, do you have any tip?

pschatzmann, 2022-12-12:
You will not like the answer: use an ESP32, it is cheap, fast and has a lot of memory...

VictorHAS, 2022-12-12:
hahahah I do have a esp32, I'll try it. Can I use the same lib for get audio from s3, right? I think you can now close the topic, compilation problem has been solved! Thank you!!

pschatzmann, 2022-12-12:
Never used s3, so I don't know. 
If you can access the file from your browser with a url, it should work...

VictorHAS, 2022-12-12:
If you want, I can submit as an example if it's not something like 'streams-url-file'

pschatzmann, 2022-12-12:
cool