pschatzmann/arduino-audio-tools #526: Allow configuration of AnalogDriverESP32 channel_format

aSemy, 2022-12-15:
Hi,

I've got a MAX9814 microphone and for whatever reason it only works when `channel_format` is set to `I2S_CHANNEL_FMT_ONLY_LEFT`.

The default of `I2S_CHANNEL_FMT_RIGHT_LEFT` causes errors in the logs, and I can't receive any audio

I would like to provide my own configuration to adjust `channel_format`, but at the moment it is hard coded.

https://github.com/pschatzmann/arduino-audio-tools/blob/267455d3bbea215c18723dedf0b2863d1c3e2048/src/AudioAnalog/AnalogAudioESP32.h#L59

Perhaps there is another way to drop the right channel? But I think it would be nicer to adjust the channels using built-in options, rather than introducing extra streams or copiers.

#### Log with errors

```text
======================================================================================================= [SUCCESS] Took 26.07 seconds =======================================================================================================
--- Terminal on /dev/cu.SLAB_USBtoUART | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets S���Ғ�Ң�j

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Starting input stream...[I] AnalogAudioBase.h : 54 - I2S_MODE_ADC_BUILT_IN
[I] AnalogAudio.h : 57 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AnalogAudioESP32.h : 48 - virtual bool audio_tools::AnalogDriverESP32::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 56 - sample_rate: 44100
[I] AudioTypes.h : 57 - channels: 1
[I] AudioTypes.h : 58 - bits_per_sample: 16
[I] AnalogAudioBase.h : 75 - input pin1: 35
[I] AnalogAudioESP32.h : 87 - RX_MODE
 Input stream started
[I] CodecWAV.h : 455 - is_streamed! because length is 2147418112
...
IP address: 192.168.178.157
[I] AudioServer.h : 216 - New Client.
[I] AudioServer.h : 357 - sendReply - Returning encoded stream...
[I] CodecWAV.h : 455 - is_streamed! because length is 4294967295
[I] AudioCopy.h : 69 - buffer_size=1024
[I] CodecWAV.h : 499 - Writing Header
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[  6724][E][WiFiClient.cpp:422] write(): fail on fd 49, errno: 104, "Connection reset by peer"
[I] AudioCopy.h : 285 - try write - 2 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 3 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 4 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 5 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 6 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 7 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 8 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 9 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 10 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 11 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 12 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 13 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 14 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 15 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 16 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 17 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 18 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 19 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 20 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 21 (open 1024 bytes) 
[E] AudioCopy.h : 279 - write to target has failed! (1024 bytes)
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 0 bytes - in 22 hops
[I] AudioServer.h : 122 - stop client...
[I] AudioServer.h : 216 - New Client.
[I] AudioServer.h : 357 - sendReply - Returning encoded stream...
[I] CodecWAV.h : 455 - is_streamed! because length is 4294967295
[I] AudioCopy.h : 69 - buffer_size=1024
[I] CodecWAV.h : 499 - Writing Header
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[ 12457][E][WiFiClient.cpp:422] write(): fail on fd 50, errno: 104, "Connection reset by peer"
[I] AudioCopy.h : 285 - try write - 2 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 3 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 4 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 5 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 6 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 7 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 8 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 9 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 10 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 11 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 12 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 13 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 14 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 15 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 16 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 17 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 18 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 19 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 20 (open 1024 bytes) 
[I] AudioCopy.h : 285 - try write - 21 (open 1024 bytes) 
[E] AudioCopy.h : 279 - write to target has failed! (1024 bytes)
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 0 bytes - in 22 hops
[I] AudioServer.h : 122 - stop client...
```

#### Sample code

Here is the code I am using:

```cpp
#include "AudioTools.h"
#include "WiFiCredentials.h"
#include "AudioLibs/MemoryManager.h"

AudioWAVServer server(WIFI_SSID, WIFI_PASS);
AnalogAudioStream audioSource;
ConverterAutoCenter<int16_t> converter; // Unsigned integer to +/-

const int SAMPLE_RATE = 44100;
const int CHANNELS = 1;

MemoryManager memory(500); // Activate SPI RAM for objects > 500 bytes

void setup(){
  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  AudioBaseInfo baseAudioConf = AudioBaseInfo();
  baseAudioConf.bits_per_sample = 16;
  baseAudioConf.channels = CHANNELS;
  baseAudioConf.sample_rate = SAMPLE_RATE;

  Serial.print("Starting input stream...");
  auto audioSourceConf = audioSource.defaultConfig(RX_MODE);
  audioSourceConf.copyFrom(baseAudioConf);
  audioSourceConf.mode_internal = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN;
  audioSourceConf.port_no = I2S_NUM_0;
  audioSourceConf.use_apll = false;
  audioSourceConf.buffer_count = 4;
  audioSourceConf.buffer_size = 1024;
  audioSourceConf.setInputPin1(GPIO_NUM_35);
  audioSource.begin(audioSourceConf);
  Serial.println(" Input stream started");

  server.begin(audioSource, audioSourceConf, &converter);
}

void loop() {
  server.copy();
}
```

pschatzmann, 2022-12-15:
Well, the recommended way is to start with a CSV output and then you can see what you need to do with the signal. I have a converter that can copy the audio from the good channel to the bad one so that you get 2 properly working channels...

aSemy, 2022-12-15:
I wasn't able to get the Arduino Serial Plotter working (something about it not being able to connect to the serial port). I'm using CLion and PlatformIO, so it didn't fit into the work flow. I tried finding an alternative graphing tool, but I couldn't find anything that worked. Setting up a WAV server was the first thing that worked. Maybe you can recommend something else? :)

I'll see if I can expose an option and play around with it, and get something working.

pschatzmann, 2022-12-15:
Nop, I am usually coding my library in VS Code and I check my (always small) sketches in Arduino.
I was considering to use PlatformIO more, but I didn't because I did not find a good alternative to the Serial Plotter!

mohaddeseh-rezaee, 2022-12-16:
I am using PlatformIO...
when you want to use the plotter of arduino, you have to comment `monitor_speed` in platform.ini file
although I use hterm because it lets me save data