pschatzmann/arduino-audio-tools #528: stack collision/corruption when trying to read sample mp3 with URLStream

0x0fe, 2022-12-15:
So, while experimenting with the multioutput on the sdfat-ffti2s example (modified to play an UrlStream) i face a stack corruption / stack smashing protection which is not random as it occurs everytime (in loop).
The file is this one, 1.61MB and the stack corruption occurs after about 5 or 6 seconds
https://filesamples.com/samples/audio/mp3/sample3.mp3
In the code below i commented the FFT display since it is not useful to show the bug and it fills the debug output.
What i notice is that after few second streamcopy::copy get messed up and it crashes right after that.
`StreamCopy::copy 1024 -> 4294967295 -> 0 bytes`
What could be the cause of this issue?
The platform is ESP32S3 and it plays on an external DAC.

```
#include "AudioTools.h"
#include "AudioLibs/AudioSourceSD.h"
#include "AudioCodecs/CodecMP3Helix.h"
#include "AudioLibs/AudioRealFFT.h" // or AudioKissFFT or others
const char *urls[] = {
  "https://filesamples.com/samples/audio/mp3/sample3.mp3",
  "https://ice6.somafm.com/groovesalad-128-mp3",
  "http://stream.srg-ssr.ch/m/rsj/mp3_128",
};
const char *startFilePath="/";
const char* ext="mp3";
const char *wifi = "SSID";
const char *password = "PASS";
//AudioSourceSD source(startFilePath, ext);
URLStream urlStream(wifi, password);
AudioSourceURL source(urlStream, urls, "audio/mp3");
MultiOutput multi_output;
MP3DecoderHelix decoder;
AudioPlayer player(source, multi_output, decoder);
I2SStream i2s; // output to i2s
AudioRealFFT fft; // or AudioKissFFT or others

#define MCK             17
#define WS              18
#define BCK             21
#define DAT             4
#define MUTE            3

// display fft result
void fftResult(AudioFFTBase &fft){
//    int diff;
//    auto result = fft.result();
//    if (result.magnitude>100){
//        Serial.print(result.frequency);
//        Serial.print(" ");
//        Serial.print(result.magnitude);  
//        Serial.print(" => ");
//        Serial.print(result.frequencyAsNote(diff));
//        Serial.print( " diff: ");
//        Serial.println(diff);
//    }
}

void setup() {
  
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup I2S
  auto cfg = i2s.defaultConfig(TX_MODE);
  cfg.pin_mck = MCK;
  cfg.pin_bck = BCK;
  cfg.pin_ws = WS;
  cfg.pin_data = DAT;    
  i2s.begin(cfg);

  // Setup FFT
  auto tcfg = fft.defaultConfig();
  tcfg.copyFrom(cfg);
  tcfg.length = 1024;
[... 533 lines cut ...]
[I] HttpHeader.h : 162 - HttpHeader::readLine -> ؎G⸮*⸮/⸮⸮2⸮)J⸮.7⸮⸮@⸮⸮+⸮SO⸮⸮⸮ӗ⸮⸮⸮⸮be(⸮⸮j⸮⸮(⸮⸮⸮e⸮D `
⸮3⸮ ⸮⸮>⸮⸮⸮	⸮rÎYT⸮kW⸮⸮ ⸮zʺT⸮⸮G ⸮}i9$⸮I⸮4⸮⸮⸮`⸮>4`U:؉⸮8 x((⸮⸮l⸮⸮⸮⸮⸮⸮
\
[I] HttpHeader.h : 162 - HttpHeader::readLine -> 
⸮⸮⸮⸮⸮B⧥⸮⸮⸮⸮+⸮w⸮@⸮9餒}⸮⸮⸮⸮⸮⸮?⸮r⸮'&⸮7 IM}'⸮⸮$⸮⸮⸮⸮⸮⸮C⸮ 5Z⸮e⸮̲⸮⸮0⸮\A⸮@=⸮)⸮囥⸮ 2%Gw⸮*⸮⸮e⸮_&ϡ⸮`8 /⸮
[I] HttpHeader.h : 162 - HttpHeader::readLine -> 8⸮8ౠ
[I] HttpHeader.h : 162 - HttpHeader::readLine -> ⸮⸮'⸮	⸮=%⸮⸮⸮@@⸮b⸮
[E] HttpLineReader.h : 73 - Line cut off: ⸮⸮s⸮釄⸮D⸮⸮⸮ror]H⸮@⸮$⸮⸮=⸮!⸮⸮4⸮$⸮@⸮⸮⸮x⸮I⸮⸮⸮"⸮⸮⸮P?7@⸮⸮⸮⸮⸮d⸮3⸮<Z⸮I⸮Jț=%⸮
⸮⸮g1'⸮l⸮⸮	Л⸮eK$⸮⸮⸮⸮a⸮⸮⸮q9⸮h
[I] HttpHeader.h : 162 - HttpHeader::readLine -> ⸮⸮s⸮釄⸮D⸮⸮⸮ror]H⸮@⸮$⸮⸮=⸮!⸮⸮4⸮$⸮@⸮⸮⸮x⸮I⸮⸮⸮"⸮⸮⸮P?7@⸮⸮⸮⸮⸮d⸮3⸮<Z⸮I⸮Jț=%⸮
⸮⸮g1'⸮l⸮⸮	Л⸮eK$⸮⸮⸮⸮a⸮⸮⸮q9⸮h
[I] HttpHeader.h : 162 - HttpHeader::readLine -> w⸮⸮
[I] HttpHeader.h : 162 - HttpHeader::readLine -> ⸮9e⸮N⸮⸮⸮⸮ߡ
[E] HttpLineReader.h : 73 - Line cut off: W⸮⸮⸮⸮@⸮͒%⸮⸮*⸮⸮⸮⸮
[I] HttpHeader.h : 162 - HttpHeader::readLine -> W⸮⸮⸮⸮@⸮͒%⸮⸮*⸮⸮⸮⸮
[E] HttpLineReader.h : 73 - Line cut off: "
[I] HttpHeader.h : 162 - HttpHeader::readLine -> "
[I] HttpHeader.h : 162 - HttpHeader::readLine -> 

Stack smashing protect failure!

abort() was called at PC 0x4201f8d3 on core 1

Backtrace:0x4037791e:0x3fce2c900x4037cc1d:0x3fce2cb0 0x40382691:0x3fce2cd0 0x4201f8d3:0x3fce2d50 0x420096fe:0x3fce2d70 0x42007259:0x3fce00f0 0x4200727f:0x3fce0014  |<-CORRUPTED
```

pschatzmann, 2022-12-15:
You were really taking the combination of the most RAM demanding functionalities possible:

- mp3 decoding uses a lot of RAM - (wav would be more efficient)
- wifi uses a lot of RAM - (reading from files would less costly)
- FFT uses a lot of RAM - (try out the different implementations and decrease the fft size)

I assume that you pushed it a litte bit too far.  I suggest that you print out the available ram in the loop as well to check if my assumption is right..

0x0fe, 2022-12-15:
i see, that makes sense. The S3 variant i use has and additional 2MB internal PSRAM but it is not put to contribution yet.
As a side note i tested with another sample URL  (https://file-examples.com/wp-content/uploads/2017/11/file_example_MP3_2MG.mp3) and it worked fine, i could read the file in full, so the issue is also related to how the file is served apparently, maybe the drops increases buffer usage and lead to the stack collision.

Anyway i will disable the FFT (of no use here) and write to the file. Later i will try to figure how to assign some of the PSRAM to audio-tools so i can increase buffering.

closed 2023-02-13 by pschatzmann

ffaerber, 2024-09-18:
I had this "Line cut off"  problem with URLStream too.
the problem is the server. the mp3 is not send via chunked. 
i fixed it via highWaterMark

here is a nestjs controller example

` @Get(':id/audio')
  @Header('Content-Type', 'audio/mpeg')
  @Header('Cache-Control', 'no-cache')
  async getAudio(
    @Param('id') id: string,
    @Req() req, 
    @Res({ passthrough: true }) res
  ): Promise<StreamableFile> {
    const message: Message = await this.messagesService.findUnique({ where: { id } });
    try {
      const filePath = path.join(messagesPath, message.fileName);
      const readStream = createReadStream(filePath, { highWaterMark: 1024 });

      res.set({
        'Content-Disposition': 'attachment; filename=audio.mp3',
        'Transfer-Encoding': 'chunked',  // Enable chunked transfer
        'Connection': 'keep-alive',      // Keep the connection open
      });
      return new StreamableFile(readStream);
    } catch (error) {
      throw new Error('File not found or could not be read');
    }
  }`

took me a few hours to figure this out.