pschatzmann/arduino-audio-tools #535: Compile Error with I2S Source & A2DP Sink

udponly, 2022-12-21:
Hi Phil, 
first of all: thanks for the huge work on the library!

I am trying to use a PMOD I2S2 as Audio Input/Output module with an ESP32, currently using the D1_mini32 Board.

Not so sure about the pin mappings, but I *think* they are somewhat working after switching `pin_data` and `pin_data_rx`.
I was first assuming `pin_data_rx` meant the *Arduino MCU* was receiving on this end, but I had always 0,0 output in the CSV copier sketch. After Switching the `SDOUT` to `pin_data` at least I would see some non-negative numbers...

Anyway, here is the compilation error I am facing atm:
```
n file included from f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioI2S/I2SStream.h:11,
                 from f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools.h:60,
                 from F:\Documents\Arduino\I2STest\I2STest.ino:48:
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioI2S/I2SESP32.h: In member function 'bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfig, int, int)':
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioI2S/I2SESP32.h:158:7: warning: missing initializer for member 'i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
       };
       ^
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioI2S/I2SESP32.h:158:7: warning: missing initializer for member 'i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
In file included from f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioAnalog/AnalogAudio.h:5,
                 from f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools.h:62,
                 from F:\Documents\Arduino\I2STest\I2STest.ino:48:
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioAnalog/AnalogAudioESP32.h: In member function 'virtual bool audio_tools::AnalogDriverESP32::begin(audio_tools::AnalogConfig)':
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioAnalog/AnalogAudioESP32.h:67:9: warning: missing initializer for member 'i2s_driver_config_t::fixed_mclk' [-Wmissing-field-initializers]
         };
         ^
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioAnalog/AnalogAudioESP32.h:67:9: warning: missing initializer for member 'i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioAnalog/AnalogAudioESP32.h:67:9: warning: missing initializer for member 'i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
In file included from F:\Documents\Arduino\I2STest\I2STest.ino:49:
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioLibs/AudioA2DP.h: In member function 'void audio_tools::A2DPStream::begin(audio_tools::A2DPConfig)':
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioLibs/AudioA2DP.h:140:20: error: enumeration value 'UNDEFINED_MODE' not handled in switch [-Werror=switch]
             switch (cfg.mode){
                    ^
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioLibs/AudioA2DP.h:140:20: error: enumeration value 'RXTX_MODE' not handled in switch [-Werror=switch]
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioLibs/AudioA2DP.h: In static member function 'static int32_t audio_tools::A2DPStream::a2dp_stream_source_sound_data(Frame*, int32_t)':
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioLibs/AudioA2DP.h:344:58: error: 'void* memset(void*, int, size_t)' clearing an object of non-trivial type 'struct Frame'; use assignment or value-initialization instead [-Werror=class-memaccess]
                         memset(data, 0, len*sizeof(Frame));
                                                          ^
In file included from f:\Documents\Arduino\libraries\ESP32-A2DP\src/BluetoothA2DPCommon.h:41,
                 from f:\Documents\Arduino\libraries\ESP32-A2DP\src/BluetoothA2DPSink.h:18,
                 from f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioLibs/AudioA2DP.h:13,
                 from F:\Documents\Arduino\I2STest\I2STest.ino:49:
f:\Documents\Arduino\libraries\ESP32-A2DP\src/SoundData.h:26:32: note: 'struct Frame' declared here
 struct __attribute__((packed)) Frame {
                                ^~~~~
In file included from f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools.h:56,
                 from F:\Documents\Arduino\I2STest\I2STest.ino:48:
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/Converter.h: In instantiation of 'void audio_tools::ConverterFillLeftAndRight<T>::setup(T*, size_t) [with T = short int; size_t = unsigned int]':
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/Converter.h:234:17:   required from 'size_t audio_tools::ConverterFillLeftAndRight<T>::convert(uint8_t*, size_t) [with T = short int; size_t = unsigned int; uint8_t = unsigned char]'
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/Converter.h:231:16:   required from here
f:\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/Converter.h:260:20: error: variable 'sample' set but not used [-Werror=unused-but-set-variable]
                 T *sample = (T*) src;
                    ^~~~~~
cc1plus.exe: some warnings being treated as errors
```
Sketch used:
```
#include <Arduino.h>
#include <WiFi.h>
/**
 * @file streams-i2s-serial.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-i2s-serial/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioLibs/AudioA2DP.h"

uint16_t sample_rate = 44100;
uint16_t channels = 2;
I2SStream i2sStream; // Access I2S as stream
A2DPStream a2dpStream; // access A2DP as stream

VolumeStream volume(a2dpStream);
StreamCopy copier(volume, i2sStream); // copy i2sStream to a2dpStream
ConverterFillLeftAndRight < int16_t > filler(LeftIsEmpty); // fill both channels

// Arduino Setup
void setup(void) {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);
  volume.setVolume(0.3);

  Serial.println("starting A2DP...");
  auto cfgA2DP = a2dpStream.defaultConfig(TX_MODE);
  cfgA2DP.name = "ES-151A";
  a2dpStream.begin(cfgA2DP);

  // start I2S in
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.sample_rate = sample_rate;
  config.bits_per_sample = 16;
  config.i2s_format = I2S_STD_FORMAT;
  config.is_master = true;
  config.port_no = 0;
  config.pin_ws = 18;
  config.pin_bck = 5;
  /*config.pin_data = 19;
  config.pin_data_rx = 17; //original*/
  // switcheroo:
  config.pin_data = 17;
  config.pin_data_rx = 19;
  config.pin_mck = 0;
  config.use_apll = true;
  a2dpStream.setNotifyAudioChange(i2sStream); // i2s is using the info from a2dp
  i2sStream.begin(config);
  Serial.println("I2S started...");
}

// Arduino loop - copy data
void loop() {
  copier.copy();
}
```

my assumption is that there might be a naming collision between RX_MODE, etc. between the libraries?

thanks a lot in advance and best regards!

pschatzmann, 2022-12-21:
I am not able to reproduce your issue. I am neither getting any warnings nor errors.
Are you sure that you installed the latest version of all libraries and that you use the latest ESP board version which is 2.0.5

udponly, 2022-12-21:
thanks for your reply! I re-verified that I have the latest commits of this repo, ESP32-A2DP and ESP board (2.0.5).
Here is a more complete log. I am unable to see what might be wrong. can you? 
[compile_error.log](https://github.com/pschatzmann/arduino-audio-tools/files/10278372/compile_error.log)

regarding RXTX_MODE and pin "remapping", thanks for clearing that up, explains a lot!

pschatzmann, 2022-12-21:
Please make sure that you are treating warnings not as errors. In Preferences the setting Compiler Warnings should be set to normal

udponly, 2022-12-21:
ow, thanks a lot! I thought this settings was merely for *outputting* not for handling.....
thanks for solving! <3

closed 2022-12-21 by udponly