pschatzmann/arduino-audio-tools #563: Stuttering audio with faust examples

hjanetzek, 2023-01-09:
Hello this library looks really awesome!

The stream-stk-audiokit.ino example works well, but when I try the Faust examples the sound is stuttering and distorted. Is this a known issue with the audiokit boards (2957 3478) or do you have a hint what to look for?

thanks alot :)

hjanetzek, 2023-01-09:
Checking the wave output with an oscilloscope looks as if only half of a buffer per channel is filled

btw there's a little off-by-one in the stk-audiokit example `freq = notes_array[random((sizeof(notes_array)-1)/sizeof(uint16_t))];` seems to work

pschatzmann, 2023-01-10:
I converted the notes form integers to floats, but missed to adapt this correctly. 
The logic should be

      int count = sizeof(notes_array)/sizeof(float);
      freq = notes_array[random(count)]; // generate random number from 0 to count-1

closed 2023-01-10 by pschatzmann

hjanetzek, 2023-01-10:
Sorry for having intermingled two issues.. could you verify if the Faust flute example works with the Audiokit board (2957 3478) ? Running the flute faust example from Arduino IDE produces this discontinuity for example:
[image]
[image]

hjanetzek, 2023-01-10:
ok it seems to be related to channel interleaving: changing to 
```
virtual int getNumOutputs() {
		return 1;
	}
```
(and of course `FAUSTFLOAT* output1 = outputs[0];`)
produces a continuous wafevorm

hjanetzek, 2023-01-10:
should L/R channel be interleaved per sample or per frame by ` readBytes(uint8_t *data, size_t len)`?

hjanetzek, 2023-01-10:
this makes the two channel flute work for me:
```
    fix Faust readBytes/write for more than one channel

-------------------------- src/AudioLibs/AudioFaust.h --------------------------
index c13b9069..9f5e32ad 100644
@@ -117,5 +117,5 @@ class FaustStream : public AudioStreamX {
             TRACED();
             result = len;
-            int samples = len / bytes_per_sample;
+            int samples = (len / bytes_per_sample) / cfg.channels;
             allocateFloatBuffer(samples, false);
             p_dsp->compute(samples, nullptr, p_buffer);
@@ -131,9 +131,10 @@ class FaustStream : public AudioStreamX {
         if (is_write){
             TRACED();
-            int samples = len / bytes_per_sample;
+            int samples = (len / bytes_per_sample) / cfg.channels;
             allocateFloatBuffer(samples, with_output_buffer);
             int16_t *data16 = (int16_t*) write_data;
             // convert to float
-            int frameCount = samples/cfg.channels;
+            // TODO support different number of input and output channels
+            int frameCount = samples;
             for(int j=0;j<frameCount;j++){
                 for(int i=0;i<cfg.channels;i++){
@@ -263,5 +264,5 @@ class FaustStream : public AudioStreamX {
     void convertFloatBufferToInt16(int samples, void *data, FAUSTFLOAT**p_float_out){
         int16_t *data16 = (int16_t*) data;
-        int frameCount = samples/cfg.channels;
+        int frameCount = samples;
         for (int j=0;j<frameCount;j++){
             for (int i=0;i<cfg.channels;i++){
```