pschatzmann/arduino-audio-tools #571: sd.begin faild

BeijingUncle, 2023-01-14:
[W] AudioA2DP.h : 300 - ==> state: Connecting
[W] AudioA2DP.h : 300 - ==> state: Connected
[E] AudioSourceSDFAT.h : 89 - sd.begin failed
[E] SDDirect.h : 231 - Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

SPI setup:
SPI.begin(14, 2, 15, 13);

The other sketch which use sdfat to list files on tf card works fine:

#include <SPI.h>
#include "SdFat.h"
#include <iostream>
using namespace std;

SdFs sd;     
SPIClass hspi(HSPI);

void setup() {
  Serial.begin(115200);

  delay(3000);
  hspi.begin(14, 2, 15, 13);

  if(!sd.begin(SdSpiConfig(13, DEDICATED_SPI, 18000000, &hspi)))
  {
    Serial.println("error");
    return;
  }
...

Thanks!

copyCATandpasteDOG, 2023-01-14:
Hi BeijingUncle,
isn't spi.begin to be declared like that: //spiXXX.begin(CLK,MISO,MOSI,CS); //spiXXX is fake of course

pschatzmann, 2023-01-14:
You did not explain what you are doing so that it is not working. 
So, I can't really help you!
I suggest that you have a closer look at the following example which is working with the custom pins used by the AudioKit: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/player-sdfat-audiokit/player-sdfat-audiokit.ino
You could also use the SD or SDMMC implementations that do not required the installation of any additional library.

BeijingUncle, 2023-01-14:
> You did not explain what you are doing so that it is not working. So, I can't really help you! I suggest that you have a closer look at the following example which is working with the custom pins [...]

Sorry pschatzmann,

I'm trying to test player-sdfat-a2dp example:
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-sdfat-a2dp
on my LilyGO T4 V1.3:
https://github.com/Xinyuan-LilyGO/LilyGo_Txx

In the example, 

// Setting up SPI if necessary with the right SD pins by calling 
  // SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);

So I just uncomment the "SPI.begin" line and enter the pin number and also cfg.name = "JBL TUNE660NC"; 

The example can connect my JBL TUNE660NC, and then I got the error:

[W] AudioA2DP.h : 300 - ==> state: Connecting
[W] AudioA2DP.h : 300 - ==> state: Connected
[E] AudioSourceSDFAT.h : 89 - sd.begin failed
[E] SDDirect.h : 231 - Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.

pschatzmann, 2023-01-14:
Just use the complete AudioSourceSDFAT definition that I have proposed in the example above. This also corresponds to your working example!
By the way I would recommend to use the example using the Basic API instead. Plenty of people had some problems with the A2DP stability/quality with this example.

copyCATandpasteDOG, 2023-01-14:
I do an equivalente thing with a sdcard and a tft,i use HSPI for the TFT and VSPI for the sdcard,
but the HSPI pins are : (14,12,13,CS) and the VSPI pins are : (18,19,23,CS)

It seems you didn't give the right pins order:
// SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);

I think because these pins are mandatory for each SPI class
for HSPI : CLK=14,MISO=12,MOSI=13,CS as you want
for VSPI : CLK=18,MISO=19,MOSI=23,CS as you want

pschatzmann, 2023-01-14:
My understanding is that the both for HSPI and VSPI all pins are freely mappable and you can use whatever you want. I think there might however be a performance advantage to use the default pins....

copyCATandpasteDOG, 2023-01-14:
I believe you're right,you have to suit those pinout (lyraT),but he said he was using also a LilyGO T4 wich also use spi,
i had problems configuring 2 spi modules at the same time,and the better way to do it for me was to
use the mandatory pinouts (CS is freely use of course) of the 2 classes at the same time,and had it get to work:)
I would be glad you give me another wayout that can fit a lyraT and a tft together on a esp32:)

pschatzmann, 2023-01-14:
To complete the picture: You can also have multiple SPI devices working with the same SPI port. For each device you would have a separate CS pins, but all other pins would be shared.

copyCATandpasteDOG, 2023-01-14:
I tried that before but it never worked,had to look at forums and saw a good solution working,but i was
using a sdcard and a tft,maybe the two together on the same spi port was not a good idea :)

pschatzmann, 2023-01-14:
I agree - for Audio speed is important. To use a dedicated port should be much faster...

BeijingUncle, 2023-01-14:
I got it works!

#include <SPI.h>
#include <SdFat.h>
#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"
#include "AudioLibs/AudioA2DP.h"

File audioFile;
SdFs sd;
SPIClass hspi(HSPI);

A2DPStream out;
EncodedAudioStream decoder(&out, new MP3DecoderHelix()); // Decoding stream
StreamCopy copier;

void setup() {
  Serial.begin(115200);

  hspi.begin(14, 2, 15, 13);

  if(!sd.begin(SdSpiConfig(13, DEDICATED_SPI, 18000000, &hspi))) {
    Serial.println("error");
    return;
  }

  audioFile = sd.open("/wrong-place.mp3");

  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "JBL TUNE660NC";

  out.begin(cfg);
  out.setVolume(0.3);

  decoder.setNotifyAudioChange(out);
  decoder.begin();

  copier.begin(decoder, audioFile);
}

void loop() {
  if (!copier.copy()) {
    stop();
  }
}

pschatzmann, 2023-01-14:
Actually, you don't need to create your own SPIClass hspi(HSPI); This is actually the same like the exising SPI provided by Arduino, so you can remove that line and replace all hspi with SPI...

BeijingUncle, 2023-01-14:
There is another problem:

I copy a big mp3 (11M) and it can be played but  lost some, and got lots of warning:

libhelix -  -> invalid frame size: -1 / max: 1600

Any suggestion?

pschatzmann, 2023-01-14:
I am not sure if this is a problem. I suggest that you try with some other mp3 files

BeijingUncle, 2023-01-14:
> Actually, you don't need to create your own SPIClass hspi(HSPI); This is actually the same like the exising SPI provided by Arduino, so you can remove that line and replace all hspi with just the standard SPI...

Yes, it works just fine! Thanks!
---------------------------  
  SPI.begin(14, 2, 15, 13);

  if(!sd.begin(SdSpiConfig(13, DEDICATED_SPI, 18000000))) {
    Serial.println("error");
    return;
  }

pschatzmann, 2023-01-14:
you could also use sd.begin(SdSpiConfig(13, DEDICATED_SPI, 18000000, &SPI))) if you want to provide all parameters

BeijingUncle, 2023-01-14:
> 

Change another MP3, still same problem, sounds lost some data to encode, SPI speed problem?

pschatzmann, 2023-01-14:
Difficult to say. It could also be the A2DPStream. 
You would need to test the output with something which is more reliable: E.g. I2SStream or the AnalogAudioStream.
As I have indicated above you should use the Basic API for A2DP which uses callbacks. This is usually much more reliable than the A2DPStream.

BeijingUncle, 2023-01-14:
> Difficult to say. It could also be the A2DPStream. You would need to test the output with something which is more reliable: E.g. I2SStream or the AnalogAudioStream. As I have indicated above you [...]

Thank you for your help pschatzmann, I will try tomorrow, It's too late here, have a nice day!

BeijingUncle, 2023-01-15:
I found an issue in arduino/libhelix:
https://github.com/pschatzmann/arduino-libhelix/issues/1

Then I change it in MP3DecoderHelix.h:

//#define MP3_MAX_OUTPUT_SIZE 1024 * 5
#define MP3_MAX_OUTPUT_SIZE 1024 * 20

//#define MP3_MAX_FRAME_SIZE 1600
#define MP3_MAX_FRAME_SIZE 3200

and the sketch now works just fine!

pschatzmann, 2023-01-15:
Cool, but I am still surprised that you need such high values. 
Where did you get the mp3 files from ?
Can you share one such file ?

BeijingUncle, 2023-01-15:
> Cool, but I am still surprised that you need such high values: The higher you set the values the more (valuable) RAM is reserved. Where did you get the mp3 files from ? Can you share one such file ?

Have a look at https://www.jamendo.com/start, there are lots of songs for personal use.

BeijingUncle, 2023-01-15:
play-sdfat-a2dp example works fine!

/**
 * @file player-sdfat-a2dp.ino
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-sdfat-a2dp/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioLibs/AudioA2DP.h"
#include "AudioLibs/AudioSourceSDFAT.h"
#include "AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/";
const char* ext="mp3";
SdSpiConfig sdcfg(13, DEDICATED_SPI, SD_SCK_MHZ(18));  //  <--------  add this line
AudioSourceSDFAT source(startFilePath, ext, sdcfg); // <------- add the third param

A2DPStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  SPI.begin(14, 2, 15, 13);   // <------ add this line

  // setup output - We send the test signal via A2DP - so we conect to the "LEXON MINO L" Bluetooth Speaker
  auto cfg = out.defaultConfig(TX_MODE);
  // cfg.name = "LEXON MINO L";
  cfg.auto_reconnect = true;  // if this is use we just quickly connect to the last device ignoring cfg.name
  out.begin(cfg);

  // setup player
  player.setVolume(0.1);
  player.begin();

}

void loop() {
  player.copy();
}

pschatzmann, 2023-01-15:
Strange, I tried a song from your proposed url, but I can't reproduce your issue with the mp3 parser error messages

BeijingUncle, 2023-01-15:
> Strange, I tried a song from your proposed url, but I can't reproduce your issue with the mp3 parser error messages

How can I send you the mp3 file with problem?

BeijingUncle, 2023-01-15:
Two more questions:
1. AudioPlayer, play() -> stop() -> play(), can here some noise;
2. AudioPlayer, player.setMetadataCallback(printMetaData) dos not work.

BeijingUncle, 2023-01-17:
> > Strange, I tried a song from your proposed url, but I can't reproduce your issue with the mp3 parser error messages
> 
> https://github.com/BeijingUncle/CWS/blob/master/docs/test.zip)

Did you get the problem what I met with this mp3?

pschatzmann, 2023-01-19:
Yes. It seems that some parts have a too high variable bitrate for the defined max frame size in this file. Increasing the frame size e.g. by calling.
<code>
  decoder.setMaxFrameSize(2100);
</code>

Is resolving the issue.
I did not get any messages in the log, though...

BeijingUncle, 2023-01-20:
> There is another problem:
> 
> I copy a big mp3 (11M) and it can be played but lost some, and got lots of warning:
> 
> libhelix - -> invalid frame size: -1 / max: 1600
> 
> Any suggestion?

I think I can avoid these info:

1. reduce SCK frequency of SD to 4M(default value);
2. reformat my SD card.

closed 2023-01-22 by pschatzmann