pschatzmann/arduino-audio-tools #574: a2dp connect->disconnect->reconnect

BeijingUncle, 2023-01-16:
I'm playing with player-sdfat-a2dp example, when I power off BT Speaker and power on again after a while, a2dp connected again but player has no action. Is there an event or callback for reconnected, and how can I tell player to restart?

09:40:44.600 -> entry 0x400805e4                                                                                           <---- power on
09:40:46.512 -> [W] AudioA2DP.h : 300 - ==> state: Connecting                                           <---- first time connecting
09:40:51.624 -> [W] AudioA2DP.h : 300 - ==> state: Disconnected                                       <---- always disconnected at first time
09:40:52.447 -> [W] AudioA2DP.h : 300 - ==> state: Connecting                                           <---- try to connect again
09:40:52.646 -> [W] AudioA2DP.h : 300 - ==> state: Connected                                            <---- connected
09:40:54.097 -> [W] AudioA2DP.h : 209 - is_a2dp_active -> true with 59904 bytes            <---- ready, player start playing
09:41:11.416 -> [W] AudioA2DP.h : 300 - ==> state: Disconnected                                          <---- BT Speaker power off
09:41:16.630 -> [W] AudioA2DP.h : 300 - ==> state: Connecting                                             <---- try to connect
09:41:21.745 -> [W] AudioA2DP.h : 300 - ==> state: Disconnected
09:41:26.623 -> [W] AudioA2DP.h : 300 - ==> state: Connecting
09:41:31.736 -> [W] AudioA2DP.h : 300 - ==> state: Disconnected
09:41:36.616 -> [W] AudioA2DP.h : 300 - ==> state: Connecting
09:41:41.733 -> [W] AudioA2DP.h : 300 - ==> state: Disconnected
09:41:46.620 -> [W] AudioA2DP.h : 300 - ==> state: Connecting
09:41:51.742 -> [W] AudioA2DP.h : 300 - ==> state: Disconnected
09:41:56.628 -> [W] AudioA2DP.h : 300 - ==> state: Connecting
09:42:01.739 -> [W] AudioA2DP.h : 300 - ==> state: Disconnected
09:42:03.817 -> [W] AudioA2DP.h : 300 - ==> state: Connecting                                            <---- BT Speaker power on
09:42:03.983 -> [W] AudioA2DP.h : 300 - ==> state: Connected                                            <---- connected but player cannot resume

pschatzmann, 2023-01-16:
I guess you need to restart the player again when you get the connected event.

BeijingUncle, 2023-01-16:
> I guess you need to restart the player again when you get the connected event.

How can I get the event?

pschatzmann, 2023-01-16:
Check the A2DP documentation for the set_on_connection_state_changed callback

BeijingUncle, 2023-01-18:
I do think I was wrong, but how can I do it in a right way?

#include "AudioTools.h"
#include "AudioLibs/AudioA2DP.h"
#include "AudioLibs/AudioSourceSDFAT.h"
#include "AudioCodecs/CodecMP3Helix.h"

SdSpiConfig sdcfg(13, DEDICATED_SPI, SD_SCK_MHZ(18));
AudioSourceSDFAT source("/", "mp3", sdcfg);
BluetoothA2DPSource a2dp_source;
A2DPStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  Serial.println(a2dp_source.to_str(state));
}

void setup() {
  Serial.begin(115200);
  delay(500);
  
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  a2dp_source.set_on_connection_state_changed(connection_state_changed);
  
  SPI.begin(14, 2, 15, 13);

  auto cfg = out.defaultConfig(TX_MODE);
  cfg.auto_reconnect = false;
  cfg.name = "";
  
  out.begin(cfg);

  player.begin();
}

void loop() {
  player.copy();
}

BeijingUncle, 2023-01-20:
I set a set_on_connection_state_changed callback after out.begin():

out.begin(cfg);
out.source().set_on_connection_state_changed(connection_state_changed);

and do nothing but printout the state change of connection in the callback function.

When music is playing, I switch off the speaker, the callback is called(Disconnected).

And then I switch on the speaker, the callback is called twice(Connecting & Connected).

It's in a right way, and of course music is not resume.

Then I uncomment the action in the callback function, and switch off the speaker, when I got Disconnected callback, I did:

player.stop();

Then I switch on the speaker, I cannot get any callback this time.

What I did wrong?

The sketch
---------------------------

#include <Button2.h>
#include "AudioTools.h"
#include "AudioLibs/AudioA2DP.h"
#include "AudioLibs/AudioSourceSDFAT.h"
#include "AudioCodecs/CodecMP3Helix.h"

#define SD_CS   13
#define SD_MISO 2
#define SD_MOSI 15
#define SD_CLK  14

#define BTN_1 37
#define BTN_2 38
#define BTN_3 39

Button2 btn1, btn2, btn3;

float vol = 0.5;
boolean isConnected = false;

SdSpiConfig sdcfg(SD_CS, DEDICATED_SPI, SD_SCK_MHZ(4));
AudioSourceSDFAT source("/", "mp3", sdcfg);

A2DPStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

void longClick(Button2& btn) {
  unsigned int time = btn.wasPressedFor();

  if (time > 1000) {
    if (btn == btn2) {
      player.previous();
    } else {
      player.next();
    }
  } else {
    if (btn == btn2) {
      if (vol > 0.05) {
        vol = vol - 0.05;
        player.setVolume(vol);
      }
    } else {
      if (vol < 0.95) {
        vol = vol + 0.05;
        player.setVolume(vol);
      }
    }
  }
}

void togglePlay(Button2& btn) {
  if (player.isActive()) {
    player.stop();
  } else {
    player.play();
  }
}

void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  Serial.println("Connection State Changed");
  Serial.println(out.source().to_str(state));
  
  if (state == ESP_A2D_CONNECTION_STATE_DISCONNECTED) {
    player.stop();
  } else if (state == ESP_A2D_CONNECTION_STATE_CONNECTED) {
    player.begin();
  }
}

void setup() {
  SPI.begin(SD_CLK, SD_MISO, SD_MOSI, SD_CS);

  btn1.begin(BTN_1);
  btn1.setPressedHandler(togglePlay);

  btn2.begin(BTN_2);
  btn2.setLongClickHandler(longClick);

  btn3.begin(BTN_3);
  btn3.setLongClickHandler(longClick);

  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  auto cfg = out.defaultConfig(TX_MODE);
  cfg.auto_reconnect = false;
  cfg.name = "";
  out.begin(cfg);
  out.source().set_on_connection_state_changed(connection_state_changed);

  player.setVolume(vol);
  player.begin();
}

void loop() {

  btn1.loop();
  btn2.loop();
  btn3.loop();

  if (player.isActive()) {
    player.copy();
  }
}

BeijingUncle, 2023-01-22:
When I got reconnect callback, do player.begin(), and the ESP32 rebooted:

16:53:50.592 -> [W] AudioA2DP.h : 300 - ==> state: Connecting
16:53:51.746 -> [W] AudioA2DP.h : 300 - ==> state: Connected
16:53:52.637 -> [W] AudioA2DP.h : 209 - is_a2dp_active -> true with 59904 bytes
16:54:08.833 -> A2DP connection State Changed: Disconnected
16:54:25.783 -> A2DP connection State Changed: Connecting
16:54:25.980 -> A2DP connection State Changed: Connected
16:54:25.980 -> player.begin()
16:54:25.980 -> 
16:54:25.980 -> assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())
16:54:26.012 -> 
16:54:26.012 -> assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())
16:54:26.012 -> 
16:54:26.012 -> 
16:54:26.012 -> Backtrace: 0x40095af7:0x3ffe17e0 0x40083bdd:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 0x40095af7:0x3ffe18a0 0x40095af7:0x3ffe18c0 0x40095af7:0x3ffe1930 0x4008007d:0x3ffe17e0 0x400837ca:0x3ffe1800 0x40095af7:0x3ffe1820 0x40095af7:0x3ffe1850 0x40095af7:0x3ffe1880 |<-CONTINUES
16:54:26.207 -> 

The sketch
-----------------
void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  Serial.print("A2DP connection State Changed: ");
  Serial.println(out.source().to_str(state));
  
  if (state == ESP_A2D_CONNECTION_STATE_CONNECTED) {
    Serial.println("player.begin()");
    player.begin();
  }
}

void setup() {
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.auto_reconnect = false;
  cfg.name = "";
  out.begin(cfg);
  out.source().set_on_connection_state_changed(connection_state_changed);

  player.setVolume(vol);
  player.begin();

pschatzmann, 2023-01-22:
Please learn how to analyse a stack trace and resolve issues yourself

closed 2023-01-22 by pschatzmann