pschatzmann/arduino-audio-tools #575: AudioPlayer, player.setMetadataCallback can not print out metadata

BeijingUncle, 2023-01-16:
void printMetaData(MetaDataType type, const char* str, int len){
  Serial.print("==> ");
  Serial.print(toStr(type));
  Serial.print(": ");
  Serial.println(str);
}

player.setMetadataCallback(printMetaData);
player.setVolume(vol);
player.begin();

[image]

Sorry in chinese

pschatzmann, 2023-01-16:
I committed a correction which also provides some additinal information messages and warnings. I was getting only the title before, now I am getting:

11:14:44.469 -> [I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
11:14:44.469 -> [I] AudioSourceSD.h : 64 - selectStream: 0
11:14:44.782 -> [I] AudioSourceSD.h : 68 - Using file /test2/Old_Money_(feat._Jade_Gritty_&amp;_AURC)_-_Tab_(2).mp3
11:14:44.890 -> [I] AudioCopy.h : 69 - buffer_size=1024
11:14:44.920 -> [I] MetaDataID3.h : 508 - callback TPE1
11:14:44.920 -> ==> Artist: Tab
11:14:44.920 -> [I] MetaDataID3.h : 508 - callback TIT2
11:14:44.920 -> ==> Title: Old Money (feat. Jade Gritty &amp; AURC)
11:14:44.954 -> [I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:14:44.954 -> [I] AudioCopy.h : 134 - Stre

BeijingUncle, 2023-01-16:
It's strange, I use the same mp3 (Old Money from Jamendo), and I get the album name, which you don't get, but I cannot get
both title and artist which you get.

18:52:59.863 -> entry 0x400805e4
18:53:01.776 -> [W] AudioA2DP.h : 300 - ==> state: Connecting
18:53:07.487 -> [W] AudioA2DP.h : 300 - ==> state: Disconnected
18:53:11.872 -> [W] AudioA2DP.h : 300 - ==> state: Connecting
18:53:13.223 -> [W] AudioA2DP.h : 300 - ==> state: Connected
18:53:14.079 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.079 -> ==> Album: Old Money (feat. Jade Gritty &amp; AURC)
18:53:14.079 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.111 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.111 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.111 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.144 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.144 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.176 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.176 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.176 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.209 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.209 -> libhelix -  -> invalid frame size: 0 / max: 3200
18:53:14.243 -> libhelix -  -> invalid frame size: 0 / max: 3200

pschatzmann, 2023-01-16:
Did you update the audio-tools library ?

BeijingUncle, 2023-01-16:
> Did you update the audio-tools library ? That's what has been corrected in the last commit

I download the zip and install again, now get some compilation error:

In file included from /Users/gaopeng/Documents/Arduino/libraries/arduino-audio-tools/src/AudioCodecs/CodecHelix.h:5,
                 from /Users/gaopeng/Documents/Arduino/libraries/arduino-audio-tools/src/AudioCodecs/AudioCodecs.h:16,
                 from /Users/gaopeng/Documents/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:71,
                 from /Users/gaopeng/Documents/Arduino/player-sdfat-a2dp_copy_20230115165357/player-sdfat-a2dp_copy_20230115165357.ino:14:
/Users/gaopeng/Documents/Arduino/libraries/arduino-audio-tools/src/AudioCodecs/CodecMP3Helix.h: In member function 'size_t audio_tools::MP3DecoderHelix::maxFrameSize()':
/Users/gaopeng/Documents/Arduino/libraries/arduino-audio-tools/src/AudioCodecs/CodecMP3Helix.h:153:38: error: 'virtual size_t libhelix::MP3DecoderHelix::maxFrameSize()' is protected within this context
             return mp3->maxFrameSize();
                                      ^

pschatzmann, 2023-01-16:
you also need to upgrade libhelix

BeijingUncle, 2023-01-16:
Now I get Title and Artist, but missing Album ;-)

pschatzmann, 2023-01-16:
The title was before reported as album, which was a mistake: You can check the beginning of the MP3 File if it contains a TALB tag

BeijingUncle, 2023-01-16:
I change back to "Welcome To New York", got some info:

19:38:28.526 -> entry 0x400805e4
19:38:30.407 -> [W] AudioA2DP.h : 314 - ==> state: Connecting
19:38:36.152 -> [W] AudioA2DP.h : 314 - ==> state: Disconnected
19:38:40.507 -> [W] AudioA2DP.h : 314 - ==> state: Connecting
19:38:41.069 -> [W] AudioA2DP.h : 314 - ==> state: Connected
19:38:41.333 -> [W] MetaDataID3.h : 448 - TAG TALB ignored
19:38:41.333 -> [W] MetaDataID3.h : 448 - TAG TPE1 ignored
19:38:41.366 -> [W] MetaDataID3.h : 448 - TAG TIT2 ignored
19:38:41.366 -> [W] MetaDataID3.h : 448 - TAG TCON ignored
19:38:41.432 -> [W] AudioA2DP.h : 216 - is_a2dp_active -> true with 13824 bytes

pschatzmann, 2023-01-16:
I had plenty of issues with invalid tag content that made the system crash, so I added a check that only accepts the tag if if finds some valid ASCII characters. If you get this message this check fails
File MetaDataID3.h line 445 isAscii() check is failing....

BeijingUncle, 2023-01-16:
And the player begin to lost data again, I already change:
// #define MP3_MAX_OUTPUT_SIZE 1024 * 5
// #define MP3_MAX_FRAME_SIZE 1600 

#define MP3_MAX_OUTPUT_SIZE 1024 * 20
#define MP3_MAX_FRAME_SIZE 1600 * 2

in MP3DecoderHelix.h

So I have to downgrade both libs to previous version, and songs play well again

pschatzmann, 2023-01-16:
I suggest that you use the new methods in your sketch. See the readme of libhelix

pschatzmann, 2023-01-22:
Chinese characters are not supported

closed 2023-01-22 by pschatzmann