pschatzmann/arduino-audio-tools #578: Detect SD end

oldmanskier, 2023-01-18:
Is there a way to detect the end of a SD card. Card has 200 or more songs.  I presently count the files on the SD but this take several seconds before playing begins.  I have saved the last stream number in preferences so I can begin where I turned off the player without starting over from the beginning. I compare the last stream number with the total file count and reset last stream to zero when file count is reached.  I would like to detect the last file so I could start from the beginning without the long delay to count the files.

pschatzmann, 2023-01-19:
Feel free to extend the functionality.

pschatzmann, 2023-01-22:
I assume you could resolve the issue with the proposed solution.

closed 2023-01-22 by pschatzmann