pschatzmann/arduino-audio-tools #580: Of the correct usage of player.setPath ([W] AudioPlayer.h : 144 - -> begin: no data found)

0x0fe, 2023-01-23:
So, i may have a wrong usage of player.setPath, or there is a bug. I am reading mp3 from SD card via AudioSourceSDMMC, when i play files inside directories at the root of the SD it works fine, i set the parth with  
`player.setPath("/K0012"); `
And it will automatically find the files in this directory and play them one after the other.
However, when i move all the directories inside a subdirectory MKI01, then i got no data found error
```
player.setPath("/MKI01/K0012"); 
[W] AudioPlayer.h : 144 - -> begin: no data found
```
I dont understand what the cause can be, here is a picture of the SD structure:
[image]
That is how i setup the player:
```
player.setVolume(0.9); 
player.setMetadataCallback(printMetaData);
player.setAutoNext(true);
```

0x0fe, 2023-01-23:
there is no error to reach the directory, but it fails to find the datas, which it did not when the directory was at the root of SD. Each directory has 3 mp3 files inside, named d1f1.mp3, d1f2.mp3, d1f3.mp3 for K0010 for example.

pschatzmann, 2023-01-23:
Strange: it should explore the path recursively, so all of the following should find the data
/
/MKI01
/MKI01/K0012

You can set the log level to Info or Debug to see in more detail what the system is doing...

0x0fe, 2023-01-23:
[image]
even more strange, it can see the file

0x0fe, 2023-01-23:
tried with another file, same problem. However all these files are working fine when their directory is at SD root. So there is really some kind of bug it seems.
[image]

pschatzmann, 2023-01-23:
Just read me answer above: you are not using setPath() correctly !

pschatzmann, 2023-01-23:
I was just double checking the code and I think your code should work.
What happens if you give the start path only in the constructor ?

0x0fe, 2023-01-23:
`begin :  no data found`,
even though it can clearly find the file.
[image]

If i remove MKI01 and move K0010 at the root of the SD card then it works fine, it can read all files inside each directory as long as i set the directory path : player.setPath("/K0012");

pschatzmann, 2023-01-23:
Maybe this is a version issue. Double check in the Board Manager: What ESP32 version are you using ? 
Are you using the latest version of the SDMMC library.
At some point of time the filename() reported the full path and that was changed to report the file name only...

0x0fe, 2023-01-23:
i use 2.0.6

SD_MMC version 2.0.0 dans le dossier: FS version 2.0.0 dans le dossier: arduino-audio-tools-1505524d5925643ba7378542f101229d54551f3f version 0.9.3

0x0fe, 2023-01-23:
i think you are right, it must be the SD_MMC version maybe, because : 
[image]

0x0fe, 2023-01-23:
but, i could not find SD_MMC in local libraries, so i assume its part of the ESP32 core, which is up to date

0x0fe, 2023-01-23:
i checked, actually the implmentation is in FS, not SD_MMC, and i can confirm file.name only returns the short name, to get the path it is file.path
"You could try to use the SD library instead which is part of the ESP32 and does not require any separate installation."
not sure about that, SD_MMC is part of ESP32 core, and my hardware implementation is 1bit SDIO, i cannot use SD via SPI.

0x0fe, 2023-01-23:
so, one more clue, it tries to open the wrong path, even though this time i removed the MKI01 directory and move K0012 to the SD card root.

[image]

pschatzmann, 2023-01-23:
I think there is a bug in the   const char* fileNamePath(FileT &file) method in SDIndex.h.

0x0fe, 2023-01-23:
ok, i check

0x0fe, 2023-01-23:
[image]

mmh, so, it seems that when we dont use SDFAT, which is my case here, it returns just the filename without path

pschatzmann, 2023-01-23:
I just committed a correction

0x0fe, 2023-01-23:
ok i replace with last comit and test

0x0fe, 2023-01-23:
so, i just tested and i can confirm the commit indeed fixe dthe problem, now it can read correctly file files regardless of the directory level. 
However i notice a new bug, which i never noticed before (maybe i missed it). 

So i have two test directories, each has 3 files inside, and the bug is that it is playing randomly either 2 or 3 files. On the first directory it starts on file1 and plays ile 2, sometimes it will play file3. On the second directory for some reason it always starts on file 2 and and only read file 2 and file 3.

 I have made some test mp3 files which tell their names, so i attached them if ever you want to test. There is also a problem of digital "fart" when it transition between two files and when i use player.stop() (for pause / breakpoint function) this problem is unrelated to the path bug, i guess i will open this in another ticket. I have found one very dirty "fix" but it does not quite work well and surely doesnt solve the issue at the source.

[MKI01.zip](https://github.com/pschatzmann/arduino-audio-tools/files/10481706/MKI01.zip)

0x0fe, 2023-01-24:
so, on this issue i noticed a pattern,

regarding first directory (K0010) 
first playback : file 2, file 3, file 1
subsequent playbacks : file 2, file 3

```
path : /MKI01/K0010
[I] I2SConfig.h : 104 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 105 - port_no: 0
[I] I2SConfig.h : 106 - is_master: Master
[I] I2SConfig.h : 107 - sample rate: 44100
[I] I2SConfig.h : 108 - bits per sample: 1path : /MKI01/K0010
[I] AudioSourceSDMMC.h : 75 - -> selectStream: /MKI01/K0010
[I] AudioEncoded.h : 321 - virtual void audio_tools::EncodedAudioPrint::end()
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 70 - buffer_size=2048
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSourceSDMMC.h : 62 - selectStream: 0
[I] AudioSourceSDMMC.h : 66 - Using file /MKI01/K0010/d1f2.mp3
[I] AudioCopy.h : 70 - buffer_size=2048
[I] I2SStream.h : 68 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioBaseInfo)
[I] AudioTypes.h : 56 - sample_rate: 44100
[I] AudioTypes.h : 57 - channels: 2
[I] AudioTypes.h : 58 - bits_per_sample: 16
[I] AudioCopy.h : 147 - mute
[W] AudioPlayer.h : 419 - -> timeout - moving by 1
[I] AudioSourceSDMMC.h : 57 - nextStream: 1
[I] AudioSourceSDMMC.h : 62 - selectStream: 1
[I] AudioSourceSDMMC.h : 66 - Using file /MKI01/K0010/d1f3.mp3
[I] AudioEncoded.h : 321 - virtual void audio_tools::EncodedAudioPrint::end()
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 70 - buffer_size=2048
[I] I2SStream.h : 68 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioBaseInfo)
[I] AudioTypes.h : 56 - sample_rate: 44100
[I] AudioTypes.h : 57 - channels: 2
[I] AudioTypes.h : 58 - bits_per_sample: 16
[I] AudioCopy.h : 147 - mute
[W] AudioPlayer.h : 419 - -> timeout - moving by 1
[I] AudioSourceSDMMC.h : 57 - nextStream: 1
[I] AudioSourceSDMMC.h : 62 - selectStream: 2
[I] AudioSourceSDMMC.h : 66 - Using file /MKI01/K0010/d1f1.mp3
[I] AudioEncoded.h : 321 - virtual void audio_tools::EncodedAudioPrint::end()
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 70 - buffer_size=2048
[I] I2SStream.h : 68 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioBaseInfo)
[I] AudioTypes.h : 56 - sample_rate: 44100
[I] AudioTypes.h : 57 - channels: 2
[I] AudioTypes.h : 58 - bits_per_sample: 16
[I] AudioCopy.h : 147 - mute
[W] AudioPlayer.h : 419 - -> timeout - moving by 1
[I] AudioSourceSDMMC.h : 57 - nextStream: 1
[I] AudioSourceSDMMC.h : 62 - selectStream: 3
[I] AudioEncoded.h : 321 - virtual void audio_tools::EncodedAudioPrint::end()
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
```

```
path : /MKI01/K0010
[I] AudioSourceSDMMC.h : 75 - -> selectStream: /MKI01/K0010
[I] AudioEncoded.h : 321 - virtual void audio_tools::EncodedAudioPrint::end()
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 70 - buffer_size=2048
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSourceSDMMC.h : 62 - selectStream: 0
[I] AudioSourceSDMMC.h : 66 - Using file /MKI01/K0010/d1f2.mp3
[... 100 lines cut ...]
[I] AudioCopy.h : 147 - mute
[W] AudioPlayer.h : 419 - -> timeout - moving by 1
[I] AudioSourceSDMMC.h : 57 - nextStream: 1
[I] AudioSourceSDMMC.h : 62 - selectStream: 1
[I] AudioSourceSDMMC.h : 66 - Using file /MKI01/K0012/d2f2.mp3
[I] AudioEncoded.h : 321 - virtual void audio_tools::EncodedAudioPrint::end()
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 70 - buffer_size=2048
[I] I2SStream.h : 68 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioBaseInfo)
[I] AudioTypes.h : 56 - sample_rate: 44100
[I] AudioTypes.h : 57 - channels: 2
[I] AudioTypes.h : 58 - bits_per_sample: 16
[I] AudioCopy.h : 147 - mute
[W] AudioPlayer.h : 419 - -> timeout - moving by 1
[I] AudioSourceSDMMC.h : 57 - nextStream: 1
[I] AudioSourceSDMMC.h : 62 - selectStream: 2
[I] AudioEncoded.h : 321 - virtual void audio_tools::EncodedAudioPrint::end()
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
```

So there is something wrong with the way it is indexing, for some reason only the first playback reads the correct number of files, any subsequent playback seems to ignore one file (in my case) and forsome reason, in one directory, alwyas the same, it does not index the file ending with 1 (d1f1.mp3) as the index 0.
The naming convention is the same in both directories, the first iles are d1f1.mp3 (K0010) and d2f1.mp3 (K0012).
I parsed SDindex and SDdirect to find here the indexing is done but it is still unclear to me, i noticed some functions creating and index file but it doesnt seem to be used in my case since i did not saw any index file in teh SD. 
Could you point me to the location where this indexing is determine, in particular the section of the first file and the counting of all files in a directry, so i can maybe fix them.

pschatzmann, 2023-01-24:
The SDdirect is not indexing! It is just recursively reading the files and directories using the File API. 
So I guess the issue must be in the basic File API provided by the ESP32.

0x0fe, 2023-01-24:
yes, indeed i can confirm the issue is reltaed to ESP32 VFS API, i have tested outside the library and i get the same result : 

```
  sprintf(filePath,"/MKI01/K%s",suffix);
  File root = SD_MMC.open(filePath);
  root.rewindDirectory();
  File entry = root.openNextFile();
  player.setPath(entry.path());
  printf("\npath : %s\n",entry.path());
  root.close();
  entry.close();
```
`path : /MKI01/K0010/d1f2.mp3`

i have a working implementation with QSort, now i need to see how to integrate this into the library 

https://www.onlinegdb.com/03wbO1F06

[image]

0x0fe, 2023-01-24:
should i implement it here? 

const char *operator[](int idx) or SDDirect.h

[image]

pschatzmann, 2023-01-24:
I don't think that this is possible. I would suggest to use the implementation which uses an index or to implement your own Data source which would operate on one directory only.

0x0fe, 2023-01-24:
i am not sure how to use the implementation with index file, i havent seen an example of it 

"implement your own Data source which would operate on one directory only." 
is there any example of that in the library? 

Operating on one directory is fine, because currently that is how it works, the devise is always set to a specific directory by some external means (nfc).

pschatzmann, 2023-01-24:
Have a look at player-callback-i2s.ino. This is creating a custom data source using the SD library with the help of AudioSourceCallback. Or you could just create your own subclass or AudioSource

0x0fe, 2023-01-24:
oh i see, nice, i look into this..
I also noticed this test https://github.com/pschatzmann/arduino-audio-tools/blob/9c4f3fd2497154e8b97c5e0c5c54000dcd20a449/examples/tests/sd/test-index-sd/test-index-sd.ino
Cna it be used with SD_MMC or only with SPI SD?

pschatzmann, 2023-01-24:
It should work with all implementations

0x0fe, 2023-01-24:
yes i see, i will try the callback first, maybe with small modification i can integrate the qsort and handle various directories more easily. I can keep the filelist of one directory in ram.