pschatzmann/arduino-audio-tools #587: BluetoothA2DPSink.h: No such file or directory

QBGROUP, 2023-01-25:
Trying to compile streams-i2s-a2dp I get this error.
The file is missing and I cannot find it anywhere, what is wrong?
Thank you

pschatzmann, 2023-01-25:
I guess you just forgot to install the [a2dp](https://github.com/pschatzmann/ESP32-A2DP) library!
I recommend that you read the documentation before posting any issues. This is e.g. documented in https://github.com/pschatzmann/arduino-audio-tools/wiki/Optional-Libraries

closed 2023-01-25 by pschatzmann

QBGROUP, 2023-01-25:
I'm really sorry! thank you for your reply, I'll read more, I'm new to the library.
Is it possible for an esp32 act as a bluetooth microphone? I'd like to get audio from a i2s mic and be able to get the source with a bt device on request.
Have a nice evening

pschatzmann, 2023-01-25:
It's pretty challenging and full of pitfalls...