pschatzmann/arduino-audio-tools #599: Should free function mapFloat in header be inline?

chrissbarr, 2023-01-30:
I am including `AudioTools.h` in one of my project's header files. This header file is then included by multiple source files.

`AudioTools.h` includes `AudioTools/AudioTypes.h`. `AudioTypes.h` declares and defines the function `mapFloat` within the header:

https://github.com/pschatzmann/arduino-audio-tools/blob/f33df5e65fc186cf67d05b4f5f94d137a27a5c07/src/AudioTools/AudioTypes.h#L211

Because this function is defined in the header, `AudioTools/AudioTypes.h` being included repeatedly is giving me multiple definition errors for `mapFloat` during linking:

```
...
/.pio/libdeps/esp32/audio-tools/src/AudioTools/AudioTypes.h:211: multiple definition of `audio_tools::mapFloat(float, float, float, float, float)';
./.pio/libdeps/esp32/audio-tools/src/AudioTools/AudioTypes.h:211: first defined here
...
```

My understanding (which may be incomplete) is that there are generally two simple solutions:
1 - Separating the definition of the function into a source file
2 - Marking the function definition in the header inline

Is there any reason why this function couldn't be marked inline?

pschatzmann, 2023-01-30:
Since this is a header only library, option 1 does not fit.
That's a recent change and I agree that it should be inline or static.
Can you submit a PR ?

chrissbarr, 2023-01-30:
Yep, will do shortly!

closed 2023-01-30 by chrissbarr