pschatzmann/arduino-audio-tools #608: Feature request: Is it possible to add LoRa communication method like ESPNow with Codec2 ?

happytm, 2023-02-04:
Thank you for providing this great library for using audio features of ESP32.

I would like to use this library to make audio link using Codec2 protocol between 2 ESP32 devices equipped with LoRa radios.

I have simple project using ESPNow or Lora with codec2 protocol here https://github.com/happytm/AudioTransport/tree/main/test

Is it even possible?

Thanks.

pschatzmann, 2023-02-04:
There is nothing to do from my side, just use the standard Lora library: As you can see [LoRaClass](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.h) is a subclass of Stream

closed 2023-02-04 by pschatzmann

happytm, 2023-02-04:
Excellent. I looked for ESPNow example with Codec2 protocol but did not find one. Can you point me to such example if there is one? 

Thanks.

pschatzmann, 2023-02-05:
There is an [example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-communication/esp-now-codec/communication-codec-espnow-receive/communication-codec-espnow-receive.ino) that shows how to use esp now with a codec. Just replace the used codec with any other...

happytm, 2023-02-05:
Thank you Sir.