pschatzmann/arduino-audio-tools #610: Sourcing Audio problem

bool-shark, 2023-02-06:
Amazing library. Most examples tried worked very well with nary a hitch.
Interestingly the one I'm keen on "player-sdfat-a2dp" isn't working for me.
Audio is breaking up, cutting in and out with a bit of a lag.
Tried both Audiokit and generic wroom board. Tried a couple different
Mp3s, 2 channel, 16 bit, 44100 as well as a few other variations and different sinks.
Welcome any suggestions. 
Thanks

pschatzmann, 2023-02-06:
I think the [example using the basic API](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/base-player-a2dp/base-player-a2dp.ino) is behaving a little better

bool-shark, 2023-02-06:
That was fast.  Can confirm your suggestion working. Much appreciated Phil.

bool-shark, 2023-02-06:
I should also mention in case this anybody else  has this experience and I'm sure this is outside of Phil's control (with the many variations of these thinker boards) that I get a "failure to compile for board" in a few examples, 1  in every 3 attempts . I try it again immediately and it compiles, not sure why. I have a V2.2 A210 board using option 5 in Audiokit settings.  Thanks again Phil.

closed 2023-02-07 by bool-shark