pschatzmann/arduino-audio-tools #611: Inputmixer.readBytes function gives a reboot when using 2 channels, while kit.readbytes etc. works fine.

onyx99, 2023-02-07:
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

AudioKitStream kit;
InputMixer<int16_t> mixer;

uint16_t sample_rate=44100;
uint8_t channels = 2;

void setup(){
  AudioBaseInfo info;
  info.channels = channels;
  info.sample_rate = sample_rate;
  info.bits_per_sample = 16;
  
  //kit config here
  
  mixer.add(kit);
  mixer.begin(info);
 
}

uint8_t buffer[1024];

void loop() {
  
  if(mixer.available()>0){
    int samplesRead = mixer.readBytes(buffer, sizeof(buffer)); //reboots here
  }

}

00:07:51.016 -> Rebooting...
00:07:51.016 -> ets Jun  8 2016 00:22:57
00:07:51.016 -> 
00:07:51.016 -> rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
00:07:51.016 -> configsip: 0, SPIWP:0xee
00:07:51.016 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:07:51.016 -> mode:DIO, clock div:1
00:07:51.016 -> load:0x3fff0030,len:1344
00:07:51.016 -> load:0x40078000,len:13864
00:07:51.016 -> load:0x40080400,len:3608
00:07:51.016 -> entry 0x400805f0

#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

AudioKitStream kit;
InputMixer<int16_t> mixer;

uint16_t sample_rate=44100;
uint8_t channels = 2;

void setup(){
  AudioBaseInfo info;
  info.channels = channels;
  info.sample_rate = sample_rate;
  info.bits_per_sample = 16;
  
  //kit config here
  
  mixer.add(kit);
  mixer.begin(info);
 
}

uint8_t buffer[1024];

void loop() {

  if(kit.available()>0){
    int samplesRead = kit.readBytes(buffer, sizeof(buffer)); //works fine
  }

}

i really need the mixer so i hope it can be solved, thanks.

pschatzmann, 2023-02-08:
My first guess would be that the issue is, that you forgot to start the kit!

onyx99, 2023-02-08:
Kit works fine here, and i get the readBytes from kit, but when i try to take the bytes from mixer when i use 2 channels then it crashes on line mixer.readByte, 1 channel seems to work when i read the bytes, but 2 channels fails, i need stereo

I also tried a working example from the examples: streams-generator_inputmixer-audiokit and modified just 1 line so that i could take the decoded frames, but it also crashes when i try to take the bytes, but if i set the channels to 1 then it works 

see below:

/**
 * @file streams-generator_inputmixer-audiokit.ino
 * @brief Tesing input mixer
 * @author Phil Schatzmann
 * @copyright GPLv3
 */
 
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

uint16_t sample_rate=32000;
uint8_t channels = 2;                                  // The stream will CRASH with 2 channels, but works when set to 1
SineWaveGenerator<int16_t> sineWave1(32000);                // subclass of SoundGenerator with max amplitude of 32000
SineWaveGenerator<int16_t> sineWave2(32000);                // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int16_t> sound1(sineWave1);             // Stream generated from sine wave
GeneratedSoundStream<int16_t> sound2(sineWave2);             // Stream generated from sine wave
InputMixer<int16_t> mixer;
AudioKitStream out; 
StreamCopy copier(out, mixer);                             // copies sound into i2s
AudioBaseInfo info;

uint8_t buffer[1024];

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup audio info
  info.channels = channels;
  info.sample_rate = sample_rate;
  info.bits_per_sample = 16;

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  config.copyFrom(info); 
  out.begin(config);

  // Setup sine wave
  sineWave1.begin(channels, sample_rate, N_B4);
  sineWave2.begin(channels, sample_rate, N_E4);

  mixer.add(sound1);
  mixer.add(sound2);
  mixer.begin(info);

  Serial.println("started...");
}

// Arduino loop - copy sound to out 
void loop() {

  if(mixer.available()>0) {
    int samplesRead = mixer.readBytes(buffer, sizeof(buffer)); //crashes here
  }
  
  //copier.copy(); //makes no difference when i comment this out or not
}

11:06:25.107 -> rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
11:06:25.107 -> configsip: 0, SPIWP:0xee
11:06:25.107 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:06:25.107 -> mode:DIO, clock div:1
11:06:25.107 -> load:0x3fff0030,len:1344
11:06:25.154 -> load:0x40078000,len:13864
11:06:25.154 -> load:0x40080400,len:3608
11:06:25.154 -> entry 0x400805f0

onyx99, 2023-02-09:
Any second guess ?

pschatzmann, 2023-02-09:
Are you sure that you use the latest version. I can't reproduce your errors and the following test sketch seems to work just fine:
```

#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

AudioKitStream kit;
InputMixer<int16_t> mixer;
CsvStream<int16_t> csv(Serial);
uint16_t sample_rate = 44100;
uint8_t channels = 2;
StreamCopy copier(csv, mixer); 
AudioBaseInfo info;

void setup() {
  Serial.begin(119200);
  info.channels = channels;
  info.sample_rate = sample_rate;
  info.bits_per_sample = 16;

  //kit config here
  auto cfg = kit.defaultConfig();
  cfg.input_device = AUDIO_HAL_ADC_INPUT_LINE2;
  cfg.copyFrom(info);
  kit.begin(cfg);

  mixer.add(kit);
  mixer.begin(info);
  csv.begin(info);
}

void loop() {
  copier.copy();
}
```

onyx99, 2023-02-09:
Are you sure you tried with these lines added in the loop ?, the sketch works fine until i add:

uint8_t buffer[1024];

if(mixer.available()>0) {
int samplesRead = mixer.readBytes(buffer, sizeof(buffer)); //crashes here
}

I use v1.06 of ESP32 and i tried with 2x Ai-Thinker ESP32-Audio-Kit board, and option 6 in the settings to select the board.

onyx99, 2023-02-09:
I am very happy that it wont crash at your side, with me it crashes only when i try to take samples from the mixer with 2 channels in the config, with 1 channel it works fine, i will go try to check the boards with a scope (voltage etc.) ps. did you test with similair boards? i also have a bunch of ESP32-A1S chips to test with.

pschatzmann, 2023-02-09:
Did you delete the AudioTools library and installed the latest version ?

onyx99, 2023-02-09:
yes i used the latest version from giti also tried to delete and re-install, buy what i still didn't is to remove the whole Arduino IDE, and do a fresh/clean install. so i will try that first.

pschatzmann, 2023-02-09:
The Arduino IDE is not relevant. The latest ESP32 Version in the Board manager is 2.0.6

onyx99, 2023-02-09:
Yes it was a typo, i ment v2.0.6 instead of v1.0.6, with trouble i always install the latest sources, i know that the Arduino IDE itself is not really relevant, but i ment all the sources installed with and after it, because the faillure has to be a reason, and then why not install also the latest version of the IDE.

pschatzmann, 2023-02-09:
I added some more debug statements,  so if it is still crashing you can set the log level to debug to figure out where the error is coming from.
The only issue that I noticed, is that when I output a sine wave via the Input mixer I get the wrong frequency out. However the same test case using jupyter is working w/o issues.

onyx99, 2023-02-09:
It works now! the only problem still is that the mixer sounds double the speed when i use the readbytes sample output, i tested with tone input and also with line input at 44k1 and 2 channels, when i set it to 44k1 and 1 channel then the tone height is correct.

pschatzmann, 2023-02-10:
I finally found the issue with the wrong frequencies. The Sine generator did not support readBytes with a length smaller then the frame size. The correction has been committed

onyx99, 2023-02-10:
Thank you! the sinegenerator works fine now, the kit as inputdevice seems to have the same or similair problem trough inputmixer.

closed 2023-02-10 by pschatzmann