pschatzmann/arduino-audio-tools #612: Issue in the URLStream destruction order

hummusawy, 2023-02-08:
First, thank you for your amazing effort!

In the distractor of the `URLStream`  the `clientSecure` and the `clientInsecure` get deleted before calling the `end()` method, which lead to a `null pointer exception` when the `URLStream` get distracted because the `end()` method calls `        request.stop()` which calls `client_ptr->stop()` so the client should be there when the `end()` method get called. 

I pushed the `end()` method call to the beginning of the distractor and the distraction working fine with me after doing this change. 

```
        ~URLStream(){
            TRACEI();
            end();
#ifdef USE_WIFI_CLIENT_SECURE
            if (clientSecure!=nullptr){
                delete clientSecure;
                clientSecure = nullptr;
            }
#endif
            if (clientInsecure!=nullptr){
                delete clientInsecure;
                clientInsecure = nullptr;
            }
        }
```

pschatzmann, 2023-02-08:
correction has been committed

closed 2023-02-08 by pschatzmann

hummusawy, 2023-02-08:
Super!