pschatzmann/arduino-audio-tools #618: I updated the library to use the latest commit but only the speedproblem with the iputdevice trough inputmixer still exists.

onyx99, 2023-02-10:
Why did you close the issue? i already used the latest commit d6a58d1 before i tried, only the SineWaveGenerator issue is fixed, but the line-in still has the wrong frequency when playing 2 channels as i mentioned.

pschatzmann, 2023-02-10:
No, you did not mention this. 
Please provide a complete sketch where I can reproduce this!

onyx99, 2023-02-10:
You are now referring to a topic (issue) wich is not mine! 

my first was #611 but you decided to close #611, so i opened #617 because a (line in trough inputmixer) frequency issue still existed after updating to the latest commit d6a58d1 but you closed that one immediatly so i opened a new one #618 because that issue still exists.

pschatzmann, 2023-02-10:
I start to have the impression that you are wasting my time.
Here is the sketch that is proofing that it is working as expected:
```

#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

uint16_t sample_rate = 44100;
uint8_t channels = 2;
uint8_t bits_per_sample = 16;

AudioKitStream kit;
InputMixer<int16_t> mixer;
CsvStream<int16_t> csv(Serial);
StreamCopy copier(kit, mixer); 
AudioBaseInfo info;

void setup() {
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  Serial.begin(119200);

  info.channels = channels;
  info.sample_rate = sample_rate;
  info.bits_per_sample = bits_per_sample;

  //kit config here
  auto cfg = kit.defaultConfig(RXTX_MODE);
  cfg.copyFrom(info);
  cfg.input_device = AUDIO_HAL_ADC_INPUT_LINE2;
  cfg.output_device = AUDIO_HAL_DAC_OUTPUT_LINE1;
  kit.begin(cfg);

  mixer.add(kit);
  mixer.begin(info);
  csv.begin(info);
}

void loop() {
  copier.copy();
}
```
I was recording the input from the microphones , running thru the mixer and was recrdoing the result in Audacity. It has the same frequency like the recorded test signal!